MichaIng/DietPi #6483: Odriod N2+ | rngd service doesn't find any entropy source on boot

Zer0x00, 2023-07-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
```
- Distro version | `bookworm`
- Kernel version | `Linux pi 6.1.11-meson64 #23.02.2 SMP PREEMPT Sat Feb 18 00:07:55 UTC 2023 aarch64 GNU/Linux`
- SBC model | `Odroid N2 (aarch64)` (it's actually an Odroid N2+)
- Power supply used | Vendor supplied
- SD card used | 128 GB eMMC Vendor supplied

#### Additional Information (if applicable)
- Software title | None
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi? | Yes

- Bug report ID | `8dbe998a-fa90-423f-b941-1794af4bf499`

#### Steps to reproduce

1. Install dietpi freshly on a ODROID N2+ with the latest release
2. Reboot (don't know if this is really necessary or a fresh install does behave like this also)
3. Type `service rngd status`
4. You'll see that the service isn't running

#### Expected behaviour

- The rngd service should start just fine and be running

#### Actual behaviour

- The rngd service doesn't find and entropy source and gets deactivated

#### Extra details
```bash
root@pi:~# journalctl
[...]
Jul 15 22:48:38 pi systemd[1]: Started rngd.service - Start entropy gathering daemon (rngd).
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
[...]
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: No entropy sources working, exiting rngd
Jul 15 22:48:38 pi systemd[1]: rngd.service: Deactivated successfully
```

```bash
root@pi:~# service rngd status
○ rngd.service - Start entropy gathering daemon (rngd)
     Loaded: loaded (/lib/systemd/system/rngd.service; enabled; preset: enabled)
     Active: inactive (dead) since Sat 2023-07-15 22:48:38 CEST; 18min ago
   Duration: 10ms
       Docs: man:rngd(8)
   Main PID: 920 (code=exited, status=0/SUCCESS)
        CPU: 3ms

Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: read error
Jul 15 22:48:38 pi rngd[920]: No entropy sources working, exiting rngd
Jul 15 22:48:38 pi systemd[1]: rngd.service: Deactivated successfully.
```

**Fix / Workaround**
Inserting `ExecStartPre=/bin/sleep 5` under `[Service]` in /lib/systemd/system/rngd.service fixes this behaviour reliably.

MichaIng, 2023-07-15:
Strange, it works well here:
```console
root@OdroidN2:~# journalctl -u rngd
May 03 16:31:43 OdroidN2 systemd[1]: Started rngd.service - Start entropy gathering daemon (rngd).
root@OdroidN2:~# cat /proc/device-tree/model
Hardkernel ODROID-N2Plus
root@OdroidN2:~# uname -a
Linux OdroidN2 6.1.11-meson64 #23.02.2 SMP PREEMPT Sat Feb 18 00:07:55 UTC 2023 aarch64 GNU/Linux
```
Seem all identical. The `rngd.service` is extremely simple, just with default dependencies and no further ordering. The kernel driver should be loaded automatically. Can you check:
```sh
lsmod
```
The order in which modules are listed from buttom to top is the chronological order in which they are loaded. So we can check whether `meson_rng` is loaded moreless at the same relative time, or e.g. later in your case. Sadly `dmesg` seems to not contain any log about it, just one about when the entropy pool has been filled:
```console
root@OdroidN2:~# dmesg | grep -iE 'rng|rand'
[    5.992676] random: crng init done
```

Zer0x00, 2023-07-15:
I did multiple fresh installs and this issue occurs on every install.

```bash
root@pi:~# lsmod
Module                  Size  Used by
snd_soc_hdmi_codec     24576  1
snd_soc_meson_g12a_tohdmitx    20480  1
snd_soc_meson_g12a_toacodec    20480  1
snd_soc_meson_codec_glue    16384  2 snd_soc_meson_g12a_toacodec,snd_soc_meson_g12a_tohdmitx
snd_soc_meson_axg_tdmout    24576  2
meson_vdec             81920  0
snd_soc_meson_axg_toddr    28672  3
videobuf2_dma_contig    24576  1 meson_vdec
snd_soc_meson_axg_tdmin    24576  4
v4l2_mem2mem           36864  1 meson_vdec
snd_soc_meson_axg_frddr    36864  3
snd_soc_meson_axg_fifo    20480  2 snd_soc_meson_axg_toddr,snd_soc_meson_axg_frddr
ir_nec_decoder         20480  0
dw_hdmi_i2s_audio      16384  0
polyval_ce             16384  0
meson_rng              16384  0
videobuf2_memops       20480  1 videobuf2_dma_contig
snd_soc_meson_axg_sound_card    16384  0
polyval_generic        16384  1 polyval_ce
snd_soc_meson_card_utils    16384  1 snd_soc_meson_axg_sound_card
snd_soc_meson_axg_tdm_interface    16384  3 snd_soc_meson_axg_sound_card
snd_soc_meson_axg_tdm_formatter    16384  3 snd_soc_meson_axg_tdmin,snd_soc_meson_axg_tdm_interface,snd_soc_meson_axg_tdmout
snd_soc_meson_t9015    24576  1
snd_soc_simple_amplifier    16384  1
videobuf2_v4l2         32768  2 meson_vdec,v4l2_mem2mem
rc_odroid              16384  0
meson_gxbb_wdt         16384  0
snd_soc_core          245760  15 snd_soc_meson_axg_tdm_formatter,snd_soc_meson_t9015,snd_soc_meson_g12a_toacodec,snd_soc_meson_axg_sound_card,snd_soc_hdmi_codec,snd_soc_meson_g12a_tohdmitx,snd_soc_meson_axg_tdmin,snd_soc_simple_amplifier,snd_soc_meson_axg_tdm_interface,snd_soc_meson_axg_fifo,snd_soc_meson_axg_tdmout,snd_soc_meson_axg_toddr,snd_soc_meson_axg_frddr,snd_soc_meson_card_utils,snd_soc_meson_codec_glue
videobuf2_common       57344  5 meson_vdec,videobuf2_dma_contig,videobuf2_v4l2,v4l2_mem2mem,videobuf2_memops
snd_compress           24576  1 snd_soc_core
panfrost               69632  0
videodev              241664  4 meson_vdec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
meson_ir               16384  0
gpu_sched              36864  1 panfrost
meson_saradc           24576  0
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
mc                     65536  4 videodev,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
rc_core                57344  4 ir_nec_decoder,meson_ir,rc_odroid
snd_pcm               131072  8 snd_soc_hdmi_codec,snd_compress,snd_soc_meson_axg_tdm_interface,snd_soc_core,snd_soc_meson_axg_fifo,snd_soc_meson_axg_toddr,snd_soc_meson_codec_glue,snd_pcm_dmaengine
gpio_fan               16384  0
simpledrm              24576  0
nvmem_meson_efuse      16384  0
snd_timer              40960  1 snd_pcm
drm_shmem_helper       24576  2 panfrost,simpledrm
snd                    98304  5 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
dm_mod                143360  0
efi_pstore             16384  0
ip_tables              32768  0
x_tables               45056  1 ip_tables
autofs4                45056  3
meson_gxl              20480  0
realtek                32768  1
dwmac_generic          16384  0
axg_audio              77824  49
rtc_pcf8563            24576  0
sclk_div               16384  1 axg_audio
reset_meson_audio_arb    16384  6
clk_phase              16384  1 axg_audio
rtc_meson_vrtc         20480  1
dwmac_meson8b          16384  0
```

MichaIng, 2023-07-15:
Hmm, it loads indeed later, compared to the other modules, as in my case:
```console
root@OdroidN2:~# lsmod
Module                  Size  Used by
8821cu               2035712  0
cfg80211              430080  1 8821cu
rfkill                 32768  3 cfg80211
snd_soc_hdmi_codec     24576  1
snd_soc_meson_axg_sound_card    16384  0
panfrost               69632  0
snd_soc_meson_axg_tdm_interface    16384  3 snd_soc_meson_axg_sound_card
snd_soc_meson_axg_toddr    28672  3
snd_soc_meson_axg_frddr    36864  3
snd_soc_meson_card_utils    16384  1 snd_soc_meson_axg_sound_card
snd_soc_meson_g12a_tohdmitx    20480  1
snd_soc_meson_axg_tdmout    24576  2
snd_soc_meson_axg_tdmin    24576  4
snd_soc_simple_amplifier    16384  1
snd_soc_meson_axg_fifo    20480  2 snd_soc_meson_axg_toddr,snd_soc_meson_axg_frddr
snd_soc_meson_t9015    24576  1
polyval_ce             16384  0
snd_soc_meson_axg_tdm_formatter    16384  3 snd_soc_meson_axg_tdmin,snd_soc_meson_axg_tdm_interface,snd_soc_meson_axg_tdmout
snd_soc_meson_g12a_toacodec    20480  1
ir_nec_decoder         20480  0
snd_soc_meson_codec_glue    16384  2 snd_soc_meson_g12a_toacodec,snd_soc_meson_g12a_tohdmitx
snd_soc_core          245760  15 snd_soc_meson_axg_tdm_formatter,snd_soc_meson_t9015,snd_soc_meson_g12a_toacodec,snd_soc_meson_axg_sound_card,snd_soc_hdmi_codec,snd_soc_meson_g12a_tohdmitx,snd_soc_meson_axg_tdmin,snd_soc_simple_amplifier,snd_soc_meson_axg_tdm_interface,snd_soc_meson_axg_fifo,snd_soc_meson_axg_tdmout,snd_soc_meson_axg_toddr,snd_soc_meson_axg_frddr,snd_soc_meson_card_utils,snd_soc_meson_codec_glue
polyval_generic        16384  1 polyval_ce
gpu_sched              36864  1 panfrost
snd_compress           24576  1 snd_soc_core
rc_odroid              16384  0
meson_ir               16384  0
meson_vdec             81920  0
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
rc_core                57344  4 ir_nec_decoder,meson_ir,rc_odroid
videobuf2_dma_contig    24576  1 meson_vdec
snd_pcm               131072  8 snd_soc_hdmi_codec,snd_compress,snd_soc_meson_axg_tdm_interface,snd_soc_core,snd_soc_meson_axg_fifo,snd_soc_meson_axg_toddr,snd_soc_meson_codec_glue,snd_pcm_dmaengine
meson_gxbb_wdt         16384  0
v4l2_mem2mem           36864  1 meson_vdec
videobuf2_memops       20480  1 videobuf2_dma_contig
videobuf2_v4l2         32768  2 meson_vdec,v4l2_mem2mem
meson_saradc           24576  0
meson_rng              16384  0
videobuf2_common       57344  5 meson_vdec,videobuf2_dma_contig,videobuf2_v4l2,v4l2_mem2mem,videobuf2_memops
snd_timer              40960  1 snd_pcm
videodev              241664  4 meson_vdec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
dw_hdmi_i2s_audio      16384  0
mc                     65536  4 videodev,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
simpledrm              24576  0
snd                    98304  5 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
nvmem_meson_efuse      16384  0
drm_shmem_helper       24576  2 panfrost,simpledrm
gpio_fan               16384  0
efi_pstore             16384  0
dm_mod                143360  0
ip_tables              32768  0
x_tables               45056  1 ip_tables
autofs4                45056  2
meson_gxl              20480  0
realtek                32768  0
dwmac_generic          16384  0
axg_audio              77824  49
reset_meson_audio_arb    16384  6
rtc_pcf8563            24576  0
sclk_div               16384  1 axg_audio
rtc_meson_vrtc         20480  1
clk_phase              16384  1 axg_audio
dwmac_meson8b          16384  0
```
In my case it is before the bulk of audio and video drivers, in your case more in the middle of them. I rebooted one time and it is still pretty much the same.

I tried to find some revision code, but couldn't find any. Is it a 2 GiB or 4 GiB RAM variant? And do you boot from SD card or eMMC module, or via SPI bootloader from USB?

Could you check for kernel errors?
```sh
dmesg -l 0,1,2,3
```

Zer0x00, 2023-07-15:
It's the 4 GiB RAM variant and I'm booting from the eMMC module.

```bash
root@pi:~# dmesg -l 0,1,2,3
[    2.810394] genirq: Setting trigger mode 3 for irq 23 failed (meson_gpio_irq_set_type+0x0/0x68)
[    4.721090] panfrost ffe40000.gpu: error -ENODEV: _opp_set_regulators: no regulator (mali) found
```

MichaIng, 2023-07-15:
All exactly the same as here, 4 GiB, eMMC and:
```console
root@OdroidN2:~# dmesg -l 0,1,2,3
[    2.812104] genirq: Setting trigger mode 3 for irq 23 failed (meson_gpio_irq_set_type+0x0/0x68)
[    5.648008] panfrost ffe40000.gpu: error -ENODEV: _opp_set_regulators: no regulator (mali) found
```
I just tested to remove the module and restart the service, and it fails the same way, hence it is not somehow loaded on demand but just automatically, and must have been loaded already when the service starts. Given that the service has absolutely not ordering or additional dependencies, it seems to be expected that such HW random number generator drivers do load automatically at sufficiently early boot stage.

Does it change something when you explicitly force the module to be loaded?
```sh
echo 'meson_rng' > /etc/modules-load.d/hwrng.conf
```

Zer0x00, 2023-07-15:
Yes, explicitly forcing the module to be loaded moves the loading order to be before the Audio/Video drivers:

```bash
root@pi:~# lsmod
Module                  Size  Used by
snd_soc_hdmi_codec     24576  1
panfrost               69632  0
snd_soc_meson_g12a_tohdmitx    20480  1
snd_soc_meson_g12a_toacodec    20480  1
polyval_ce             16384  0
meson_vdec             81920  0
snd_soc_meson_axg_sound_card    16384  0
gpu_sched              36864  1 panfrost
snd_soc_meson_codec_glue    16384  2 snd_soc_meson_g12a_toacodec,snd_soc_meson_g12a_tohdmitx
snd_soc_meson_card_utils    16384  1 snd_soc_meson_axg_sound_card
videobuf2_dma_contig    24576  1 meson_vdec
ir_nec_decoder         20480  0
polyval_generic        16384  1 polyval_ce
snd_soc_meson_axg_toddr    28672  3
v4l2_mem2mem           36864  1 meson_vdec
videobuf2_memops       20480  1 videobuf2_dma_contig
snd_soc_meson_axg_frddr    36864  3
snd_soc_meson_axg_tdmout    24576  2
snd_soc_meson_axg_tdm_interface    16384  3 snd_soc_meson_axg_sound_card
snd_soc_meson_t9015    24576  1
snd_soc_meson_axg_tdmin    24576  4
dw_hdmi_i2s_audio      16384  0
snd_soc_simple_amplifier    16384  1
meson_gxbb_wdt         16384  0
snd_soc_meson_axg_fifo    20480  2 snd_soc_meson_axg_toddr,snd_soc_meson_axg_frddr
rc_odroid              16384  0
snd_soc_meson_axg_tdm_formatter    16384  3 snd_soc_meson_axg_tdmin,snd_soc_meson_axg_tdm_interface,snd_soc_meson_axg_tdmout
videobuf2_v4l2         32768  2 meson_vdec,v4l2_mem2mem
meson_ir               16384  0
rc_core                57344  4 ir_nec_decoder,meson_ir,rc_odroid
snd_soc_core          245760  15 snd_soc_meson_axg_tdm_formatter,snd_soc_meson_t9015,snd_soc_meson_g12a_toacodec,snd_soc_meson_axg_sound_card,snd_soc_hdmi_codec,snd_soc_meson_g12a_tohdmitx,snd_soc_meson_axg_tdmin,snd_soc_simple_amplifier,snd_soc_meson_axg_tdm_interface,snd_soc_meson_axg_fifo,snd_soc_meson_axg_tdmout,snd_soc_meson_axg_toddr,snd_soc_meson_axg_frddr,snd_soc_meson_card_utils,snd_soc_meson_codec_glue
videobuf2_common       57344  5 meson_vdec,videobuf2_dma_contig,videobuf2_v4l2,v4l2_mem2mem,videobuf2_memops
videodev              241664  4 meson_vdec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
snd_compress           24576  1 snd_soc_core
mc                     65536  4 videodev,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
meson_saradc           24576  0
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               131072  8 snd_soc_hdmi_codec,snd_compress,snd_soc_meson_axg_tdm_interface,snd_soc_core,snd_soc_meson_axg_fifo,snd_soc_meson_axg_toddr,snd_soc_meson_codec_glue,snd_pcm_dmaengine
simpledrm              24576  0
drm_shmem_helper       24576  2 panfrost,simpledrm
nvmem_meson_efuse      16384  0
gpio_fan               16384  0
snd_timer              40960  1 snd_pcm
snd                    98304  5 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
meson_rng              16384  0
dm_mod                143360  0
efi_pstore             16384  0
ip_tables              32768  0
x_tables               45056  1 ip_tables
autofs4                45056  3
meson_gxl              20480  0
realtek                32768  1
dwmac_generic          16384  0
axg_audio              77824  49
rtc_pcf8563            24576  0
sclk_div               16384  1 axg_audio
clk_phase              16384  1 axg_audio
rtc_meson_vrtc         20480  1
reset_meson_audio_arb    16384  6
dwmac_meson8b          16384  0
```

..and the service just starts fine:
```bash
root@pi:~# service rngd status
● rngd.service - Start entropy gathering daemon (rngd)
     Loaded: loaded (/lib/systemd/system/rngd.service; enabled; preset: enabled)
     Active: active (running) since Sun 2023-07-16 00:18:22 CEST; 13s ago
       Docs: man:rngd(8)
   Main PID: 924 (rngd)
      Tasks: 1 (limit: 4213)
     Memory: 212.0K
        CPU: 3ms
     CGroup: /system.slice/rngd.service
             └─924 /usr/sbin/rngd -f

Jul 16 00:18:22 pi systemd[1]: Started rngd.service - Start entropy gathering daemon (rngd).
```

MichaIng, 2023-07-15:
Okay great. Of course delaying the rngd with a sleep order some ordering is also possible, but it could result in other services, which wait for the entropy pool to be filled, to be delayed as well. So as this module is quite mandatory anyway, forcing it to be loaded does not hurt and may even speed other things a little up: It is loaded earlier in your case now that it was in mine.

I'll add this config to our future Odroid N2 images and to all running systems with next DietPi update. Even if the case may be rare, it does not hurt, and a working entropy daemon is essential.

Zer0x00, 2023-07-15:
I wasn't looking for any better solutions for me but to improve the experience for other dietpi users.
But indeed your solution seems to speed up the boot time.

This may not be statistically accurate but ~3 reboot cycles with each setting confirms that the startup time changed from round about 7 seconds down to 5.5 seconds:

Before:
```
Jul 16 00:27:31 pi systemd[1]: Startup finished in 3.587s (kernel) + 3.458s (userspace) = 7.045s.
Jul 16 00:29:56 pi systemd[1]: Startup finished in 3.543s (kernel) + 3.473s (userspace) = 7.016s.
Jul 16 00:31:12 pi systemd[1]: Startup finished in 3.561s (kernel) + 3.524s (userspace) = 7.085s.
```

After:
```
Jul 16 00:32:49 pi systemd[1]: Startup finished in 3.546s (kernel) + 1.871s (userspace) = 5.417s.
Jul 16 00:33:26 pi systemd[1]: Startup finished in 3.566s (kernel) + 1.813s (userspace) = 5.379s.
Jul 16 00:34:00 pi systemd[1]: Startup finished in 3.586s (kernel) + 1.906s (userspace) = 5.493s.
```

Thanks for the fast & great help, keep up the good work! 🥇

MichaIng, 2023-07-15:
Great. I wasn't even expecting such a significant change. Will check on my N2+ as well whether it makes a difference there.

MichaIng, 2023-07-27:
Fixed with: https://github.com/MichaIng/DietPi/commit/8942203

closed 2023-07-29 by MichaIng