MichaIng/DietPi #6484: Can't access web interfaces for bitwarden, mineos through remote.it (or really anything)

cyberqueenmeg, 2023-07-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'

- Distro version | bookworm 0

- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | RPi 4 Model B (aarch64)

- Power supply used | default Canakit power supply
- SD card used | amazon basics 64 gig

#### Additional Information (if applicable)
- Software title | remote.it, bitwarden, mineos
- Was the software title installed freshly or updated/migrated?: yes
- Can this issue be replicated on a fresh installation of DietPi?: yes
- are these services running according to the OS when checked? YES

#### Steps to reproduce
Install software and try to navigate to localhost:8001 (vaultwarden) or localhost:8443 (mineos) (or use a remote.it url)
Get Service Offline error

#### Expected behaviour
I should be able to navigate to the vaultwarden/mineos interface

#### Actual behaviour
I get a service offline error

#### Extra details
I can access the dietpi dashboard through remote.it, but can't interact with it.
If I access it through my LAN, I can interact with it.

MichaIng, 2023-07-16:
> Install software and try to navigate to localhost:8001 (vaultwarden) or localhost:8443 (mineos)

Do you really try to access them via `localhost` in browser address bar form a different system? This hostname always points to the local system itself, while form any other system you must use the LAN IP address of the server. Additionally, both services are available via `https://` only, and at least the auto-generated certificate of `vaultwarden` is generated with a SAN list which does not contain `localhost`, but only the local systems actual hostname and LAN IP. So accessing it via `localhost` would at least throw a browser warning, or won't work at all.

So please try again using `https://<IPaddress>:<port>` to access MineOS and vaultwarden.

cyberqueenmeg, 2023-07-16:
@MichaIng - I had tried the IP itself earlier and it didn't work. I was using localhost for remote.it (what it calls for). I tried again doing https://IP:8001 and it worked. Thank you!

closed 2023-07-16 by cyberqueenmeg