MichaIng/DietPi #6486: Documentation: Ansible bootstrapping

port19x, 2023-07-17:
To get ansible working on my 5 NanoPi Fire3 Nodes I used the following steps:

1. switch ssh server from dropbear to openssh
2. install python3
3. `ansible all -m ping`

Now ansible works. I have only tested ad-hoc commands thus far.

Closing, as this is intended to be found through search. Will update if I encounter further issues that need workarounds on dietpi

closed 2023-07-17 by port19x

MichaIng, 2023-07-17:
Thanks for sharing this info. Could you find out why OpenSSH is required? Does Ansible rely of some PAM features not supported by Dropbear, or SFTP?

port19x, 2023-07-18:
> Thanks for sharing this info. Could you find out why OpenSSH is required? Does Ansible rely of some PAM features not supported by Dropbear, or SFTP?

I believe it relies on sftp / scp. I got warnings about connection via those methods failing while the nodes ran dropbear. Changing to openssh aleviated that issue