MichaIng/DietPi #6488: Clean Installation on new Raspberry Pi 3Model B

EinfachAleks, 2023-07-17:
#### Details:
- Date           | Mon Jul 17 21:51:01 BST 2023
- Bug report     | d00e18be-665e-48de-a0e3-60de6c22c96d
- DietPi version | v8.19.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install ffmpeg`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ffmpeg : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)
          Depends: libavfilter7 (= 8:4.3.6-0+deb11u1+rpt2)
 libavdevice58 : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)
                 Depends: libavfilter7 (= 8:4.3.6-0+deb11u1+rpt2)
 libavformat58 : Depends: libavcodec58 (= 8:4.3.6-0+deb11u1+rpt2)
                 Depends: libsrt1.4-gnutls (>= 1.4.2) but it is not installable
E: Unable to correct problems, you have held broken packages.

MichaIng, 2023-07-17:
Please check the solution here: #6461

closed 2023-07-17 by MichaIng

EinfachAleks, 2023-07-17:
@MichaIng thanks