MichaIng/DietPi #6489: libssl1.1

Dylantje, 2023-07-18:
Hope there is a option.. for install libssl1.1

Drive me cray.

I run: Odroid N2 with Armbian.

___ _ _ _ _ _ ____
/ _ \ __| |_ __ ___ (_) __| | | \ | |___ \
| | | |/ _` | '__/ _ \| |/ _` | | \| | __) |
| |_| | (_| | | | (_) | | (_| | | |\ |/ __/
\___/ \__,_|_| \___/|_|\__,_| |_| \_|_____|

Welcome to Armbian 23.05.1 Bookworm with Linux 6.1.30-meson64

System load: 2% Up time: 31 min
Memory usage: 9% of 3.69G IP: 192.168.5.70
CPU temp: 34°C Usage of /: 9% of 57G
RX today: 499.3 MiB

For  Domoticz, in need, libssl1.1

Still get an error when run:

pi@Domoticz:~$ cd domoticz
pi@Domoticz:~/domoticz$ ./domoticz
./domoticz: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
pi@Domoticz:~/domoticz$

Where i google and try to find solutions.. No option to install libssl1.1 on my ondroid. [ also not with jammy ]
I find here some solutions, only they are not working,
[https://github.com/domoticz/domoticz/issues/5233](url)
[https://www.domoticz.com/forum/viewtopic.php?t=40008](url)

No where a solution? Find something that i need to run Domoticz in docker,.... after more then 10 years running on armbian..:-(

Hope there is a other solution.

Help please......

THANKS!!!

Joulinar, 2023-07-18:
First question, as you are running Armbian, why you are posting your question on DietPi GitHub 🤔 

`libssl1.1` package is available on Debian Buster and Bullseye only. On Debian Bookworm package would be `libssl3`. Therefore, a software that requires `libssl1.1` is not able to run on Bookworm. At least from our side we disable the possibility to install Domoticz on Debian Bookworm.

Dylantje, 2023-07-18:
sorry wrong topic..

Thanks for the answer anyway!

closed 2023-07-18 by Joulinar

MichaIng, 2023-07-18:
Referencing the report at the Domoticz repo: https://github.com/domoticz/domoticz/issues/5233

Best solution IMO is to compile Domoticz from source against the new LibSSL instead of using the pre-compiled binary. I will probably create a build script and provide packages from our server which could then be used on Armbian as well.

MichaIng, 2024-01-22:
Solved with latest builds, which were compiled against libssl3. However, this means that it won't work on Buster and Bullseye anymore, i.e. we need to pull an earlier version there 😄.