MichaIng/DietPi #6497: Open Beta v8.20 | Please help testing and hardening our upcoming release

MichaIng, 2023-07-23:
RC version | v8.20.0
----------- | -------
v8.20.0 PR | #6496
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2023-07-29**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.20
____
### Beta v8.20.0
_(2023-07-23)_

#### New software
- Homebridge | Bringing HomeKit support where there is none. Many thanks to @Zer0x00 for implementing it: https://github.com/MichaIng/DietPi/pull/6493

#### Enhancements
- DietPi-LetsEncrypt | Updated the Lighttpd SSL config syntax and options according to latest Mozilla SSL config generator recommendation with intermediate client compatibility. Many thanks to @JappeHallunken for implementing this enhancement: https://github.com/MichaIng/DietPi/pull/6481
- DietPi-Software | WiFi Hotspot: The default DHCP server settings have been cleaned up and enhanced, with the default lease time increased from 10 minutes to 12 hours, the max lease time increased from 2 hours to 1 day, and the IP range extended up to 192.168.42.250.
- DietPi-Software | Apache/ownCloud/Pi-hole: The X-XSS-Protection header is now set to "0" in default configs to match recent security recommendations. This change is also applied to all systems on next DietPi update. Many thanks to @Zer0x00 for implementing this enhancement: https://github.com/MichaIng/DietPi/pull/6491

#### Bug fixes
- Raspberry Pi | Resolved an issue on Bookworm systems where FFmpeg and related A/V libraries and development headers could not be installed, since the raised epoch version of those from the Raspberry Pi repository is leading to conflicts with the newer ones from the Debian Bookworm repository.
- DietPi-LetsEncrypt | Resolved a DietPi v8.19 regression where applying the HTTPS certificate for Lighttpd fails. Many thanks to @midniteca for reporting this issue: https://github.com/MichaIng/DietPi/issues/6460
- DietPi-Config | Resolved an issue where enabling Bluetooth on SBCs with Armbian firmware failed, because of a conflict between armbian-firmware and bluez-firmware packages. bluez-firmware will not be tried to be installed anymore if armbian-firmware is. Many thanks to `@innovodev` for reporting this issue: https://dietpi.com/forum/t/upgrading-dietpi-from-bullseye-to-bookworm/15963/16
- DietPi-Software | Lighttpd: Resolved a DietPi v8.19 regression where the installation of Pi-hole, ownCloud, Nextcloud, Pydio and WikiMedia failed if Lighttpd was selected as webserver and HTTPS not yet enabled via dietpi-letsencrypt. Many thanks to @bruno-briner for reporting this issue: https://github.com/MichaIng/DietPi/issues/6455
- DietPi-Software | PaperMC: Resolved an issue where the install failed when selecting the Geyser and Floodgate plugins, as the download URL changed. Many thanks to @NatureHog for reporting and fixing this issue: https://github.com/MichaIng/DietPi/issues/6471

Zer0x00, 2023-07-26:
Would it be possible to add the fix for https://github.com/MichaIng/DietPi/issues/6483 also to the 8.20 release?

MichaIng, 2023-07-27:
Jep, just implemented the fix: https://github.com/MichaIng/DietPi/commit/8942203

MichaIng, 2023-07-29:
#### Beta v8.20.1 has been merged: #6499

MichaIng, 2023-07-29:
#### DietPi v8.20 has been released: #6498
Many thanks to all testers ❤️ !

closed 2023-07-29 by MichaIng

hazaops, 2023-07-30:
Hey guys! I am very sorry to intrude but, I did not intend to update to the beta, just realized today I was on it, so far after reading I was able to change the branch back to master but I have no idea how to go back to 8.19 for stability. Can someone enlighten me please? Thanks in advance!

Joulinar, 2023-07-30:
There is no way back. We don't host different versions due to the nature how our updates are working. Feel free to open an own new issue if you are facing issues.

MichaIng, 2023-07-30:
DietPi v8.20 has been released, so no stability problems. Furthermore, this release did not contain any intrusive core code changes, but mostly bug fixes and security updates. So even the first beta could be seen as more stable (less bugs) than v8.19 🙂.