MichaIng/DietPi #6500: Bookworm | Poweroff/reboot returns error message but does the job anyway.

twikedk, 2023-07-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.19.1
- Distro version | bookworm
- Kernel version | 5.15.93-sunxi64
- SBC model | PINE A64 (aarch64)
- Power supply used | 5V 3A
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
It is a fresh installed bookworm dietpi on PINE A64

#### Steps to reproduce

1. login with ssh
2. call for a poweroff or reboot

It does reboot or shutdown and I'm not sure if the error is just due to being ssh connected.

#### Extra details

attached screenshot of the error message
[image]

**Side Note**
I also did a fresh install on a Raspberry Pi 3 with image 'DietPi_RPi-ARMv7-Bookworm' where I end with the exact same error message and yes I know it is a v7 image but it is meant for a Raspberry Pi 2 v1.1

MichaIng, 2023-07-25:
Thanks for reporting. Indeed a known issue on Bookworm which I did not find a satisfying solution so far:
- The shutdown/reboot/poweroff commands (actually all `systemctl`) try to use `dbus` to communicate with `systemd-logind`.
- This was previously only the case with an explicit scheduled shutdown, like `shutdown +5` to shutdown after 5 minutes. This is managed by `systemd-logind`.
- Since Bookworm (or the respective newer systemd version) the attempt to contact dbus is somehow done even for immediate shutdowns/reboots, even that it is not required. Not sure what this is for, probably some sort of session de-registration. As long as `systemd-logind` was not used for session registration in the first place, this is however also not required.

DietPi ships without `systemd-logind` enabled and hence also without `dbus` installed. It is required for desktop sessions, and a few other applications use it, like Kodi and Chromium, and those which use the systemd PAM modules. But it is then installed on demand to not run two additional services which remain completely unused in common server environments.

MichaIng, 2023-07-25:
Sent to Debian bug tracker: https://bugs.debian.org/1042015
Most likely this is an upstream issue, but probably someone from the Debian community has some more insights, can verify that it is indeed an upstream issue and, if we are lucky, even an idea how to patch it. Otherwise we can report at the systemd repo: https://github.com/systemd/systemd/issues

MichaIng, 2023-08-20:
Reported to `systemd` directly: https://github.com/systemd/systemd/issues/28902

MichaIng, 2023-09-19:
Solved via shell functions defined in `dietpi-globals`: https://github.com/MichaIng/DietPi/commit/e591aeb

Not perfect, also does not cover `systemctl poweroff/reboot/halt` or `shutdown now`, but best I could think of for now.

Btw, does someone know the `--no-wtmp`/`--wtmp-only` options? https://manpages.debian.org/poweroff
- `/var/log/wtmp` contains logs about boot and user sessions, which can be checked via `last` command.
- Of course it is boot persistent and hence really usable only with `dietpi-ramlog` uninstalled.
- However, `--wtmp-only` makes the `poweroff/...` commands no-ops for me, no `wtmp` entry is created, and of course, as intended, no actual poweroff/reboot/halt done.
- Generally, for me `wtmp` contains only one log for the current boot session, which is created at boot, not at shutdown, and one for each user session, also created when the session starts.
- The poweroff/reboot/halt does not add any additional log in general, hence `--no-wtmp` has also no effect. On next boot, of course you see the last boot session as finished, but this is always the case, same as with closed user sessions.
- All tested with dbus and logind running, of course.