MichaIng/DietPi #6501: Diet PI not booting in an Orange Pi 5 Plus via Nvme + SPI Flash

yandritos, 2023-07-25:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`  **--> No way to execute but last ISO released**
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`  **--> No way to execute but last ISO released**
- Kernel version | `uname -a` **--> No way to execute but last ISO released**
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)  **--> No way to execute but Orange PI 5 Plus**
- Power supply used | (EG: 5V 1A RAVpower)  **--> 5v  3 A 33w**
- SD card used | (EG: SanDisk ultra)  **--> No SD card, SPI Flash for boot from PCIE Nvme Crucial P3 1TB**

#### Additional Information (if applicable)
Other ISOs provided at the Orange PI site are working properly booting without problems from the Nvme.

MichaIng, 2023-07-25:
Many thanks for reporting. Discussion about this is here: https://dietpi.com/forum/t/orange-pi-5-boot-from-nvme/16457
Workaround is to update the SPI bootloader either from the official Orange Pi image via `orangepi-config` or from a recent Armbian image via `armbian-config`.

Armbian has not updated their APT repository since February, so latest versions are missing, but their images ship with more recent bootloader and kernel packages which are not available via APT yet. I'll try to extract the bootloader DEB package from there and host it on our server. We can then install it form there on our Orange Pi 5 images.

However, bootstrapping the SPI bootloader is required in any case, as the one shipped with the board seems to not support the full `boot.scr` syntax. So it will be required to flash the image to SD card first, flash the SPI bootloader from there via `dietpi-config`, and only after this, NVMe boot works.

Do you have a UART adapter to check serial console logs for details?

yandritos, 2023-07-25:
Hi Michalng, thanks a lot for the very fast reply and the link.

I have an USB-TTL UART module somewere, I´ll look for it in order to see if I am able to get any console log, I´ll need some time.

Best regards!

MichaIng, 2023-07-25:
Great, that would be interesting. Probably it is similar or the same as with the ROCK 5B: #6415

yandritos, 2023-07-25:
Hi @MichaIng ,

Today I am a little bit lucky, is the first time in years I look for an old small device and I find it (and the best part, it works!).
Please find attached the logs extracted with MobaXterm and the UST-TTL UART device.
[Orange Pi 5 plus logs.txt](https://github.com/MichaIng/DietPi/files/12165781/Orange.Pi.5.plus.logs.txt)

I hope you can find something relevant.

Thanks in advance!

garlem-rid, 2023-07-31:
So I have the same problem. When I had connect opi5(which is no bootable when it is wo sd card, even with a flashed by the dietpi spi ) via flipper usbuart, and after some check I send the 'boot' command to uart, and after that it immediately started to boot from m2ssd

MichaIng, 2023-08-06:
Orange Pi 5 systems will get a kernel and U-Boot update with next DietPi update, which should also enable NVMe support, after flashing the new bootloader binary to SPI via `dietpi-config` > Advanced Settings > Update SPI bootloader: #6532

USB boot works fine afterwards, would be great if some could test it with NVMe. We should then add an entry in our documentation about this and a link from the download page.

New images with this kernel to test are available here: https://dietpi.com/downloads/images/testing/

Another bonus is that this kernel should fully support the Orange Pi 5B as well, i.e. its onboard WiFi.

BUT: While it boots on Orange Pi 5 Plus, the device tree of the non-plus variant causes constant nasty kernel errors spamming the console. And with the non-plus bootloader, the plus device tree is not automatically loaded. So I decided to add dedicated support for Orange Pi 5 Plus with an own hardware ID and own images: #6533

Images are building: https://github.com/MichaIng/DietPi/actions/runs/5779343116/job/15661509102
Will be available here if successful: https://dietpi.com/downloads/images/testing/
But they use the temporary `orangepi5plus` feature branch, which needs to be adjusted once DietPi v8.21 has been released.

garlem-rid, 2023-08-07:
I'll try next weekend

yandritos, 2023-08-08:
Hi, I can try as well, please provide the instructions when possible.

Joulinar, 2023-08-08:
> please provide the instructions when possible.

What are you looking for? Download the new image as usual. See above.

yandritos, 2023-08-08:
Hi @Joulinar maybe my English is so bad, I read this in the last entry from MichaIng:

"_If someone has its OPi 5 Plus already fully setup, the kernel will be upgraded anyway and it is not hard to migrate to the Plus bootloader and device tree. Just let me know and I'll provide instructions_."

Joulinar, 2023-08-08:
Somehow the information was missing that you already have a 5+ running DietPi

yandritos, 2023-08-08:
My bad, I thought I mentioned it. Anyway that is not the case anymore, I assume that if I restart from scrat that can save some time to Michal so I am just rewriting the NVME. I´ll keep you posted.

MichaIng, 2023-08-08:
@yandritos 
To migrate any existing system, first switch to the Orange Pi 5 Plus bootloader:
```sh
dpkg -P linux-u-boot-orangepi5-legacy
curl -fo package.deb 'https://dietpi.com/downloads/binaries/linux-u-boot-orangepi5-plus-legacy.deb'
dpkg -i package.deb
rm package.deb
```
Then switch to the DietPi `dev` branch, which will automatically update the kernel as well (skip the reboot for now!):
```sh
G_DEV_BRANCH dev
```
Then adjust the hardware ID to represent the Plus variant and reboot:
```sh
echo 82 > /etc/.dietpi_hw_model_identifier
reboot
```
Finally you can move back to the `master` branch, so you will be back on stable code once DietPi v8.21 is released:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
```

yandritos, 2023-08-09:
Hi, sorry for the delay. @MichaIng  thanks for the information, finally I tried to make it work from scratch and I formatted the NVME unit, but for some reason now is not working anymore. I am trying to get it back to life, otherwise I´ll be on hold till get a new unit.

Joulinar, 2023-08-09:
Did you tried booting of SD card for testing?

MichaIng, 2023-08-09:
You still have the `U-Boot 2017.09-orangepi` on SPI then? I tested it only with our new bootloader, but if yours was able to boot the old image, it should be able to boot the new one as well. It is a slightly newer kernel, and it does not enforce the (wrong) device tree, but that shouldn't make a difference. What does the serial console say?

yandritos, 2023-08-16:
Hi @Joulinar , not this time, I did that after the first post (I'll try as I am forced by my current spi status) @MichaIng My bad, when I broke my nvme I started from Scratch flashing the 3 points (first is the spi). This week I receive the new nvme. I'll keep you posted.

zappydood, 2023-08-29:
Can I ask what the current status of this is? I installed the new orange pi 5 plus image but there is no update boot loader to SPI or MTD just update booloader to mmc which is there because I have it running off microsd. There are no other options past that and I want to move the image to NVME, I just installed everything and configured everything to my liking and then I stumbled upon issues trying to flash it to my NVME, it won't boot and gives me errors. There are no real current docs about what to do and how to do it and I get why it was just released so early but should I just install to NVME and then resize the partition? I just came from Joshua-rieks ubuntu image but despite being somewhat satisfied with it I miss Diet Pi and want to get it to work on NVME like it can, its just there is no update bootloader to SPI or MTD, just MMC. I am wondering if I am missing a step or something I've been used to installing Armbian with armbian-config and it's just not there so I want to find out if I'm missing a step. Thanks so much for dietpi, I was so happy it came to orange pi 5 plus because maaaan trying to get it to work on the opi 5 image wasn't working too well and I had to throw in the towel for a while with time constraints. I am willing to do any testing as well because I have the orange pi 5 with NVME already installed and have had Armbian and Joshua rieks image installed and running off name without issue. Do I need to update the bootloader anyway after updating it last with [Joshua rieks ubuntu image? ](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5-Plus)

Thank you, any help is appreciated. I think I may wait to hear back before I try installing to NVME again. Hope it's okay if I posted on this issue, too.. for now I at least have dietpi on it again and it runs great! I just wanna get it moved to NVME like it was with the other images, I am a little more new to dietpi than I am with Armbian. Thanks!

Joulinar, 2023-08-29:
did you boot of SD card and tried this one as stated by @MichaIng above?

> Flash this bootloader to MMC and SPI via dietpi-config > Advanced Options > "Update MMC bootloader" and afterwards "Update SPI bootloader" (the second one needed for NVMe and USB boot).

zappydood, 2023-08-29:
yes, I did, but maybe I am confused or something because the "update bootloader to spi/mtd" doesn't appear anywhere, the only available option is "update bootloader to MMC" and then that's all there is, nothing after that as well after selecting it. I've looked long and hard all over trying to find more details pertaining to this board specifically and nvme with and without dietpi in the mix, and this issue request is the only viable solution I can find but I am still slightly confused what piece of information I'm missing because there is no actual option for updating bootloader to spi, just mmc and I feel like I'm going crazy trying to figure out why its missing when I feel like it is supposed to be there. 

I'm running version 8.21.1 and I'm wondering if I need to even update the bootloader since it was running from nvme before trying all this and now I'm just running dietpi from my microsd card because I haven't been able to fully move it to the NVME yet. It does TRY to boot from nvme when I take the microsd card out but I'm trying the write again after creating a NEW disk image my last one weirdly ended up as 

sudo e2fsck -f /dev/nvme0n1p1
e2fsck 1.47.0 (5-Feb-2023)
/dev/nvme0n1p1: recovering journal
ext2fs_check_desc: Corrupt group descriptor: bad block for block bitmap
e2fsck: Group descriptors look bad... trying backup blocks...
e2fsck: unable to set superblock flags on /dev/nvme0n1p1

so I took that as a warning I skipped or messed up a step on resizing the partition and to try again. I don't know why its doing that though, as its a virtually brand new dietpi image running on microsd and the microsd is new, so we will see what the next go at this looks like now that I've verified there is nothing wrong with either the nvme nor the microsd.

ill admit I've never needed to do it this way, so I may be missing a step/messing a step up lol

MichaIng, 2023-08-29:
Ah sorry, my mistake. I added a new hardware ID for the Plus variant but forgot to show the menu option for this ID as well. Please do the following:
```sh
sed -i 's/G_HW_MODEL == 80/G_HW_MODEL == 82/' /boot/dietpi/dietpi-config
```
Fixed with: https://github.com/MichaIng/DietPi/commit/a57a2ae

MichaIng, 2023-09-01:
Great does it boot for you from NVMe? We have the challenge that on OPi 5 non-plus, NVMe boot is broken since the kernel upgrade, so would be good to know whether this only affects non-plus.

dirkhh, 2023-09-01:
I can test the 5plus - when I first played with NVMe this was seriously unstable, but I suspected bad power supply. This is now fixed so I can give this a go.

MichaIng, 2023-09-01:
Okay great. Then we only need to find out what the issue with non-Plus is.

closed 2023-09-01 by MichaIng

dirkhh, 2023-09-02:
sorry for taking so long to respond...
using the current `dev` branch (plus my adsb-feeder commits) I did the following
- `dietpi-config`: Flash current U-Boot to /dev/mtdblock0
- copy a fresh DietPi image onto the SD card and then `dd` it to the NVMe drive
- reboot

So far so good, seems to work like a charm.

MichaIng, 2023-09-03:
Many thanks. So we only need to debug the non-Plus then.

ppsilv, 2023-09-05:
I bought the orange pi 5 plus, and that was my biggest regret lately, because I have about 6 raspberry pi and I never had any problems with them.
The orange pi 5 plus has serious problems with the NVME ssd I tested it with 3 different models and it only started to boot through NVME on the third disk I bought NVME GEN 3, because gen 4 not even thinking about booting, but with all operating systems that I tested, ubuntu jammy, orange pi OS, all the official ones, I also tried with an image created by Riek, they all boot but after some time working it starts to give an input/output error for every command executed in the terminal, I could see that this happens because the disk is gone for the operating system.

ppsilv, 2023-09-05:
> sorry for taking so long to respond... using the current `dev` branch (plus my adsb-feeder commits) I did the following
>
> * `dietpi-config`: Flash current U-Boot to /dev/mtdblock0
> * copy a fresh [...]

Try run a lot of process and verify if it do not freezes, for me after do that it freezes.

MichaIng, 2023-09-05:
Hmm, the Orange Pi 5 supports Gen 3 only, but of course Gen 4 SSDs should be backwards-compatible 🤔. I'm actually also waiting for a Gen 4 SSD to be come available to order, also to test on OPi 5 among other uses. WD_BLACK SN770 to be exact. Which ones did you test?

Serial console logs are always a good start to find out where and in case why it fails.

One SSD which has proven to work is KIOXIA KBG40ZNS128G, with M.2 2230 form factor on the non-Plus model.

dirkhh, 2023-09-05:
I would love to push back gently here.
The Orange Pi5+ works extremely well with DietPi - it's actually one of my build systems that I use in my development workflow. 
BUT: it is extremely picky when it comes to power supplies. And that is especially true with NVMe drives (as those can draw quite a bit of power very rapidly when they go from idle to running out of onboard cache and having to hammer the flash).
So with errors like this, I think it's really important to talk about the power supply that you use. The typical Raspberry Pi power supply will reliably fail and cause lockups when I do parallel builds on the NVMe. But once I switched to a reliable 5V/5A power supply (a brick, not one of those little itsy plug type power supplies that are usually sold for RPis) I have had zero problems even under exceptionally high loads.

I'm not saying this is the cause of your issues. But I have had three different NVMe drives in my 5+ (a WD Blue, a WD Black, and a noname garbage one) and they work reliably here.

MichaIng, 2023-09-07:
It is pretty much possible that not all NVMe SSDs work perfectly reliably. But also PSU impact is a good hint, at least worth to test a good quality PSU that is not mainly a charger but really for powering devices and holding voltage reliably a sufficient current. The official one from Orange Pi is of course a good reference.

@dirkhh 
Which WD Black is it exactly? The SN770 or another PCIe Gen 4, or Gen 3?

dirkhh, 2023-09-07:
SN770

MichaIng, 2023-09-07:
Thanks, what a relieve as this is the favourite I'm awaiting to become available in shops here 🙂.

dirkhh, 2023-09-07:
I wish I had more faith in "same name == same technology". Sadly it's anything but that.
But yeah, Gen4 fallback to Gen3 should work quite reliably.