MichaIng/DietPi #6502: NFS server fails to start - missing dependency

sewe75, 2023-07-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```bash
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=19
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
`bookworm`

- Kernel version | `uname -a`
`Linux soquartz-2 6.3.11 #1 SMP PREEMPT Sat Jul  1 16:03:21 UTC 2023 aarch64 GNU/Linux`

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`Quartz64 (aarch64)`

- Power supply used | (EG: 5V 1A RAVpower)
```
As the module runs on a TuringPi2, it gets power from the attached 400W desktop PSU (Aerocool Cylon)
``` 

- SD card used | (EG: SanDisk ultra)
`32GB eMMC`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
  - NFS server
  - NFS client
(both installed via dietpi-software)

- Was the software title installed freshly or updated/migrated?
`fresh install`

- Can this issue be replicated on a fresh installation of DietPi?
`yes`

- Bug report ID | `echo $G_HW_UUID`
`aedaf433-6368-4349-9201-35f83db32be5`

#### Steps to reproduce

1. Run `dietpi-software`
2. Search for and install `nfs-server` and `nfs-client`

#### Expected behaviour

- NFS server service should start successfully

#### Actual behaviour

- NFS server fails to start due to a missing dependency

#### Extra details

- Install output (partial)
```sh
Setting up nfs-kernel-server (1:2.6.2-4) ...
Created symlink /etc/systemd/system/nfs-client.target.wants/nfs-blkmap.service → /lib/systemd/system/nfs-blkmap.service.
Created symlink /etc/systemd/system/multi-user.target.wants/nfs-server.service → /lib/systemd/system/nfs-server.service.
nfs-mountd.service is a disabled or a static unit, not starting it.
nfsdcld.service is a disabled or a static unit, not starting it.
Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.

Creating config file /etc/exports with new version

Creating config file /etc/default/nfs-kernel-server with new version
A dependency job for nfs-server.service failed. See 'journalctl -xe' for details.
invoke-rc.d: initscript nfs-kernel-server, action "start" failed.
○ nfs-server.service - NFS server and services
     Loaded: loaded (/lib/systemd/system/nfs-server.service; enabled; preset: enabled)
     Active: inactive (dead)

Jul 26 08:18:34 soquartz-1 systemd[1]: Dependency failed for nfs-server.service - NFS server and services.
Jul 26 08:18:34 soquartz-1 systemd[1]: nfs-server.service: Job nfs-server.service/start failed with result 'dependency'.
Jul 26 08:18:36 soquartz-1 systemd[1]: Dependency failed for nfs-server.service - NFS server and services.
Jul 26 08:18:36 soquartz-1 systemd[1]: nfs-server.service: Job nfs-server.service/start failed with result 'dependency'.
Failed to start nfs-kernel-server, ignoring.
[  OK  ] DietPi-Software | APT install nfs-kernel-server
[  OK  ] DietPi-Software | systemctl stop nfs-kernel-server
[  OK  ] DietPi-Software | mkdir /etc/exports.d

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Finalising install

[ INFO ] DietPi-Software | Enabling installed services
[  OK  ] DietPi-Software | systemctl --no-reload enable nfs-kernel-server
[  OK  ] DietPi-Software | systemctl daemon-reload
2023-07-26 08:18:40 [ INFO ] DietPi-RAMlog | Storing /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store...
2023-07-26 08:18:40 [  OK  ] DietPi-RAMlog | Stored /var/log to /var/tmp/dietpi/logs/dietpi-ramlog_store.

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Install completed

[  OK  ] DietPi-Survey | Sending survey data
[ SUB1 ] DietPi-Services > restart
[FAILED] DietPi-Services | restart : nfs-kernel-server
[  OK  ] DietPi-Services | restart : cron
```

- Journal output (partial)
```bash
Jul 26 08:18:34 soquartz-1 mount[2294]: mount: /proc/fs/nfsd: unknown filesystem type 'nfsd'.
Jul 26 08:18:34 soquartz-1 mount[2294]:        dmesg(1) may have more information after failed mount system call.
Jul 26 08:18:34 soquartz-1 systemd[1]: proc-fs-nfsd.mount: Mount process exited, code=exited, status=32/n/a
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An n/a= process belonging to unit proc-fs-nfsd.mount has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 32.
Jul 26 08:18:34 soquartz-1 systemd[1]: proc-fs-nfsd.mount: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit proc-fs-nfsd.mount has entered the 'failed' state with result 'exit-code'.
Jul 26 08:18:34 soquartz-1 systemd[1]: Failed to mount proc-fs-nfsd.mount - NFSD configuration filesystem.
░░ Subject: A start job for unit proc-fs-nfsd.mount has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit proc-fs-nfsd.mount has finished with a failure.
░░ 
░░ The job identifier is 192 and the job result is failed.
Jul 26 08:18:34 soquartz-1 systemd[1]: Dependency failed for nfsdcld.service - NFSv4 Client Tracking Daemon.
░░ Subject: A start job for unit nfsdcld.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit nfsdcld.service has finished with a failure.
░░ 
░░ The job identifier is 204 and the job result is dependency.
Jul 26 08:18:34 soquartz-1 systemd[1]: nfsdcld.service: Job nfsdcld.service/start failed with result 'dependency'.
Jul 26 08:18:34 soquartz-1 rpc.statd[2300]: Version 2.6.2 starting
Jul 26 08:18:34 soquartz-1 systemd[1]: Dependency failed for nfs-server.service - NFS server and services.
░░ Subject: A start job for unit nfs-server.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit nfs-server.service has finished with a failure.
░░ 
░░ The job identifier is 189 and the job result is dependency.
Jul 26 08:18:34 soquartz-1 rpc.statd[2300]: Flags: TI-RPC
Jul 26 08:18:34 soquartz-1 systemd[1]: Dependency failed for nfs-idmapd.service - NFSv4 ID-name mapping service.
░░ Subject: A start job for unit nfs-idmapd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit nfs-idmapd.service has finished with a failure.
░░ 
░░ The job identifier is 212 and the job result is dependency.
```

Joulinar, 2023-07-26:
can you try following for testing. Reboot your system and wait for 10 Minutes. Afterwards try to restart NFS service

```
systemctl restart nfs-kernel-server
```

sewe75, 2023-07-26:
Hi @Joulinar - Thanks for your reply!  
Here's the output of `lsmod` - will follow your instructions in a minute.  

```bash
root@soquartz-1:~# lsmod
Module                  Size  Used by
btsdio                 16384  0
bluetooth             618496  1 btsdio
brcmfmac_wcc           16384  0
brcmfmac              348160  1 brcmfmac_wcc
crct10dif_ce           16384  1
cfg80211              929792  1 brcmfmac
brcmutil               24576  1 brcmfmac
display_connector      16384  0
hantro_vpu            139264  0
v4l2_vp9               24576  1 hantro_vpu
videobuf2_dma_contig    20480  1 hantro_vpu
videobuf2_memops       16384  1 videobuf2_dma_contig
v4l2_h264              16384  1 hantro_vpu
v4l2_mem2mem           40960  1 hantro_vpu
videobuf2_v4l2         32768  2 hantro_vpu,v4l2_mem2mem
videobuf2_common       65536  5 videobuf2_dma_contig,videobuf2_v4l2,hantro_vpu,v4l2_mem2mem,videobuf2_memops
videodev              266240  3 videobuf2_v4l2,hantro_vpu,v4l2_mem2mem
mc                     57344  5 videodev,videobuf2_v4l2,hantro_vpu,videobuf2_common,v4l2_mem2mem
loop                   32768  0
dm_mod                139264  0
ip_tables              28672  0
x_tables               40960  1 ip_tables
```

MichaIng, 2023-07-26:
Can you check for the module:
```sh
modinfo nfsd
modprobe nfsd
```

sewe75, 2023-07-26:
Good point - the module is missing
```bash
modinfo: ERROR: Module nfsd not found.
modprobe: FATAL: Module nfsd not found in directory /lib/modules/6.3.11
```

Joulinar, 2023-07-26:
@MichaIng could it be that the module is missing on `soquartz` kernel?

MichaIng, 2023-07-26:
Seems so. ... Ahh, the kernel config is not set at all, it never was for your Quartz64 builds. That neither me nor anyone else ever recognised this 😄. I don't think it ever was enabled by default. Here we go: https://github.com/MichaIng/DietPi/commit/b54ba04

New package builds are running: https://github.com/MichaIng/DietPi/actions/runs/5670021036
The kernel will be updated automatically on next DietPi update this Saturday night.

closed 2023-07-26 by sewe75

sewe75, 2023-07-30:
Just ran `dietpi-upgrade` -> 8.20.1 and can confirm the NFS server works now.  
[image]
Disclaimer: The logo is just the terminal background - I am not related to them in any way.

MichaIng, 2023-07-30:
Many thanks for testing and feedback 🙂.