MichaIng/DietPi #6503: xrdp black screen when connecting

vinhtq115, 2023-07-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.19.1`
- Distro version | `bullseye 0`
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | 5V 2A
- SD card used | 32GB

#### Additional Information (if applicable)
- Software title | xrdp + xfce
- Was the software title installed freshly or updated/migrated? | Installed a while ago. I removed and reinstalled too.
- Can this issue be replicated on a fresh installation of DietPi? | Don't know for sure.

- Bug report ID | 794c84e0-86ef-4b7e-a599-edd8a637e3c4

#### Steps to reproduce

1. Install xfce and xrdp with `dietpi-launcher`
2. Use Microsoft Remote Desktop to connect

#### Expected behaviour

- XFCE desktop environment should show up

#### Actual behaviour

- Black screen (or aqua one, depends on how I login).
- No error showed up

#### Extra details

- A year or two years ago, xrdp and xfce worked fine. But I haven't used them since then. Now they don't work anymore.

xrdp and xrdp-sesman's systemctl status:
```
dietpi@DietPi:~$ sudo systemctl status xrdp-sesman.service 
● xrdp-sesman.service - xrdp session manager
     Loaded: loaded (/lib/systemd/system/xrdp-sesman.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-07-27 00:25:37 +07; 5min ago
       Docs: man:xrdp-sesman(8)
             man:sesman.ini(5)
    Process: 2655 ExecStart=/usr/sbin/xrdp-sesman $SESMAN_OPTIONS (code=exited, status=0/SUCCESS)
   Main PID: 2656 (xrdp-sesman)
      Tasks: 1 (limit: 3723)
        CPU: 256ms
     CGroup: /system.slice/xrdp-sesman.service
             └─2656 /usr/sbin/xrdp-sesman

Jul 27 00:28:43 DietPi xrdp-sesman[2656]: [INFO ] ++ created session (access granted): username dietpi, ip ::ffff:192.168.1.20:63764 - socket: 12
Jul 27 00:28:43 DietPi xrdp-sesman[2656]: [INFO ] starting Xorg session...
Jul 27 00:28:43 DietPi xrdp-sesman[2656]: [INFO ] Starting session: session_pid 3319, display :10.0, width 1920, height 1080, bpp 24, client ip ::ffff:192.168.1.20:63764 - socket: 12, user>
Jul 27 00:28:43 DietPi xrdp-sesman[3319]: [INFO ] [session start] (display 10): calling auth_start_session from pid 3319
Jul 27 00:28:43 DietPi xrdp-sesman[2656]: [ERROR] sesman_data_in: scp_process_msg failed
Jul 27 00:28:43 DietPi xrdp-sesman[3319]: pam_unix(xrdp-sesman:session): session opened for user dietpi(uid=1000) by (uid=0)
Jul 27 00:28:43 DietPi xrdp-sesman[2656]: [ERROR] sesman_main_loop: trans_check_wait_objs failed, removing trans
Jul 27 00:28:43 DietPi xrdp-sesman[3319]: [INFO ] Found X server running at /tmp/.X11-unix/X10
Jul 27 00:28:43 DietPi xrdp-sesman[3319]: [INFO ] Session started successfully for user dietpi on display 10
Jul 27 00:28:43 DietPi xrdp-sesman[3319]: [INFO ] Session in progress on display 10, waiting until the window manager (pid 3320) exits to end the session
dietpi@DietPi:~$ sudo systemctl status xrdp
● xrdp.service - xrdp daemon
     Loaded: loaded (/lib/systemd/system/xrdp.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-07-27 00:25:38 +07; 5min ago
       Docs: man:xrdp(8)
             man:xrdp.ini(5)
    Process: 2657 ExecStartPre=/bin/sh /usr/share/xrdp/socksetup (code=exited, status=0/SUCCESS)
    Process: 2665 ExecStart=/usr/sbin/xrdp $XRDP_OPTIONS (code=exited, status=0/SUCCESS)
   Main PID: 2666 (xrdp)
      Tasks: 2 (limit: 3723)
        CPU: 325ms
     CGroup: /system.slice/xrdp.service
             ├─2666 /usr/sbin/xrdp
             └─3316 /usr/sbin/xrdp

Jul 27 00:28:42 DietPi xrdp[3316]: [WARN ] local keymap file for 0x00000409 found and doesn't match built in keymap, using local keymap file
Jul 27 00:28:42 DietPi xrdp[3316]: [INFO ] connecting to sesman ip 127.0.0.1 port 3350
Jul 27 00:28:42 DietPi xrdp[3316]: [INFO ] xrdp_wm_log_msg: sesman connect ok
Jul 27 00:28:42 DietPi xrdp[3316]: [INFO ] sesman connect ok
Jul 27 00:28:42 DietPi xrdp[3316]: [INFO ] sending login info to session manager, please wait...
Jul 27 00:28:43 DietPi xrdp[3316]: [INFO ] xrdp_wm_log_msg: login successful for display 10
Jul 27 00:28:43 DietPi xrdp[3316]: [INFO ] login successful for display 10
Jul 27 00:28:43 DietPi xrdp[3316]: [INFO ] loaded module 'libxup.so' ok, interface size 10296, version 4
Jul 27 00:28:43 DietPi xrdp[3316]: [INFO ] started connecting
Jul 27 00:28:43 DietPi xrdp[3316]: [INFO ] lib_mod_connect: connecting via UNIX socket
```

Joulinar, 2023-07-26:
can you try following

```
apt install dbus-x11
```

Are you able to connect now?

MichaIng, 2023-07-26:
Also try this: https://techdirectarchive.com/2022/01/15/how-to-resolve-microsoft-rdp-connection-black-screen/
The persistent bitmap caching helped in another such case.

vinhtq115, 2023-07-27:
> can you try following
> 
> ```
> apt install dbus-x11
> ```
> 
> Are you able to connect now?

`dbus-x11` was already installed so I guess it's not the cause.

```
dbus-x11 is already the newest version (1.12.24-0+deb11u1).
```

> Also try this: https://techdirectarchive.com/2022/01/15/how-to-resolve-microsoft-rdp-connection-black-screen/ The persistent bitmap caching helped in another such case.

RDP for macOS doesn't have this option so I tried on Windows and it's not working either.

vinhtq115, 2023-07-27:
Previously, it seems that I didn't wait long enough so I didn't see this error message.
[image]

[xorgxrdp.10.log](https://github.com/MichaIng/DietPi/files/12178582/xorgxrdp.10.log)

Service logs when I tried connecting:

xrdp:
```
Jul 27 08:36:36 DietPi systemd[1]: Starting xrdp daemon...
Jul 27 08:36:36 DietPi xrdp[50077]: [INFO ] address [0.0.0.0] port [3389] mode 1
Jul 27 08:36:36 DietPi xrdp[50077]: [INFO ] listening to port 3389 on 0.0.0.0
Jul 27 08:36:36 DietPi xrdp[50077]: [INFO ] xrdp_listen_pp done
Jul 27 08:36:36 DietPi systemd[1]: xrdp.service: Can't open PID file /run/xrdp/xrdp.pid (yet?) after start: Operation not permitted
Jul 27 08:36:37 DietPi systemd[1]: Started xrdp daemon.
Jul 27 08:36:38 DietPi xrdp[50079]: [INFO ] starting xrdp with pid 50079
Jul 27 08:36:38 DietPi xrdp[50079]: [INFO ] address [0.0.0.0] port [3389] mode 1
Jul 27 08:36:38 DietPi xrdp[50079]: [INFO ] listening to port 3389 on 0.0.0.0
Jul 27 08:36:38 DietPi xrdp[50079]: [INFO ] xrdp_listen_pp done
Jul 27 08:36:56 DietPi xrdp[50079]: [INFO ] Socket 12: AF_INET6 connection received from ::ffff:192.168.1.20 port 54165
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] Using default X.509 certificate: /etc/xrdp/cert.pem
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] Using default X.509 key file: /etc/xrdp/key.pem
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] Connected client computer name: Vinhs-MacBook-P
Jul 27 08:36:56 DietPi xrdp[50141]: [WARN ] Received [MS-RDPBCGR] TS_UD_HEADER type 0xc006 is unknown (ignored)
Jul 27 08:36:56 DietPi xrdp[50141]: [WARN ] Received [MS-RDPBCGR] TS_UD_HEADER type 0xc00a is unknown (ignored)
Jul 27 08:36:56 DietPi xrdp[50141]: [WARN ] Received [MS-RDPBCGR] TS_UD_HEADER type 0xc00c is unknown (ignored)
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] xrdp_load_keyboard_layout: Keyboard information sent by the RDP client, keyboard_type:[0x04], keyboard_subtype:[0x00], keylayout:[0x00000409]
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] xrdp_load_keyboard_layout: model [] variant [] layout [us] options []
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] TLS connection established from ::ffff:192.168.1.20 port 54165: TLSv1.3 with cipher TLS_AES_256_GCM_SHA384
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] xrdp_caps_process_pointer: client supports new(color) cursor
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] xrdp_process_offscreen_bmpcache: support level 0 cache size 0 MB cache entries 0
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] xrdp_caps_process_codecs: nscodec, codec id 1, properties len 3
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] Loading keymap file /etc/xrdp/km-00000409.ini
Jul 27 08:36:56 DietPi xrdp[50141]: [WARN ] local keymap file for 0x00000409 found and doesn't match built in keymap, using local keymap file
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] connecting to sesman ip 127.0.0.1 port 3350
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] xrdp_wm_log_msg: sesman connect ok
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] sesman connect ok
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] sending login info to session manager, please wait...
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] xrdp_wm_log_msg: login successful for display 10
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] login successful for display 10
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] loaded module 'libxup.so' ok, interface size 10296, version 4
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] started connecting
Jul 27 08:36:56 DietPi xrdp[50141]: [INFO ] lib_mod_connect: connecting via UNIX socket
Jul 27 08:40:26 DietPi xrdp[50141]: [INFO ] connection problem, giving up
Jul 27 08:40:26 DietPi xrdp[50141]: [INFO ] some problem
Jul 27 08:40:41 DietPi xrdp[50141]: [ERROR] xrdp_sec_recv: xrdp_mcs_recv failed
Jul 27 08:40:41 DietPi xrdp[50141]: [ERROR] xrdp_rdp_recv: xrdp_sec_recv failed
Jul 27 08:40:41 DietPi xrdp[50141]: [ERROR] libxrdp_process_data: xrdp_rdp_recv failed
Jul 27 08:40:41 DietPi xrdp[50141]: [ERROR] xrdp_process_data_in: xrdp_process_loop failed
Jul 27 08:40:41 DietPi xrdp[50141]: [ERROR] SSL_shutdown: I/O error
```

xrdp-sesman:
```
Jul 27 08:36:36 DietPi systemd[1]: Starting xrdp session manager...
Jul 27 08:36:36 DietPi xrdp-sesman[50068]: [INFO ] starting xrdp-sesman with pid 50068
Jul 27 08:36:36 DietPi systemd[1]: Started xrdp session manager.
Jul 27 08:36:56 DietPi xrdp-sesman[50068]: [INFO ] Socket 8: AF_INET6 connection received from ::1 port 52126
Jul 27 08:36:56 DietPi xrdp-sesman[50068]: [INFO ] Terminal Server Users group is disabled, allowing authentication
Jul 27 08:36:56 DietPi xrdp-sesman[50068]: [INFO ] ++ created session (access granted): username dietpi, ip ::ffff:192.168.1.20:54165 - socket: 12
Jul 27 08:36:56 DietPi xrdp-sesman[50068]: [INFO ] starting Xorg session...
Jul 27 08:36:56 DietPi xrdp-sesman[50068]: [INFO ] Starting session: session_pid 50145, display :10.0, width 1920, height 1080, bpp 24, client ip ::ffff:192.168.1.20:54165 - socket: 12, us>
Jul 27 08:36:56 DietPi xrdp-sesman[50145]: [INFO ] [session start] (display 10): calling auth_start_session from pid 50145
Jul 27 08:36:56 DietPi xrdp-sesman[50068]: [ERROR] sesman_data_in: scp_process_msg failed
Jul 27 08:36:56 DietPi xrdp-sesman[50145]: pam_unix(xrdp-sesman:session): session opened for user dietpi(uid=1000) by (uid=0)
Jul 27 08:36:56 DietPi xrdp-sesman[50068]: [ERROR] sesman_main_loop: trans_check_wait_objs failed, removing trans
Jul 27 08:36:57 DietPi xrdp-sesman[50145]: [INFO ] Found X server running at /tmp/.X11-unix/X10
Jul 27 08:36:57 DietPi xrdp-sesman[50145]: [INFO ] Session started successfully for user dietpi on display 10
Jul 27 08:36:57 DietPi xrdp-sesman[50145]: [INFO ] Session in progress on display 10, waiting until the window manager (pid 50146) exits to end the session
```

vinhtq115, 2023-07-28:
> Actually
> 
> > Black screen (or aqua one, depends on how I login).
> 
> You do not see the desktop wallpaper/background, do you? Does this help?
> 
> ```shell
> GGI_PRESERVE=1 G_CONFIG_INJECT 'Option[[:blank:]]+"DefaultServerLayout"' '    Option "DefaultServerLayout" "X11 Server"' /etc/X11/xrdp/xorg.conf 'Section[[:blank:]]+"ServerFlags"'
> ```
> 
> Some similar reports:
> 
> * https://forums.debian.net/viewtopic.php?t=154096
> * https://superuser.com/questions/1770194/rdp-to-xrdp-debian-black-screen
> * https://dietpi.com/forum/t/lxde-xrdp-cant-get-a-connection/6514
> 
> No reason identified or solution found in any of theses cases 🤔.

Previously, I didn't wait long enough (4-5 minutes) so I didn't know that the error would pop up. I have uploaded the screenshot in [this reply](https://github.com/MichaIng/DietPi/issues/6503#issuecomment-1652790804). About your question, yes, no background or anything.

I tried your command and still got the same black screen and then error box 5 minutes later.

vinhtq115, 2023-07-28:
So I decided to try the installation script from [this site](https://c-nergy.be/products.html) and it worked for me.

MichaIng, 2023-07-29:
Does it help to reduce the resolution on the RDP client?

It should not have an effect, but is KMS enabled or still the legacy graphics driver? To switch to KMS:
```sh
G_CONFIG_INJECT 'dtoverlay=' 'dtoverlay=vc4-kms-v3d' /boot/config.txt
reboot
```

infinitejones, 2023-10-15:
Adding a comment for anyone else who finds themselves here - I attempted to run the xrdp installation script and in doing so realised that somehow `/home/dietpi` had ended up owned by root. (I did `ls -la /home` to check.)

So I did a quick `chmod -Rv dietpi: /home/dietpi` and voila... the XRDP session starts with no problem at all.

Might not be the same cause as OP's issue, but if you end up here and you're trying to rule things out, it's worth taking 30 seconds to check...

vinhtq115, 2023-10-15:
> Adding a comment for anyone else who finds themselves here - I attempted to run the xrdp installation script and in doing so realised that somehow `/home/dietpi` had ended up owned by root. (I did [...]

Probably not because of the script. My home folder’s owner is still dietpi and I haven’t run any `chmod` on that folder since I ran the script.

infinitejones, 2023-10-15:
No, not because of the script - sorry, I should have been clearer.

I tried to run the script as the dietpi user, in `/home/dietpi`, and couldn't. Realised it was because `/home/dietpi` had ended up being owned by root. 

Fixed ownership, and XRDP (as installed via `dietpi-software`) worked fine, without even needing to run the script. This was the resolution of the "black screen when connecting" problem for me.

MichaIng, 2023-10-15:
@vinhtq115 
Which arguments did you use to run the script? It does a lot of different things, up to custom builds, depending on how you call it. The most usual thing it does on Debian is installing XRDP from Debian backports, applies the same SSL access permissions fix than us, and then add
```sh
session required pam_env.so readenv=1 user_readenv=0
```
to `/etc/pam.d/xrdp-sesman`. To me this step looks wrong, as this is implied in `/etc/pam.d/common-session` already, without `readenv=1 user_readenv=0` defined, but those are both the default values.

However, the script is hard to read. Its output would be interesting, aside of the CLI arguments you use, so I can better see what functions are executed.

vinhtq115, 2023-11-04:
> @vinhtq115 Which arguments did you use to run the script? It does a lot of different things, up to custom builds, depending on how you call it. The most usual thing it does on Debian is installing XRDP from Debian backports, applies the same SSL access permissions fix than us, and then add
> 
> ```shell
> session required pam_env.so readenv=1 user_readenv=0
> ```
> 
> to `/etc/pam.d/xrdp-sesman`. To me this step looks wrong, as this is implied in `/etc/pam.d/common-session` already, without `readenv=1 user_readenv=0` defined, but those are both the default values.
> 
> However, the script is hard to read. Its output would be interesting, aside of the CLI arguments you use, so I can better see what functions are executed.

Sorry for late reply. Here is what I used and the output from terminal:

[Untitled.txt](https://github.com/MichaIng/DietPi/files/13256011/Untitled.txt)

I also tried installing xrdp using the script without `-c` flag and this also causes black screen.

closed 2023-11-05 by vinhtq115

MichaIng, 2023-11-05:
You should remove the duplicate APT list:
```sh
sudo rm /etc/apt/sources.list.d/archive_uri-http_deb_debian_org_debian-bookworm.list
```
So the important difference with the script and `-c` is that it compiled latest XRDP from source. Looks like this solved some bug affecting your case. Probably the same has been fixed with the package provided on Bookworm already.

vinhtq115, 2023-11-05:
> You should remove the duplicate APT list:
> 
> ```shell
> sudo rm /etc/apt/sources.list.d/archive_uri-http_deb_debian_org_debian-bookworm.list
> ```
> 
> So the important difference with the script and `-c` is that it compiled latest XRDP from source. Looks like this solved some bug affecting your case. Probably the same has been fixed with the package provided on Bookworm already.
> 
> EDIT: Ah, in OP you had a Bullseye system, but based on the scripts output you are now on Bookworm already? However, probably a reflash solves it.

Yeah. I upgraded from Bullseye to Bookworm yesterday through the command on the blog. I will try flashing the OS and install everything to see if there are any issues.