MichaIng/DietPi #6505: dietpi-software script >>>  simplified and made modular

robyscar, 2023-07-28:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- dietpi-software script could be simplified and made modular. 

#### Describe the solution you'd like:

- sometime, is necessary to workaround some limitations and/ or make some tweaks.

#### Describe alternatives you've considered:

- every time i spend DAYS tweaking the DIETPI-SOFTWARE script, then there is some patch or change, and all work is lost :(

#### Additional context

- I  understand that 
1) dietpi is dietpi because is a "monolithic" solution
2) all in one file is needed for the dependencies

BUT, sometime, is necessary to workaround some limitations and/ or make some tweaks.
While DietPi is my distro of choice for the RPI, sometime i need something more personalized (so i tweek) to my needs (or another user) needs.

One example is NEXTCLOUD, which i prefer to setup on different ways: Postgres (now available also in 15), without Redis and other "SOHO" environments' needs, and limiting the PHO cache.
This is just one of the examples.
Other could be to freeze the PiHole to the rel I WANT, in order to have the STABILITY that Debian offers (and so DietPi).

thank for  your time

R1oby

Joulinar, 2023-07-28:
I guess that's not going to happen. In your case it might be better to create your own fork on GitHub to host your own version and improvements. On the other hand, feel free to contribute to DietPi and open your own PR if there are improvements that might be interesting for others :)

MichaIng, 2023-07-31:
What does "SOHO" mean? Trust me, you do not ever want to run any serious Nextcloud instance without Redis for transactional file locking. Doing this on the database leads to very bad performance and causes regular problems where you need to unlock files manually when it was just a little too busy. Redis is just as mandatory as to NOT use SQLite but a real database server. PostgreSQL as alternative database is legit. Not sure what the practical benefits are over MariaDB, but we thought about a database choice in the past. However, as this seems to be more a question of personal taste than a real net benefit of one over the other, there are more important things to dedicate our limited development time. If you or anyone however wants to contribute a database choice for Nextcloud or other software options, I am happy to support it.

> limiting the PHO cache.

Do you mean PHP cache? You can just edit the PHP config as you would do on any other system. Not problem with that. I always suggest to create an own config as module to have things very transparent and easy to revert without touching default configs, like:
```sh
cat << '_EOF_' > /etc/php/8.2/mods-available/custom.ini
; Personal PHP overrides
; priority=99
opcache.memory_consumption=64
apc.shm_size=2M
_EOF_
phpenmod custom
```
This is how I limit OPcache and APCu to the size expected/needed by my personal Nextcloud instance.

> Other could be to freeze the PiHole to the rel I WANT, in order to have the STABILITY that Debian offers (and so DietPi).

What do you mean by this? There is no enforcement or automated Pi-hole update anywhere. You should always read the changelog and then apply the update manually via `pihole -up`, if wanted. The "stability that Debian" offers is implied in all parts of Pi-hole aside of its own code updated via `pihole -up`. Webserver, PHP, DNS tools etc is all official Debian packages, hence only security patches and bug fixes are applied automatically as part of DietPi updates or manual `apt upgrade`, and you should always do those upgrades, if you care about your system stability and security for a server with as sensitive software as DNS services running on it.

> sometime i need something more personalized (so i tweek) to my needs (or another user) needs.

Actually I do not really understand your request: If you want to do something manually or tweak something, you can always do so. You can install Nextcloud manually, you can install it via `dietpi-software` and then adjust the setup afterwards, even change the database server to PostgreSQL, if you want. There is nothing forcing you to use any of our tools, and nothing prevents you from tweaking the setup after using them. Just be careful then when reusing them.

> all in one file is needed for the dependencies

There are some dependencies between the scripts indeed, like the ones in `/boot/dietpi/func` are used by some other ones. E.g. `dietpi-config` does not contain all code it does, but it is only the frontend for some options and calls other scripts to apply the changes. Also `dietpi-update` pulls only all DietPi scripts at once. But that does not mean that you need to use all of them. So I am not sure if I understand the actual idea/purpose of the modularisation you think about.