MichaIng/DietPi #6506: Debian Bullseye enabled fs.protected_regular leads to failing /boot/dietpi/func/dietpi-logclear

Timoses, 2023-07-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
  ```
  G_DIETPI_VERSION_CORE=8
  G_DIETPI_VERSION_SUB=19
  G_DIETPI_VERSION_RC=1
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
  G_LIVE_PATCH_STATUS[0]='applied'
  G_LIVE_PATCH_STATUS[1]='applied'
  G_LIVE_PATCH_STATUS[2]='applied'
  ```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
  ```
  bookworm 0
  ```
- Kernel version | `uname -a`
  ```
  Linux dietpi.robinia.dynu.net 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
  ```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
  ```
  RPi 3 Model B+ (aarch64)
  ```
- Power supply used | (EG: 5V 1A RAVpower) 5V DC 3A
- SD card used | (EG: SanDisk ultra) SanDisk Extreme 32GB

#### Steps to reproduce

1. sudo -i
2. /boot/dietpi/func/dietpi-logclear 1

#### Expected behaviour
Successfully "rotate"/truncate logs in /var/log

#### Actual behaviour
```
root@dietpi:~# /boot/dietpi/func/dietpi-logclear 1
/boot/dietpi/func/dietpi-logclear: line 126: /var/log/postgresql/postgresql-15-main.log.1: Permission denied
/boot/dietpi/func/dietpi-logclear: line 126: /var/log/postgresql/postgresql-15-main.log: Permission denied
[ INFO ] DietPi-Logclear | Summary:
 - Log file directory      | /var/log
 - Processed files         | 13
 - Cleared log files       | 2
 - Unsupported files       | 0
 - Deleted files           | 0
 - Space cleared           | 97 KiB
```

#### Extra details
Probably due to the following change in Bullseye (see [here](https://wiki.debian.org/NewInBullseye)):
> [systemd](https://packages.debian.org/systemd) now enables the [fs.protected_fifos and fs.protected_regular kernel parameters](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/sysctl/fs.txt?id=30aba6656f61ed44cba445a3c0d38b296fa9e8f5) by default. This may break applications that share files in /tmp (or other sticky directories) among multiple user accounts. The protection may be disabled temporarily by running sysctl fs.protected_regular=0 and sysctl fs.protected_fifos=0 or permanently by creating a file with a name ending in .conf in the /etc/sysctl.d directory

The sticky bit on `/var/log/postgresql` directory disallows writing to the file owned by somebody else due to `fs.protected_regular` kernel param enabled now:
```
root@dietpi:~# ls -la /var/log/postgresql/
total 100
drwxrwxr-t 2 root     postgres    80 Jul 24 16:17 .
```

I tested `logrotate` on that directory which works fine.

The following may be a suitable replacement for the script in `/boot/dietpi/func/dietpi-logclear`:
```
truncate -c -s0 /var/log/postgresql/postgresql-15-main.log
```
(`-c` is required in order to not trigger the `O_CREAT` call. Without `-c- it fails).

See also Forum post: https://dietpi.com/forum/t/boot-dietpi-func-dietpi-logclear-1-boot-dietpi-func-dietpi-logclear-line-126-var-log-postgresql-postgresql-15-main-log-permission-denied/17426/9

MichaIng, 2023-07-28:
Hmm, as the directory is only group-writable, `fs.protected_regular` must be set to `2` instead of `1`: https://www.kernel.org/doc/html/latest/admin-guide/sysctl/fs.html#protected-regular

I know this behaviour from `/tmp`, which is world-writable, but have never recognised it for group-writable sticky bit dirs.

But you are right, indeed this is the case:
```console
root@dietpi:~# sysctl -a | grep 'protected_regular'
fs.protected_regular = 2
```

What I do not like about the `truncate` solution is that it implies a (slow) external command call for every file. We could temporarily chance the setting, but that short time window could be misused, breaking the security purpose of this setting, so that is not great either. So I currently see now other way either, and to be true, the overall performance penalty should not be recognisable.

What we could do to enhance performance and theoretically decrease some write overhead in case of maaaany log files, is to collect the file paths to empty in a variable, and then empty them all together in a single `truncate` call. What do you think?

Timoses, 2023-07-28:
Hm, array and gathered one-off call to truncate would be an option.

Less extensive could be this:
```
> "$i" || truncate -c -s0 "$i"
```
What do you think about that?

I did go ahead and pushed a version with your suggestion. Gather the failed truncations in a list and fire a one-time truncate call.

MichaIng, 2023-07-28:
> Less extensive could be this:

I had the exact same idea 😄. But I did finally not consider it since this does not mute the error message. And muting all errors messages of the shell redirect could hide important error messages as well, when unrelated to the sticky bit.

Timoses, 2023-07-29:
Yeh, it's not pretty. Researched a bit whether it's possible to use shell redirect without `open(O_CREAT)`, but seems to always use `O_CREAT|O_TRUNC`...

And yes, I didn't like "muting" stderr much either. It's assumptive.

So we switch completely to `truncate` then? Then I'd adjust PR accordingly.

MichaIng, 2023-07-29:
Thanks for also looking into alternatives.

> So we switch completely to `truncate` then? Then I'd adjust PR accordingly.

Jep, its best I think. I did related suggestion on the PR already.

PR #6507 (2023-07-29): Fix dietpi-logclear failing on files in sticky bit directories
fixes #6506
closed 2023-07-29 by MichaIng