MichaIng/DietPi #6509: DietPi-Config - WiFi Hotspot mode: Check for 8 characters minimum

StephanStS, 2023-07-29:
### Enhance robustness for WiFi-Hotspot configuration

See dialog for WiFi-Hotspot configuration:

[image]

#### Is your feature request related to a problem? Please describe:
If you set the key in the dialog, there is a remark to set it to at least 8 chars.

[image]

If you ignore this hint, the dialog sets the key to a smaller value and in the following error occur:

hostapt cannot be started:
`Job for hostapd.service failed because the control process exited with error code.`

One of the `journalctl` output parts is:
```
Jul 29 13:57:51 blockerneo2 hostapd[1580]: Configuration file: /etc/hostapd/hostapd.conf
Jul 29 13:57:51 blockerneo2 hostapd[1580]: Line 11: invalid WPA passphrase length 2 (expected 8..63)
```

which then gives the hint of the user's fault.

#### Describe the solution you'd like:
I propose to add a check in the key dialog that the user can not enter a key with less than 8 characters. 
At minimum a clear warning should be issued, that hostapd will not work with a shorter key.

MichaIng, 2023-07-29:
Makes sense. It is a small/trivial thing we have functions for, so maybe I can do it even for this release.

MichaIng, 2023-08-24:
Implemented with: #6574
I also added a check to `dietpi-software` with a fall back to the default if the key in `dietpi.txt` is too short. Is this okay or is it better to have it failing, so admins need to adjust via `dietpi-config`?

closed 2023-08-26 by MichaIng