MichaIng/DietPi #6511: zram swap is not enabled after boot

magicfoxt-magicfox, 2023-07-29:
#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=8 G_DIETPI_VERSION_SUB=19 G_DIETPI_VERSION_RC=1 G_GITBRANCH='master' G_GITOWNER='MichaIng' G_LIVE_PATCH_STATUS[0]='applied' G_LIVE_PATCH_STATUS[1]='applied' G_LIVE_PATCH_STATUS[2]='not applicable'
- Distro version | bookworm 1
- Kernel version | Linux DietPi0 6.1.21+ #1642 Mon Apr 3 17:19:14 BST 2023 armv6l GNU/Linux
- SBC model | RPi Zero W (armv6l)
- Power supply used | Raspberry Pi official power 2.5A
- SD card used | Lexar HC I 16GB

#### Additional Information (if applicable)
- Software title | dietpi-set_swapfile
- Fresh bookworm install on PI0W
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID | 81d9818e-fe55-4f07-b9e6-be8e1493817c

#### Steps to reproduce
1. Enable zram swap auto-choice support in config /boot/dietpi.txt
2. First boot of DietPi

#### Expected behaviour
An enabled zram swap

#### Actual behaviour
zram swap is not enabled after each reboot

free
               total        used        free      shared  buff/cache   available
Mem:          488696       40260      373288        1640       87948      448436
Swap:              0           0           0

#### Extra details
journalctl -xe
...
Jul 29 18:08:53 DietPi0 systemd[1]: Started systemd-udevd.service - Rule-based Manager for Device Events and Files.
   Subject: A start job for unit systemd-udevd.service has finished successfully
   Defined-By: systemd
   Support: https://www.debian.org/support
    
   A start job for unit systemd-udevd.service has finished successfully.
   
   The job identifier is 11.
Jul 29 18:08:54 DietPi0 kernel: zram0: detected capacity change from 0 to 487424
Jul 29 18:08:54 DietPi0 (udev-worker)[135]: zram0: Process 'chmod 0600 /dev/zram0' failed with exit code 1.
Jul 29 18:08:54 DietPi0 (udev-worker)[135]: zram0: Process '/sbin/swapon /dev/zram0' failed with exit code 255.
Jul 29 18:08:54 DietPi0 (udev-worker)[136]: zram0: /etc/udev/rules.d/98-dietpi-zram-swap.rules:1 Failed to write ATTR{/sys/devices/virtual/block/zram0/disksize}, ignoring: Device or resource busy
Jul 29 18:08:54 DietPi0 kernel: zram: Cannot change disksize for initialized device
Jul 29 18:08:54 DietPi0 (udev-worker)[136]: zram0: Process 'chmod 0600 /dev/zram0' failed with exit code 1.
Jul 29 18:08:55 DietPi0 (udev-worker)[136]: zram0: Process '/sbin/swapon /dev/zram0' failed with exit code 255.

##### Workaround
/boot/dietpi/func/dietpi-set_swapfile 1 zram

dietpi-set_swapfile
─────────────────────────────────────────────────────
 Mode: Applying 1 zram

[ INFO ] DietPi-Set_swapfile | Disabling and deleting all existing swap files
[  OK  ] DietPi-Set_swapfile | swapoff -a
removed '/etc/modules-load.d/dietpi-zram-swap.conf'
removed '/etc/udev/rules.d/98-dietpi-zram-swap.rules'
[  OK  ] DietPi-Set_swapfile | Generating new swap space
[ INFO ] DietPi-Set_swapfile | Size = 238 MiB
[ INFO ] DietPi-Set_swapfile | Path = /dev/zram0
[  OK  ] DietPi-Set_swapfile | eval echo 1 > /sys/block/zram0/reset
[  OK  ] DietPi-Set_swapfile | eval echo '238M' > /sys/block/zram0/disksize
[  OK  ] DietPi-Set_swapfile | chmod 0600 /dev/zram0
[  OK  ] DietPi-Set_swapfile | mkswap /dev/zram0
[  OK  ] DietPi-Set_swapfile | swapon /dev/zram0
[  OK  ] DietPi-Set_swapfile | eval echo 'zram' > /etc/modules-load.d/dietpi-zram-swap.conf
[  OK  ] DietPi-Set_swapfile | eval echo 'SUBSYSTEM=="block", KERNEL=="zram0", ACTION=="add", ATTR{disksize}="238M", RUN+="chmod 0600 /dev/zram0", RUN+="/sbin/mkswap /dev/zram0", RUN+="/sbin/swapon /dev/zram0"' > /etc/udev/rules.d/98-dietpi-zram-swap.rules
[  OK  ] DietPi-Set_swapfile | eval echo 'swappiness=50' > /etc/sysctl.d/98-dietpi-zram-swap.conf
[  OK  ] DietPi-Set_swapfile | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_SIZE=238
[  OK  ] DietPi-Set_swapfile | Desired setting in /boot/dietpi.txt was already set: AUTO_SETUP_SWAPFILE_LOCATION=zram
[ INFO ] DietPi-Set_swapfile | Setting /tmp tmpfs size: 357 MiB
[  OK  ] DietPi-Set_swapfile | Desired setting in /etc/fstab was already set: tmpfs /tmp tmpfs size=357M,noatime,lazytime,nodev,nosuid,mode=1777
[  OK  ] DietPi-Set_swapfile | systemctl daemon-reload
[  OK  ] DietPi-Set_swapfile | mount -o remount /tmp

free
               total        used        free      shared  buff/cache   available
Mem:          488696       39508      373932        1640       88056      449188
Swap:         243708           0      243708

magicfoxt-magicfox, 2023-07-30:
#### Persitant workaround on reboot:
``` 
nano /etc/rc.local

#!/bin/sh
# Provisional workaround cf https://github.com/MichaIng/DietPi/issues/6511 #
/usr/sbin/modprobe zram && /usr/sbin/mkswap /dev/zram0 && /usr/sbin/swapon /dev/zram0

chmod 755 /etc/rc.local
```

```
journalctl -u rc-local
Jul 30 13:25:11 DietPi0 systemd[1]: Starting rc-local.service - /etc/rc.local Compatibility...
Jul 30 13:25:11 DietPi0 rc.local[356]: mkswap: /dev/zram0: warning: wiping old swap signature.
Jul 30 13:25:11 DietPi0 rc.local[356]: Setting up swapspace version 1, size = 238 MiB (249556992 bytes)
Jul 30 13:25:11 DietPi0 rc.local[356]: no label, UUID=d5bf2ebd-9ca2-4256-84b8-3fdfac79ce2d
Jul 30 13:25:11 DietPi0 systemd[1]: Started rc-local.service - /etc/rc.local Compatibility.
```

MichaIng, 2023-07-30:
Hmm, looks like the udev rule does not work in your case. Can you check the two files:
```sh
cat /etc/sysctl.d/98-dietpi-zram-swap.conf
cat /etc/udev/rules.d/98-dietpi-zram-swap.rules
```
The first is to enforce the kernel module to be loaded, hence this should make the `modprobe zram` in your `rc.local` obsolete. The second creates and enables the swap space, once the `zram0` device is added/detected, same as the other commands in your `rc.local` do.

Little note on your workaround, until we have this fixed: You may want to add `chmod 0600 /dev/zram0` as well, to ensure only root raw R/W access. By default, the `disk` group has R/W access as well, but this group is not meant to theoretically have full insights into any internal/private application data running on the server.

Is udev reporting any errors?
```sh
journalctl -u systemd-udevd
```

magicfoxt-magicfox, 2023-07-30:
Hi [MichaIng](https://github.com/MichaIng),

```
cat /etc/sysctl.d/98-dietpi-zram-swap.conf
swappiness=50

cat /etc/udev/rules.d/98-dietpi-zram-swap.rules
SUBSYSTEM=="block", KERNEL=="zram0", ACTION=="add", ATTR{disksize}="238M", RUN+="chmod 0600 /dev/zram0", RUN+="/sbin/mkswap /dev/zram0", RUN+="/sbin/swapon /dev/zram0"
```

You're right, errors are present on udev:
```
journalctl -u systemd-udevd
Jul 30 13:59:13 DietPi0 systemd[1]: Starting systemd-udevd.service - Rule-based Manager for Device Events and Files...
Jul 30 13:59:13 DietPi0 systemd-udevd[130]: Using default interface naming scheme 'v252'.
Jul 30 13:59:13 DietPi0 systemd[1]: Started systemd-udevd.service - Rule-based Manager for Device Events and Files.
Jul 30 13:59:14 DietPi0 (udev-worker)[135]: zram0: Process 'chmod 0600 /dev/zram0' failed with exit code 1.
Jul 30 13:59:14 DietPi0 (udev-worker)[135]: zram0: Process '/sbin/swapon /dev/zram0' failed with exit code 255.
Jul 30 13:59:14 DietPi0 (udev-worker)[136]: zram0: /etc/udev/rules.d/98-dietpi-zram-swap.rules:1 Failed to write ATTR{/sys/devices/virtual/block/zram0/disksize}, ignoring: Device or resource busy
Jul 30 13:59:14 DietPi0 (udev-worker)[136]: zram0: Process 'chmod 0600 /dev/zram0' failed with exit code 1.
Jul 30 13:59:15 DietPi0 (udev-worker)[136]: zram0: Process '/sbin/swapon /dev/zram0' failed with exit code 255.
```
Regards

magicfoxt-magicfox, 2023-07-30:
In addition, I have errors too on a DietPi v8.19.1 PI3B (Debian 11.7) with zram swap enabled:
```
journalctl -u systemd-udevd
Jul 30 13:59:17 DietPi systemd[1]: Starting Rule-based Manager for Device Events and Files...
Jul 30 13:59:17 DietPi systemd-udevd[165]: Network interface NamePolicy= disabled on kernel command line, ignoring.
Jul 30 13:59:17 DietPi systemd[1]: Started Rule-based Manager for Device Events and Files.
Jul 30 13:59:17 DietPi systemd-udevd[168]: zram0: Process 'chmod 0600 /dev/zram0' failed with exit code 1.
Jul 30 13:59:18 DietPi systemd-udevd[171]: Using default interface naming scheme 'v247'.
Jul 30 13:59:18 DietPi systemd-udevd[175]: zram0: /etc/udev/rules.d/98-dietpi-zram-swap.rules:1 Failed to write ATTR{/sys/devices/virtual/block/zram0/disksize}, ignoring: Device or resource busy
Jul 30 13:59:18 DietPi systemd-udevd[175]: zram0: Process 'chmod 0600 /dev/zram0' failed with exit code 1.
Jul 30 13:59:18 DietPi systemd-udevd[175]: zram0: Process '/sbin/mkswap /dev/zram0' failed with exit code 1.
Jul 30 13:59:18 DietPi systemd-udevd[175]: zram0: Process '/sbin/swapon /dev/zram0' failed with exit code 255.
```
but zram swap is active:
```
free
               total        used        free      shared  buff/cache   available
Mem:          982532       60964      793044        3016      128524      902620
Swap:         490492           0      490492
```

MichaIng, 2023-07-30:
Ah sorry, this is the first file I mean (wrong copy&paste):
```sh
cat /etc/modules-load.d/dietpi-zram-swap.conf
```
However, the issue can be seen in udev logs.
```
Process 'chmod 0600 /dev/zram0' failed with exit code 1.
```
The exit code 1 seems to be used for any error, so without the exact output, not sure what is wrong. But I guess the `/dev` node is not immediately present when `/sys/class/block/zram0` is. Probably we simply need to wait for another second before setting it up.

The `swapon` command at least indeed returns 255 if the target does not exist:
```console
root@VM-Bookworm:~# /sbin/swapon test; echo $?
swapon: cannot open test: No such file or directory
255
```

Hmm, that everything is executed twice  makes me think that this is the problem: Two concurrent attempts to setup the swap space, breaking each other.

MichaIng, 2023-07-30:
Bingo, assuring that it runs once only, by checking that the disk size is still zero, solves the double execution. Then `swapon` still fails.

But now I see the actual issue: https://github.com/systemd/systemd/issues/23054

So here is how to solve the issue:
```sh
mkdir /etc/systemd/system/systemd-udevd.service.d
echo -e '[Service]\nSystemCallFilter=@swap' > /etc/systemd/system/systemd-udevd.service.d/dietpi-zram.conf
```

And the more cosmetic change to avoid the doubled execution:
```sh
echo 'SUBSYSTEM=="block", KERNEL=="zram0", ACTION=="add", ATTR{disksize}=="0", ATTR{disksize}="256M", RUN+="/bin/chmod 0600 /dev/zram0", RUN+="/sbin/mkswap /dev/zram0", RUN+="/sbin/swapon /dev/zram0"' > /etc/udev/rules.d/98-dietpi-zram-swap.rules
```

MichaIng, 2023-07-30:
Fixed with: https://github.com/MichaIng/DietPi/commit/55019bb

magicfoxt-magicfox, 2023-07-30:
Thanks for your job ;)

Note the sysctl swappiness is not correcty set and later forced to 1 in /etc/sysctl.d/dietpi.conf, I propose this patch:

```
--- dietpi-set_swapfile.55019bb	2023-07-30 19:40:58.000000000 +0200
+++ dietpi-set_swapfile.55019bb.new	2023-07-30 19:45:16.000000000 +0200
@@ -112,7 +112,8 @@
 			# Make persistent, proceed with failure in case of R/O rootfs
 			G_EXEC_NOHALT=1 G_EXEC eval 'echo '\''zram'\'' > /etc/modules-load.d/dietpi-zram-swap.conf' || return 1
 			G_EXEC_NOHALT=1 G_EXEC eval "echo 'SUBSYSTEM==\"block\", KERNEL==\"zram0\", ACTION==\"add\", ATTR{disksize}==\"0\", ATTR{disksize}=\"${SWAP_SIZE}M\", RUN+=\"/bin/chmod 0600 /dev/zram0\", RUN+=\"/sbin/mkswap /dev/zram0\", RUN+=\"/sbin/swapon /dev/zram0\"' > /etc/udev/rules.d/98-dietpi-zram-swap.rules" || return 1
-			G_EXEC_NOHALT=1 G_EXEC eval 'echo '\''swappiness=50'\'' > /etc/sysctl.d/98-dietpi-zram-swap.conf' || return 1
+			G_EXEC_NOHALT=1 G_EXEC eval 'echo '\''vm.swappiness=50'\'' > /etc/sysctl.d/98-dietpi-zram-swap.conf' || return 1
+			G_EXEC_NOHALT=1 G_EXEC eval "sed -i 's/^vm.swap/\#vm.swap/' /etc/sysctl.d/dietpi.conf" || return 1
 			# Bookworm: Permit udev to run swapon: https://github.com/MichaIng/DietPi/issues/6511
 			G_EXEC_NOHALT=1 G_EXEC mkdir -p /etc/systemd/system/systemd-udevd.service.d
 			G_EXEC_NOHALT=1 G_EXEC eval 'echo -e '\''[Service]\nSystemCallFilter=@swap'\'' > /etc/systemd/system/systemd-udevd.service.d/dietpi-zram.conf'
```

MichaIng, 2023-07-30:
You are right about the swappiness 😮. We create a config to set it to 50, but it is alphanumerically ordered before the default config file, so is overridden with the default 1.

[This file](https://github.com/MichaIng/DietPi/blob/master/rootfs/etc/sysctl.d/dietpi.conf) needs to be renamed to `97-dietpi.conf`, so `98-dietpi-zram-swap.conf` is applied afterwards and `99-sysctl.conf` (which is a symlink to `/etc/sysctl.conf`) is applied last, i.e. overrides everything else.

closed 2023-08-26 by MichaIng