MichaIng/DietPi #6512: Update to 8.20.1 fails bc of downgrade of ffmpeg (which should not be necessary in my case?)

JappeHallunken, 2023-07-30:
### Creating a bug report/issue

#### Required Information
- DietPi version |
- G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=20
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bookworm 0
- Kernel version | Linux RPi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4

When attending the update to 8.20.1 (from 8.19.1) the update fails with:
```
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://ftp.de.debian.org/debian bookworm InRelease
Get:2 http://ftp.de.debian.org/debian bookworm-updates InRelease [52.1 kB]
Get:3 http://ftp.de.debian.org/debian bookworm-backports InRelease [52.2 kB]
Hit:4 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Hit:6 https://downloads.plex.tv/repo/deb public InRelease
Hit:7 https://packagecloud.io/ookla/speedtest-cli/debian bullseye InRelease
Fetched 104 kB in 3s (35.1 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following packages were automatically installed and are no longer required:
  libaom0 libavcodec58 libavdevice58 libavfilter7 libavformat58 libavresample4
  libavutil56 libcodec2-0.9 libdav1d4 libpostproc55 libsrt1.4-gnutls
  libswresample3 libswscale5 libvpx6 libwavpack1 libwebp6 libx264-160
  libx265-192
Use 'apt autoremove' to remove them.
The following NEW packages will be installed:
  libavdevice59 libavfilter8 libavformat59 libcjson1 libmbedcrypto7
  libplacebo208 libpostproc56 librist4 libsrt1.5-gnutls libswscale6 libvulkan1
  libzimg2
The following packages will be DOWNGRADED:
  ffmpeg
0 upgraded, 12 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
E: Packages were downgraded and -y was used without --allow-downgrades.
```
I just removed the `--with-new-pkgs` flag (to prevent the downgrade of `ffmpg` and the update went fine. I think on my system the downgrade is not necessary, but I'm not sure. It was HA related, when you need to compile stuff, right?
Alternativley I have could used `--allow-downgrades` flag to allow the downgrade.
On my system `ffmpeg` is needed for navidrome, and maybe nextcloud, and media playback in this apps is still working.

JappeHallunken, 2023-07-30:
I run another `apt upgrade` and it still downgrades `ffmpeg`, so I should have used the `--allow-downgrades' in the beginng, instead of removing the `--with-new-pkgs`.

MichaIng, 2023-07-30:
Many thanks for the report. Hmm, I tested this up and down on RPi Zero and didn't need to pass any flag 🤔. Will test again.

The "downgrade" is actually an upgrade and needs to be applied on every RPi Bookworm system to prevent possible dependency conflicts and outdated libraries. The FFmpeg source package provides (as you can see) a lot of libraries which are used by a lot of other packages, which expect and in case only work with the new versions on Bookworm.

MichaIng, 2023-07-30:
Okay, pushed a fixed by automatically allowing downgrades during the update. I did not raise the version string, as this does not leave remaining changes on the system, but only the pre-patch and patch files downloaded and executed during the update.

closed 2023-07-30 by MichaIng