MichaIng/DietPi #6525: Dietpi Issue with updating

danielrichert62, 2023-08-01:
### Creating a bug report/issue

#### Details:
- Date           | Tue Aug  1 06:22:16 CDT 2023
- DietPi version | v8.16.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- Hardware       | Odroid XU3/XU4/MC1/HC1/HC2 (armv7l) (ID=11)
- Kernel version | `Linux hc1-server 4.14.294+ #1 SMP PREEMPT Sat Sep 24 10:24:15 UTC 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `getent hosts 8.8.8.8`
- Exit code      | 2
- Software title | DietPi-Update
#### Steps to reproduce:

1. ssh into server
2. # dietpi-update
3. Retry fails
4. Network settings
5. Test
6. Test : [Failure] |. Offline
7. # ping 8.8.8.8 "Successful
8. # ping dietpi.com "ping: dietpi.com: Temporary failure in name resolution"

#### Expected behaviour:

- Test should work.  I am connected using ethernet
#### Actual behaviour:

- Updates fail ([FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.)
#### Extra details:

- ...
#### Additional logs:
```

```

Joulinar, 2023-08-01:
looks like an issue with DNS name resolution. Which DNS server you are using? Do you use DHCP or Static IP?

danielrichert62, 2023-08-01:
I'm using the PiHole as my DNS.  Static IP.

Joulinar, 2023-08-01:
Is PiHole running on the very same system or on another one?

danielrichert62, 2023-08-01:
Same system.  This is an Odroid HC1 Arm7 with Dietpi running only PiHole.

Joulinar, 2023-08-01:
Please go to the local network settings dietpi-config and change the DNS server to a global public server. This is not necessary for DietPi to use PiHole, it just causes challenges.  Normally there is nothing in DietPi that needs to be blocked as long as the system is not used as a desktop incl browser. This is our best practice, which we recommend in order to avoid exactly such problems. 😉

SurycL, 2023-08-02:
@Joulinar I also experienced this error while running dietpi-update.

I am using pihole, and am using unbound DNS. I think it's a pretty popular configuration, so probably many others will have this issue.

---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Tue Aug  1 23:10:35 PDT 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 http://packages.azlux.fr/debian buster InRelease
Ign:6 http://giteduberger.fr rpimonitor/ InRelease
Hit:8 https://archive.raspberrypi.org/debian bullseye InRelease
Err:7 http://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
Reading package lists...
E: The repository 'http://giteduberger.fr rpimonitor/ Release' no longer has a Release file.
W: http://packages.azlux.fr/debian/dists/buster/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```
---------------------------------------------------------------------

```
```

Joulinar, 2023-08-02:
> so probably many others will have this issue.

No, your issue is not related to the issue above, PiHole and Unbound at all. It's dedicated to your system.

> Err:7 http://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
E: The repository 'http://giteduberger.fr rpimonitor/ Release' no longer has a Release file.

You have (manually) added a 3rd party `apt` package repository that doesn't seems to be working anymore.

> W: http://packages.azlux.fr/debian/dists/buster/InRelease: Key is stored in legacy trusted.gpg keyring 

Another challenge on your system, you seem to have added a Buster repository, while your system is on Debian Bookworm already.

You would need to correct or remove both configs from your system.

Hukuma1, 2023-08-02:
```
│ APT upgrade
│  - Command: apt-get -y --with-new-pkgs upgrade
│  - Exit code: 100
│  - DietPi version: v8.19.1 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6
│  - Image creator: DietPi Core Team
│  - Pre-image: From scratch
│  - Error log:
│ Reading package lists...
│ Building dependency tree...
│ Reading state information...
│ Calculating upgrade...
│ 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
│ 1 not fully installed or removed.
│ After this operation, 0 B of additional disk space will be used.
│ Setting up udev (247.3-7+deb11u4) ...^M
│ Failed to reload daemon: Failed to activate service 'org.freedesktop.systemd1': timed out
│ (service_start_timeout=25000ms)^M
│ Failed to restart udev.service: Connection timed out^M
│ See system logs and 'systemctl status udev.service' for details.^M
│ invoke-rc.d: initscript udev, action "restart" failed.^M
│ Failed to get properties: Failed to activate service 'org.freedesktop.systemd1': timed out
│ (service_start_timeout=25000ms)^M
│ dpkg: error processing package udev (--configure):^M
│  installed udev package post-installation script subprocess returned error exit status 1^M
│ Errors were encountered while processing:^M
│  udev^M
│ E: Sub-process /usr/bin/dpkg returned an error code (1)
```

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
Failed to retrieve unit state: Connection timed out
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.20.1
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.19.1
[ INFO ] DietPi-Update | Latest version  : v8.20.1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=52268 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[FAILED] DietPi-Services | stop : cron
[FAILED] DietPi-Services | stop : docker

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://download.docker.com/linux/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Get:3 https://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:4 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:5 https://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Get:6 https://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:7 https://archive.raspberrypi.org/debian bullseye/main arm64 Packages [308 kB]
Fetched 473 kB in 4s (125 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
                                                              1 not fully installed or removed.
                                                                                               After this operation, 0 B of additional disk space will be used.
                                                            Setting up udev (247.3-7+deb11u4) ...
Failed to reload daemon: Failed to activate service 'org.freedesktop.systemd1': timed out (service_start_timeout=25000ms)
Failed to restart udev.service: Connection timed out
See system logs and 'systemctl status udev.service' for details.
invoke-rc.d: initscript udev, action "restart" failed.
Failed to get properties: Failed to activate service 'org.freedesktop.systemd1': timed out (service_start_timeout=25000ms)
dpkg: error processing package udev (--configure):
 installed udev package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 udev
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade
```

I am running into this issue for the first time ever. Been running and updating DietPi for years without issue until yesterday. I have AdGuard Home running at the router level if it matters. Apologies if this is unrelated, I can easily make a new issue?

MichaIng, 2023-08-02:
Guys, please stop posting into other peoples issues if you are not 100% sure that it is the same issue 😄. This udev error is again completely unrelated to the other two. Sadly GitHub does not provide an easy way to move a comment into a new issue.

However, now we have the mess here already, let me try to do the best with it:
- @danielrichert62 just to be sure your Pi-hole host really uses Pi-hole itself for its own DNS resolution, can you please paste the output of:
    ```sh
    cat /etc/resolv.conf
    ```
    If it really contains `nameserver 127.0.0.1` or its own LAN IP, please try to use another one, like:
    ```sh
    echo 'nameserver 8.8.8.8' > /etc/resolv.conf
    ```
    But actually, Pi-hole is not stopped during DietPi updates, so while the above is recommended anyway (not using Pi-hole with its own host), it should still work to use Pi-hole. So if the step helps: Are other Pi-hole clients able to resolve hostnames?
- @danielrichert62 I didn't know that RPi-Monitor had an own APT repository. However, it has been abandoned, like the whole project. Please identify the file which contains the repo and remove it:
    ```sh
    grep -r 'giteduberger.*rpimonitor' /etc/apt/sources.list.d
    ```
- @Hukuma1 please paste the udev service logs:
    ```sh
    journalctl -u systemd-udevd
    ```

danielrichert62, 2023-08-04:
```
cat /etc/resolv.conf 
nameserver 8.8.8.8
```
```
grep -r 'giteduberger.*rpimonitor' `/etc/apt/sources.list.d
```
returned nothing

```
-- Journal begins at Thu 2022-12-22 05:55:42 CST, ends at Fri 2023-08-04 10:38:09 CDT. --
Jul 24 11:17:01 server systemd[1]: Starting Rule-based Manager for Device Events and Files...
Jul 24 11:17:01 server systemd-udevd[250]: Network interface NamePolicy= disabled on kernel command line, ignoring.
Jul 24 11:17:01 server systemd[1]: Started Rule-based Manager for Device Events and Files.
Jul 24 11:17:01 server systemd-udevd[256]: Using default interface naming scheme 'v247'.
Jul 24 11:17:12 server systemd-udevd[1006]: Using default interface naming scheme 'v247'.

```

closed 2023-08-04 by danielrichert62