MichaIng/DietPi #6526: Error while running dietpi-update

SurycL, 2023-08-02:
#### Details:
- Date           | Tue Aug  1 23:10:35 PDT 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:
run dietpi-update
#### Expected behaviour:
the update finishes without error
#### Actual behaviour:
I get the above error
#### Extra details:
I am running pihole, unbound, and rpimonitor (https://github.com/XavierBerger/RPi-Monitor)
#### Additional logs:
```
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 http://packages.azlux.fr/debian buster InRelease
Ign:6 http://giteduberger.fr rpimonitor/ InRelease
Hit:8 https://archive.raspberrypi.org/debian bullseye InRelease
Err:7 http://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
Reading package lists...
E: The repository 'http://giteduberger.fr rpimonitor/ Release' no longer has a Release file.
W: http://packages.azlux.fr/debian/dists/buster/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```

MichaIng, 2023-08-02:
Thanks for creating an own issue.

I didn't know that RPi-Monitor had an own APT repository. However, it has been abandoned, like the whole project. Please identify the file which contains the repo and remove it:
```sh
grep -r 'giteduberger.*rpimonitor' /etc/apt/sources.list.d
apt update
```

SurycL, 2023-08-02:
@MichaIng apologies for the noise on the other topic, I thought it was the same issue, just didn't want to duplicate information.

After removing those sources, I was able to run apt update and dietpi-update.

Thank you for your help, of course, very much appreciated!

closed 2023-08-02 by SurycL

MichaIng, 2023-08-02:
No problem.

Btw, to address the warning about the second repository, and to move to its latest suite:
```sh
curl -sSfLo /etc/apt/tristed.gpg.d/azlux.gpg 'https://azlux.fr/repo.gpg'
rm /etc/apt/trusted.gpg
sed -i 's/buster/stable/' /etc/apt/sources.list.d/azlux.list # assuming this is the correct filename
apt update
```