MichaIng/DietPi #6527: Unable to update from v8.19.1 to v8.20.1

Hukuma1, 2023-08-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `8.19.1`
- Distro version | `bullseye`
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | oem power supply
- SD card used | usb stick

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

```
│ APT upgrade
│  - Command: apt-get -y --with-new-pkgs upgrade
│  - Exit code: 100
│  - DietPi version: v8.19.1 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6
│  - Image creator: DietPi Core Team
│  - Pre-image: From scratch
│  - Error log:
│ Reading package lists...
│ Building dependency tree...
│ Reading state information...
│ Calculating upgrade...
│ 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
│ 1 not fully installed or removed.
│ After this operation, 0 B of additional disk space will be used.
│ Setting up udev (247.3-7+deb11u4) ...^M
│ Failed to reload daemon: Failed to activate service 'org.freedesktop.systemd1': timed out
│ (service_start_timeout=25000ms)^M
│ Failed to restart udev.service: Connection timed out^M
│ See system logs and 'systemctl status udev.service' for details.^M
│ invoke-rc.d: initscript udev, action "restart" failed.^M
│ Failed to get properties: Failed to activate service 'org.freedesktop.systemd1': timed out
│ (service_start_timeout=25000ms)^M
│ dpkg: error processing package udev (--configure):^M
│  installed udev package post-installation script subprocess returned error exit status 1^M
│ Errors were encountered while processing:^M
│  udev^M
│ E: Sub-process /usr/bin/dpkg returned an error code (1)
```

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
Failed to retrieve unit state: Connection timed out
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 8.20.1
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.19.1
[ INFO ] DietPi-Update | Latest version  : v8.20.1

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=52268 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[FAILED] DietPi-Services | stop : cron
[FAILED] DietPi-Services | stop : docker

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 https://download.docker.com/linux/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye InRelease
Get:3 https://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:4 https://deb.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:5 https://deb.debian.org/debian bullseye-backports InRelease [49.0 kB]
Get:6 https://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:7 https://archive.raspberrypi.org/debian bullseye/main arm64 Packages [308 kB]
Fetched 473 kB in 4s (125 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
                                                              1 not fully installed or removed.
                                                                                               After this operation, 0 B of additional disk space will be used.
                                                            Setting up udev (247.3-7+deb11u4) ...
Failed to reload daemon: Failed to activate service 'org.freedesktop.systemd1': timed out (service_start_timeout=25000ms)
Failed to restart udev.service: Connection timed out
See system logs and 'systemctl status udev.service' for details.
invoke-rc.d: initscript udev, action "restart" failed.
Failed to get properties: Failed to activate service 'org.freedesktop.systemd1': timed out (service_start_timeout=25000ms)
dpkg: error processing package udev (--configure):
 installed udev package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 udev
E: Sub-process /usr/bin/dpkg returned an error code (1)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade
```

I am running into this issue for the first time ever. Been running and updating DietPi for years without issue until yesterday. I have AdGuard Home running at the router level if it matters. Apologies if this is unrelated, I can easily make a new issue?

As per https://github.com/MichaIng/DietPi/issues/6525#issuecomment-1662640190 I have nothing in the output of the log requested:

```
root@DietPi:~# journalctl -u systemd-udevd
-- Journal begins at Fri 2023-07-21 01:25:00 PDT, ends at Mon 2023-07-31 08:23:46 PDT. --
-- No entries --
```

MichaIng, 2023-08-02:
Thanks for opening a dedicated issue.

Hmm strange that there is not a single log entry. Can you check:
```sh
systemctl status systemd-udevd
journalctl -u udev
```

Hukuma1, 2023-08-02:
```
root@DietPi:~# systemctl status systemd-udevd
Failed to get properties: Connection timed out
```

MichaIng, 2023-08-02:
Hmm, at least the running systemd process itself seems to have issue, not returning any service status. This is also a good reason that any service restart attempt fails.

Those "critical medium error"s  do not look good indeed. Are there probably voltage/power errors as well? And what does this return?
```sh
vcgencmd get_throttled
```

Since it aborted in the middle of systemd package upgrades, there is a risk that it may not boot up again when doing a restart. Can you check which packages have not been fully configured yet?
```sh
dpkg -l | grep -v '^ii'
```
AFAIK, `udev` should be one of the last, and there should not be much more to configure than the service restart. But lets assure that.

Hukuma1, 2023-08-02:
```
root@DietPi:~# vcgencmd get_throttled
throttled=0x0
```

```
root@DietPi:~# dpkg -l | grep -v '^ii'
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                          Version                          Architecture Description
+++-=============================-================================-============-===============================================================================
iF  udev                          247.3-7+deb11u4                  arm64        /dev/ and hotplug management daemon
```
 
```
G_AGI smartmontools
smartctl -a /dev/sda
```
I was able to install smartctl when I was thinking I would check S.M.A.R.T. data before realizing it's basically impossible with USB sticks. And it installed just fine. Something with this udev though is not letting it go.

MichaIng, 2023-08-02:
> before realizing it's basically impossible with USB sticks

Yes, USB sticks have no S.M.A.R.T. data, only 2,5"/3,5" and larger drives 😉.

Okay no powering issue, only the `udev` package post-configuration (AFAIK only the service restart) outstanding. Reboot works.

Can you try to just reinstall the `udev` package?
```sh
apt install --reinstall udev
```
Maybe just one of its files is damaged or so.

Hukuma1, 2023-08-04:
Unfortunately right after I did do a shut down, I was unable to boot back up again. :/ Weird how reboots worked just fine, but I couldn't do a shutdown/start. (RPi was stuck on green and amber light constantly on and could not SSH in.)

I had to wipe my flash drive. So it's likely some kind of corruption happened somewhere. I'll close this out. Thanks for your help!

closed 2023-08-04 by Hukuma1