MichaIng/DietPi #6528: apt update - deprecation notice for armbian bullseye repository from bookworm

fra87, 2023-08-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=20
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bookworm
```

- Kernel version | `uname -a`
```
Linux server 5.15.93-rockchip64 #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux
```

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
ROCK64 (aarch64)
```

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
`apt`

- Was the software title installed freshly or updated/migrated?
`updated`

- Can this issue be replicated on a fresh installation of DietPi?
`I don't think so; it seems a configuration carryover issue`

#### Steps to reproduce
1. Update to debian bookwork through the [command](sudo bash -c "$(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/master/.meta/dietpi-bookworm-upgrade')") shared in the [bookwork blog post](https://dietpi.com/blog/?p=3128)
2. Wait for the upgrade to finish, then restart as suggested
3. Launch `sudo apt update`

#### Expected behaviour

- The update goes smoothly
- (probably) armbian repository was changed to bookworm

#### Actual behaviour

- There is a warning about a deprecation:
```
W: https://apt.armbian.com/dists/bullseye/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```

- The armbian repository is still the bullseye one:
```
Hit:6 https://armbian.site-meganet.com/apt bullseye InRelease
```

#### Extra details

- At least for now this is just a warning message, so it's not a priority.
- In addition I did not search whether armbian already fully supports bookworm, so I don't know if it is actually feasible/advisable to change it; I just suppose all the repos shall have been updated to bullseye
- I tested it on another rockchip64 SBC (NanoPI NEO 3) and it shows the same warning

Joulinar, 2023-08-03:
can you share your source file configuration?

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

fra87, 2023-08-03:
Sure, here it is

```
/etc/apt/sources.list:
deb https://deb.debian.org/debian/ bookworm main contrib non-free non-free-firmware
deb https://deb.debian.org/debian/ bookworm-updates main contrib non-free non-free-firmware
deb https://deb.debian.org/debian-security/ bookworm-security main contrib non-free non-free-firmware
deb https://deb.debian.org/debian/ bookworm-backports main contrib non-free non-free-firmware
/etc/apt/sources.list.d/armbian.list:
deb https://apt.armbian.com/ bullseye main
/etc/apt/sources.list.d/docker.list:
deb https://download.docker.com/linux/debian/ bookworm stable

```

MichaIng, 2023-08-03:
Strange. Can you rerun the command which our upgrade script runs and check again?
```sh
G_EXEC sed -i 's/bullseye/bookworm/g' /etc/apt/sources.list.d/*.list
cat /etc/apt/sources.list.d/armbian.list
```

fra87, 2023-08-03:
The command (run as sudo) successfully modified the file:

```
root@server:~# cat /etc/apt/sources.list.d/armbian.list
deb https://apt.armbian.com/ bullseye main
root@server:~# G_EXEC sed -i 's/bullseye/bookworm/g' /etc/apt/sources.list.d/*.list
[  OK  ] sed -i s/bullseye/bookworm/g /etc/apt/sources.list.d/armbian.list /etc/apt/sources.list.d/docker.list
root@server:~# cat /etc/apt/sources.list.d/armbian.list
deb https://apt.armbian.com/ bookworm main
```

This is strange, however. Why didn't it succeed in both my servers?

In any case, even now the issue remains:

```
Hit:6 https://armbian.site-meganet.com/apt bookworm InRelease
Reading package lists...
W: https://apt.armbian.com/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```

Joulinar, 2023-08-03:
The key warning is ok and not doing any harm actually. We will work on this. 

Why the source list was not changed? Well, we found a typo in the Bookworm upgrade script already 😉

MichaIng, 2023-08-10:
You can fix the warning this way:
```sh
curl -sSfL 'https://apt.armbian.com/armbian.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-armbian.gpg --yes
rm /etc/apt/trusted.gpg
apt update
```
I'll see how far we can automate this on next DietPi update.

MichaIng, 2023-09-17:
The key migration is now done for all images which still have the Armbian keyring in `/etc/apt/trusted.gpg`: https://github.com/MichaIng/DietPi/commit/f0b5abc

closed 2023-09-22 by MichaIng