MichaIng/DietPi #6529: File browser software not installing.

lutfor-diu, 2023-08-05:
#### Details:
- Date           | Sat Aug  5 12:29:01 BST 2023
- Bug report     | 8b8d80e6-2ca3-4b3f-bb16-eff5f7058706
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=1)
- Command        | `/opt/filebrowser/filebrowser config init`
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
2023/08/05 12:29:01 ./filebrowser.db already exists

lutfor-diu, 2023-08-05:
Reference code: 8b8d80e6-2ca3-4b3f-bb16-eff5f7058706

bug report uploaded

Joulinar, 2023-08-05:
was this a new fresh installation? Independent on your issue, there is a challenge with Filebrowser v2.24.1 themselves actually.

MichaIng, 2023-08-05:
Interesting. The error others currently face is on the same step, but with a different error message
```
flag accessed but not defined: branding.DisableUsedPercentage
```
Replicated on all platforms here: https://github.com/MichaIng/DietPi/actions/runs/5772908872

Fixed upstream: https://github.com/filebrowser/filebrowser/pull/2596
But a release is still outstanding.

As your error message states that the database file does exist already, while we only run this command when it explicitly does not exist, did you retry that command after you ran into the error with the message above? Quite possible that despite this error, the database file is created, which breaks rerunning the command. This again leads to the question whether we could just ignore the message and continue, or whether some needed mandatory database entries are missing.

lutfor-diu, 2023-08-06:
> was this a new fresh installation? Independent on your issue, there is a challenge with Filebrowser v2.24.1 themselves actually.

yes. fresh new installation.

lutfor-diu, 2023-08-06:
> Interesting. The error others currently face is on the same step, but with a different error message
> 
> ```
> flag accessed but not defined: branding.DisableUsedPercentage
> ```
> 
> Replicated on all platforms here: https://github.com/MichaIng/DietPi/actions/runs/5772908872
> 
> Fixed upstream: [filebrowser/filebrowser#2596](https://github.com/filebrowser/filebrowser/pull/2596) But a release is still outstanding.
> 
> As your error message states that the database file does exist already, while we only run this command when it explicitly does not exist, did you retry that command after you ran into the error with the message above? Quite possible that despite this error, the database file is created, which breaks rerunning the command. This again leads to the question whether we could just ignore the message and continue, or whether some needed mandatory database entries are missing.

when try 2nd time it installs but file browser service failed to start. with error "run exit"

MichaIng, 2023-08-08:
The latest release has this fixed: https://github.com/filebrowser/filebrowser/releases/tag/v2.24.2
Simply reinstall it:
```sh
dietpi-software reinstall 198
```

lutfor-diu, 2023-08-09:
Thank you so much 😊

lutfor-diu, 2023-08-09:
Thanks, it works

lutfor-diu, 2023-08-09:
│ × filebrowser.service - File Browser (DietPi)                                          ││      Loaded: loaded (/etc/systemd/system/filebrowser.service; enabled; preset:         ││ enabled)                                                                               ││      Active: failed (Result: exit-code) since Wed 2023-08-09 01:44:02 BST; 10s ago     ││    Duration: 190ms                                                                     ││        Docs: https://filebrowser.org/                                                  ││     Process: 2708 `ExecStart=/opt/filebrowser/filebrowser` -d                            ││ /mnt/dietpi_userdata/filebrowser/filebrowser.db (code=exited, status=1/FAILURE)        ││    Main PID: 2708 (code=exited, status=1/FAILURE)                                      ││         CPU: 74ms                                                                      ││                                                                                        ││ Aug 09 01:44:02 DietPi systemd[1]: Started filebrowser.service - File Browser          ││ (DietPi).                                                                              ││ Aug 09 01:44:02 DietPi filebrowser[2708]: 2023/08/09 01:44:02 No config file used      ││ Aug 09 01:44:02 DietPi filebrowser[2708]: 2023/08/09 01:44:02 the resource does not    ││ exist                                                                                  ││ Aug 09 01:44:02 DietPi systemd[1]: filebrowser.service: Main process exited,           ││ code=exited, status=1/FAILURE                                                          ││ Aug 09 01:44:02 DietPi systemd[1]: filebrowser.service: Failed with result

lutfor-diu, 2023-08-09:
File browser service fail to start

Joulinar, 2023-08-09:
Remove it completely and install File browser afterwards again

MichaIng, 2023-08-10:
Ah right, init step is otherwise not repeated as the database file has been created already. Minimum step needed before trying the reinstall:
```sh
rm /mnt/dietpi_userdata/filebrowser/filebrowser.db
```

closed 2023-08-26 by MichaIng