MichaIng/DietPi #6534: When ntpsec installed: dietpi-update: Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found

zatie12, 2023-08-07:
#### Details:
- Date           | Mon Aug  7 09:35:04 BST 2023
- DietPi version | v8.19.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | NanoPi NEO3 (aarch64) (ID=56)
- Kernel version | `Linux tiarasol 5.15.93-rockchip64 #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7)
- Command        | `systemctl start systemd-timesyncd`
- Exit code      | 5
- Software title | DietPi-TimeSync
#### Steps to reproduce:

1. apt install ntpsec
#### Expected behaviour:

- dietpi-update completes successfully
#### Actual behaviour:

- dietpi-update tries and fails to start systemd-timesyncd becuase it gets removed when installing ntpsec
#### Extra details:

- 
#### Additional logs:
```
Failed to start systemd-timesyncd.service: Unit systemd-timesyncd.service not found.
```
---------------------------------------------------------------------
[FAILED] DietPi-TimeSync | Unable to continue, DietPi-TimeSync will now terminate.

Joulinar, 2023-08-07:
You would need to disable time sync within dietpi-config if you use an own tool

zatie12, 2023-08-07:
Thanks, I searched everywhere but missed that advanced section! Can close this one

closed 2023-08-07 by Joulinar