MichaIng/DietPi #6536: install failed on Rock Pi 4SE -mmc

MidG971, 2023-08-08:
#### Details:
- Date           | Tue Aug  8 23:26:14 UTC 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  |
- Pre-image      Ubuntu 20.04.6 LTS
- Hardware       |  (ID=24)
- Kernel version | `Linux rock-4se 5.10.110-11-rockchip #8c4e8d205 SMP Fri Jun 16 09:38:43 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Installer
#### Steps to reproduce:

1. ...Follow the install procees
2. ...
#### Expected behaviour:

- DietPi installation
#### Actual behaviour:

- [ INFO ] DietPi-Installer | Executing alternative command: apt-get -y -eany update
E: Malformed entry 1 in list file /etc/apt/sources.list.d/apt-radxa-com.list (Component)
E: The list of sources could not be read.
[FAILED] Alternative command execution | Exited with error
Press any key to return to error handler menu...
root@rock-4se:/tmp/DietPi-Installer# wget -O - apt.radxa.com/$DISTRO/public.key | sudo apt-key add -
--2023-08-08 23:27:06--  http://apt.radxa.com//public.key
Resolving apt.radxa.com (apt.radxa.com)... 2606:50c0:8000::153, 2606:50c0:8003::153, 2606:50c0:8002::153, ...
Connecting to apt.radxa.com (apt.radxa.com)|2606:50c0:8000::153|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-08-08 23:27:06 ERROR 404: Not Found.
#### Extra details:

- ...
#### Additional logs:
```
E: Malformed entry 1 in list file /etc/apt/sources.list.d/apt-radxa-com.list (Component)
E: The list of sources could not be read.

Joulinar, 2023-08-09:
can you share your source files configuration

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

> wget -O - apt.radxa.com/$DISTRO/public.key | sudo apt-key add -

This of course is not working because variable `$DISTRO` is not valid outside our scripts. Therefore, you request result in an incorrect URL `http://apt.radxa.com//public.key` .You would need to set it manually.

MidG971, 2023-08-09:
radxa@rock-4se:~$ for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
/etc/apt/sources.list:
deb https://deb.debian.org/debian/ bookworm main contrib non-free non-free-firmware
deb https://deb.debian.org/debian/ bookworm-updates main contrib non-free non-free-firmware
deb https://deb.debian.org/debian-security/ bookworm-security main contrib non-free non-free-firmware
deb https://deb.debian.org/debian/ bookworm-backports main contrib non-free non-free-firmware
/etc/apt/sources.list.d/apt-radxa-com.list:
deb http://apt.radxa.com//  main
deb http://apt.radxa.com//  main
deb http://apt.radxa.com//  main
/etc/apt/sources.list.d/focal-backports.list:
deb http://ports.ubuntu.com/ubuntu-ports focal-backports main restricted universe multiverse
/etc/apt/sources.list.d/focal-security.list:
deb http://ports.ubuntu.com/ubuntu-ports focal-security main restricted universe multiverse
/etc/apt/sources.list.d/focal-updates.list:
deb http://ports.ubuntu.com/ubuntu-ports focal-updates main restricted universe multiverse
/etc/apt/sources.list.d/radxa.list:
deb [signed-by=/usr/share/keyrings/radxa-archive-keyring.gpg] https://radxa-repo.github.io/focal focal main

Joulinar, 2023-08-09:
`Pre-image Ubuntu 20.04.6 LTS` we don't support Ubuntu at all. You need to use Debian.

MichaIng, 2023-08-09:
> Pre-image Ubuntu 20.04.6 LTS

This cannot work. You are mixing Debian and Ubuntu here. Please use a Debian pre-image.

> `wget -O - apt.radxa.com/$DISTRO/public.key | sudo apt-key add -`

Where did you get this command from? This does not work, neither within the script, nor outside, as the variable `$DISTRO` is not defined anywhere. Also a missing key is not the issue, but the `/etc/apt/sources.list.d/apt-radxa-com.list` which looks like you added the same line with the same non-existent `$DISTRO` variable (three times), so that the suite is missing. However, Radxa's Debian images come with the correct APT list file and key, so it is not needed to change anything.

However, I would not recommend Radxa's images, unless you require a specific kernel feature which does not work with mainline kernel. Armbian provides much cleaner ROCK 4 images with Linux 6.1.

The ROCK 4 image we provide on our download page uses Armbian's kernel packages and works on the SE variant as well.

MidG971, 2023-08-09:
I don't remember where as I was testing Ubuntu on the board, I could'nt update anything...it worked after I used it. 
I found on my notepad this : 
Fix distro ubuntu
=========================
export DISTRO=focal-stable
wget -O - apt.radxa.com/$DISTRO/public.key | sudo apt-key add -

maybe that's where the variable $DISTRO come from 

I'll use the rock 4 image you provide then. Does it enable virtualisation ?

MidG971, 2023-08-09:
I tried the solution you gave me, I still failed mid install : 

`Unpacking libx11-data (2:1.8.4-2+deb12u1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/028-libx11-data_2%3a1.8.4-2+deb12u1_all.deb (--unpack):
 unable to sync new file '/var/lib/dpkg/info/libx11-data.list-new': Input/output error
rm: cannot remove '/var/lib/dpkg/tmp.ci/control': Read-only file system
rm: cannot remove '/var/lib/dpkg/tmp.ci/md5sums': Read-only file system
dpkg: error while cleaning up:
 rm command for cleanup subprocess returned error exit status 1
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/029-libx11-6_2%3a1.8.4-2+deb12u1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/030-libx11-xcb1_2%3a1.8.4-2+deb12u1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/031-libxcb-dri2-0_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/032-libxcb-dri3-0_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/033-libxcb-present0_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/034-libxcb-randr0_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/035-libxcb-sync1_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/036-libxcb-xfixes0_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/037-libxshmfence1_1.3-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/038-libegl-mesa0_22.3.6-1+deb12u1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/039-libepoxy0_1.5.10-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/040-libevdev2_1.13.0+dfsg-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/041-libpng16-16_1.6.39-2_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/042-libfreetype6_2.12.1+dfsg-5_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/043-libfontconfig1_2.14.1-4_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/044-libfontenc1_1%3a1.1.4-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/045-libglvnd0_1.6.0-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/046-libxcb-glx0_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/047-libxcb-shm0_1.15-1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/048-libxext6_2%3a1.3.4-1+b1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/049-libxfixes3_1%3a6.0.0-2_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/050-libxxf86vm1_1%3a1.1.4-1+b2_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/051-libicu72_72.1-3_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/052-libxml2_2.9.14+dfsg-1.3~deb12u1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/053-libz3-4_4.8.12-3.1_arm64.deb (--unpack):
 error ensuring '/var/lib/dpkg/reassemble.deb' doesn't exist: Read-only file system
dpkg: error processing archive /tmp/apt-dpkg-install-LP3yWg/054-libllvm15_1%3a15.0.6-4+b1_arm64.deb (--unpack):
[... cut ...]
[ INFO ] DietPi-Software | Executing alternative command: apt-get -y --allow-change-held-packages install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-user-session xfonts-base x11-xserver-utils x11-utils
W: Not using locking for read only lock file /var/lib/dpkg/lock-frontend
W: Not using locking for read only lock file /var/lib/dpkg/lock
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
[FAILED] Alternative command execution | Exited with error
Press any key to return to error handler menu...
`

MidG971, 2023-08-09:
[[ssh 192.168.1.116]  (2023-08-10_001904)  192.168.1.116.log](https://github.com/MichaIng/DietPi/files/12307380/ssh.192.168.1.116.2023-08-10_001904.192.168.1.116.log)
This is my mobaxterm log for the install

MichaIng, 2023-08-09:
> I found on my notepad this :

Ah indeed there this variable is assigned. But I do not see this in the log of your OP. Btw, `apt-key add` is a legacy method of adding key which should not be used anymore, and `focal` is an Ubuntu version, so not compatible with Debian. However, the key is anyway not the issue.

> I tried the solution you gave me, I still failed mid install :

You mean this happened when you flashed our ROCK 4 image and when through first run setup? It seems there was an I/O error, which lead to the root filesystem being automatically read-only mounted to prevent further damage:
```
 unable to sync new file '/var/lib/dpkg/info/libx11-data.list-new': Input/output error
rm: cannot remove '/var/lib/dpkg/tmp.ci/control': Read-only file system
```
Is this an SD card or eMMC module? If it is an SD card, it might be dying. An eMMC module however should survive for a long time. You could try to remount the filesystem to continue the setup and see whether it goes through:
```sh
mount -o remount,rw /
```
This can be also done from the error handler when opening a subshell. If it works, force an `fsck` on reboot and check logs, to see whether it was a tiny issue only or larger parts of the filesystem corrupted:
```sh
touch /forcefsck
reboot
```
After reboot to check logs:
```sh
cat /run/initramfs/fsck.log
```

MidG971, 2023-08-10:
I found where I got the $DISTRO variable : https://wiki.radxa.com/Rock4/FAQs cf at the end of the page.

The install is done. I'm using a eMMC module. I remounted the filesystem as instructed. I just didn't tried to install the additional packages.

This is my ssh log file in case of : 

[[ssh 192.168.1.116]  (2023-08-10_230049)  192.168.1.116.log](https://github.com/MichaIng/DietPi/files/12317249/ssh.192.168.1.116.2023-08-10_230049.192.168.1.116.log)

Thanks a lot for your help 👍

MichaIng, 2023-08-10:
Okay looks good. Just to be sure, can you check for kernel errors:
```sh
dmesg -l 0,1,2,3
```

MidG971, 2023-08-10:
Unfortunatelly Installing the additional packages I want still fails. When I log in using the user dietpi I can't intstall them unless I log in using the root user : 
[[ssh 192.168.1.116]  (2023-08-10_232841)  192.168.1.116.log](https://github.com/MichaIng/DietPi/files/12317345/ssh.192.168.1.116.2023-08-10_232841.192.168.1.116.log)

MichaIng, 2023-08-10:
There indeed still I/O errors:
```sh
[ 1424.881162] blk_update_request: I/O error, dev mmcblk1, sector 1591296 op 0x1:(WRITE) flags 0x4000 phys_seg 8 prio class 0
[ 1424.893186] Buffer I/O error on device mmcblk1p1, logical block 194560
[ 1424.893926] Buffer I/O error on device mmcblk1p1, logical block 194561
[ 1424.894564] Buffer I/O error on device mmcblk1p1, logical block 194562
[ 1424.895186] Buffer I/O error on device mmcblk1p1, logical block 194563
[ 1424.895804] Buffer I/O error on device mmcblk1p1, logical block 194564
[ 1424.896421] Buffer I/O error on device mmcblk1p1, logical block 194565
[ 1424.897037] Buffer I/O error on device mmcblk1p1, logical block 194566
[ 1424.897720] Buffer I/O error on device mmcblk1p1, logical block 194567
[ 1424.898362] Buffer I/O error on device mmcblk1p1, logical block 194568
[ 1424.898997] Buffer I/O error on device mmcblk1p1, logical block 194569
[ 1426.519629] blk_update_request: I/O error, dev mmcblk1, sector 562624 op 0x1:(WRITE) flags 0x800 phys_seg 94 prio class 0
[ 1426.521436] Aborting journal on device mmcblk1p1-8.
[ 1476.538115] EXT4-fs error (device mmcblk1p1): ext4_journal_check_start:83: comm systemd: Detected aborted journal
[ 1476.539711] EXT4-fs (mmcblk1p1): Remounting filesystem read-only
```
The last entry shows how the kernel automatically remounts the filesystem read-only because of the errors. I just recognised from your other log that you can `cat /run/initramfs/fsck.log` before doing a reboot, right? Can you please do once again:
```sh
mount -o remount,rw /
touch /forcefsck
```
Then do a `reboot`, and afterwards:
```sh
cat /run/initramfs/fsck.log
```

MidG971, 2023-08-10:
I did as instructed : 

[[ssh 192.168.1.116]  (2023-08-11_000127)  192.168.1.116.log](https://github.com/MichaIng/DietPi/files/12317489/ssh.192.168.1.116.2023-08-11_000127.192.168.1.116.log)

MidG971, 2023-08-10:
I tried to run the install command in the subshell, still fails :

[[ssh 192.168.1.116]  (2023-08-11_001032)  192.168.1.116.log](https://github.com/MichaIng/DietPi/files/12317519/ssh.192.168.1.116.2023-08-11_001032.192.168.1.116.log)

MidG971, 2023-08-10:
I runned dmseg :
`root@DietPi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.93-rockchip64 (root@cf2b4ed83279) (aarch64-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023
[    0.000000] Machine model: Radxa ROCK Pi 4B
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xf77d6100-0xf77d7fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000000f7ffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 128 MiB at 0x00000000e9000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 29 pages/cpu s80984 r8192 d29608 u118784
[    0.000000] pcpu-alloc: s80984 r8192 d29608 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 999432
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: root=UUID=21fbd909-5d4f-4ffe-8ad7-751b251a9f2a rootfstype=ext4 rootwait console=ttyS2,1500000 console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] Memory: 3808696K/4061184K available (16000K kernel code, 2692K rwdata, 6712K rodata, 4352K init, 593K bss, 121416K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 256 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: Distributor has no Range Selector support
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: 16 PPIs implemented
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[    0.000000] ITS [mem 0xfee20000-0xfee3ffff]
[    0.000000] ITS@0x00000000fee20000: allocated 65536 Devices @500000 (flat, esz 8, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] GICv3: using LPI property table @0x00000000004d0000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x00000000004e0000
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[    0.000000] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001913] Console: colour dummy device 80x25
[    0.002917] printk: console [tty1] enabled
[    0.003044] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.003092] pid_max: default: 32768 minimum: 301
[... 606 lines cut ...]
[  411.216073] EXT4-fs warning (device mmcblk1p1): ext4_end_bio:348: I/O error 10 writing to inode 2256 starting block 153856)
[  411.216309] Buffer I/O error on device mmcblk1p1, logical block 149504
[  411.216971] Buffer I/O error on device mmcblk1p1, logical block 149505
[  411.217618] Buffer I/O error on device mmcblk1p1, logical block 149506
[  411.218285] Buffer I/O error on device mmcblk1p1, logical block 149507
[  411.218994] Buffer I/O error on device mmcblk1p1, logical block 149508
[  411.219655] Buffer I/O error on device mmcblk1p1, logical block 149509
[... cut ...]
[ 2546.490803] mmc1: running CQE recovery
[ 2546.491371] blk_update_request: I/O error, dev mmcblk1, sector 562712 op 0x1:(WRITE) flags 0x800 phys_seg 94 prio class 0
[ 2546.492826] Aborting journal on device mmcblk1p1-8.
[ 2591.687194] EXT4-fs error (device mmcblk1p1): ext4_journal_check_start:83: comm dpkg: Detected aborted journal
[ 2591.688771] EXT4-fs (mmcblk1p1): Remounting filesystem read-only
[ 2597.766843] EXT4-fs error (device mmcblk1p1): ext4_remount:5838: comm mount: Abort forced by user
root@DietPi:~#
`

Joulinar, 2023-08-11:
looks like there are still FS errors.

MidG971, 2023-08-11:
I changed my eMMC module. It seems that I can only perform the install then add the optional packages, which fail.

`DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing X.Org X Server: aka X11 - X Window System implementation

[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | Checking URL: https://raw.githubusercontent.com/MichaIng/DietPi/master/.conf/dps_6/98-dietpi-disable_dpms.conf
[ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.conf/dps_6/98-dietpi-disable_dpms.conf -o 98-dietpi-disable_dpms.conf
[ INFO ] DietPi-Software | APT install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-user-session xfonts-base x11-xserver-utils x11-utils, please wait...
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
[FAILED] DietPi-Software | APT install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-user-session xfonts-base x11-xserver-utils x11-utils
 - Command: apt-get -y --allow-change-held-packages install xserver-xorg-core xserver-xorg-input-libinput xinit dbus-user-session xfonts-base x11-xserver-utils x11-utils
root@DietPi:/tmp/DietPi-Software# dpkg --configure -a
Setting up libgomp1:arm64 (12.2.0-14) ...
Setting up libasound2-data (1.2.8-1) ...
Setting up libasound2:arm64 (1.2.8-1+b1) ...
Setting up libsamplerate0:arm64 (0.2.2-3) ...
Setting up libatopology2:arm64 (1.2.8-1+b1) ...
Setting up libfftw3-single3:arm64 (3.3.10-1) ...
Processing triggers for libc-bin (2.36-9+deb12u1) ...
dpkg: error: unable to sync new file '/var/lib/dpkg/status-new': Input/output error
root@DietPi:/tmp/DietPi-Software# touch /forcefsck
root@DietPi:/tmp/DietPi-Software# cat /run/initramfs/fsck.log
Log of fsck -C -a -V -t ext4 /dev/mmcblk1p1
Fri Aug 11 08:25:33 2023

fsck from util-linux 2.38.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk1p1] fsck.ext4 -a -C0 /dev/mmcblk1p1
/dev/mmcblk1p1: recovering journal
/dev/mmcblk1p1 contains a file system with errors, check forced.
/dev/mmcblk1p1: 19571/1871456 files (0.5% non-contiguous), 265037/7629824 blocks
fsck exited with status code 1

Fri Aug 11 08:25:34 2023
----------------
root@DietPi:/tmp/DietPi-Software# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk1      179:0    0 29.1G  0 disk
└─mmcblk1p1  179:1    0 29.1G  0 part /
mmcblk1boot0 179:32   0    4M  1 disk
mmcblk1boot1 179:64   0    4M  1 disk
root@DietPi:/tmp/DietPi-Software# df -hT
Filesystem     Type      Size  Used Avail Use% Mounted on
udev           devtmpfs  1.9G     0  1.9G   0% /dev
tmpfs          tmpfs     387M  5.2M  382M   2% /run
/dev/mmcblk1p1 ext4       29G  559M   27G   2% /
tmpfs          tmpfs     1.9G     0  1.9G   0% /dev/shm
tmpfs          tmpfs     5.0M     0  5.0M   0% /run/lock
tmpfs          tmpfs     1.9G   16K  1.9G   1% /tmp
tmpfs          tmpfs      50M   12K   50M   1% /var/log
`
I did a shutdown / force check : 
`root@DietPi:/tmp/DietPi-Software# shutdown -rF now`

Now there're no errors : 
`root@DietPi:~# cat /run/initramfs/fsck.log
Log of fsck -C -a -V -t ext4 /dev/mmcblk1p1
Fri Aug 11 08:48:08 2023

fsck from util-linux 2.38.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk1p1] fsck.ext4 -a -C0 /dev/mmcblk1p1
/dev/mmcblk1p1: clean, 19580/1871456 files, 265424/7629824 blocks

Fri Aug 11 08:48:08 2023
----------------
root@DietPi:~# dmesg -l 0,1,2,3
[    2.022633] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    3.586340] rk_gmac-dwmac fe300000.ethernet: cannot get clock clk_mac_speed
[    6.501458] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.619580] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.629245] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 25 2015 18:58:57 version 7.45.69 (r581703) FWID 01-24037f6e
root@DietPi:~# apt install apache2
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
root@DietPi:~#
`
Yet I can't add packages. I think the RockPi 4 / 4SE may be too different ...I'll try to use a sd card to eleminate the (new) eMMC out of the equation.

Joulinar, 2023-08-11:
> E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

Package install stuck now due to the issue you faced at the beginning. Therefore try running following to finished pending installations. 

```
dpkg --configure -a
````

MidG971, 2023-08-11:
> > E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
> 
> Package install stuck now due to the issue you faced at the beginning. Therefore try running following to finished pending installations.
> 
> ```
> dpkg --configure -a
> ```

` DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | ALSA will be reinstalled
[ INFO ] DietPi-Software | X.Org X Server will be installed
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
/boot/dietpi/func/dietpi-globals: line 1572:  4982 Segmentation fault      apt-get clean
[ INFO ] DietPi-Software | APT update, please wait...
[FAILED] DietPi-Software | APT update
 - Command: apt-get -y -eany update
Segmentation fault
root@DietPi:~# dpkg --configure -a
dpkg: error: unable to sync new file '/var/lib/dpkg/status-new': Input/output error`

`root@DietPi:~# mount -o remount,rw /
root@DietPi:~# dpkg --configure -a
root@DietPi:~# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[ INFO ] DietPi-Software | ALSA will be reinstalled
[ INFO ] DietPi-Software | X.Org X Server will be installed
[ INFO ] DietPi-Software | Avahi-Daemon will be installed
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
/boot/dietpi/func/dietpi-globals: line 1572:  6167 Segmentation fault      apt-get clean
[ INFO ] DietPi-Software | APT update, please wait...
[FAILED] DietPi-Software | APT update
 - Command: apt-get -y -eany update
Segmentation fault
root@DietPi:~#
`

Joulinar, 2023-08-11:
> Segmentation fault      apt-get clean

> dpkg: error: unable to sync new file '/var/lib/dpkg/status-new': Input/output error`

Quite some data corruption.

MidG971, 2023-08-11:
Indeed : 

root@DietPi:~# dmesg -l 0,1,2,3
[    2.022633] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    3.586340] rk_gmac-dwmac fe300000.ethernet: cannot get clock clk_mac_speed
[    6.501458] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.619580] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.629245] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 25 2015 18:58:57 version 7.45.69 (r581703) FWID 01-24037f6e
[ 1644.594031] blk_update_request: I/O error, dev mmcblk1, sector 769536 op 0x1:(WRITE) flags 0x800 phys_seg 48 prio class 0
[ 1644.595321] Buffer I/O error on device mmcblk1p1, logical block 92096
[ 1644.595937] Buffer I/O error on device mmcblk1p1, logical block 92097
[ 1644.596523] Buffer I/O error on device mmcblk1p1, logical block 92098
[ 1644.597108] Buffer I/O error on device mmcblk1p1, logical block 92099
[ 1644.597692] Buffer I/O error on device mmcblk1p1, logical block 92100
[ 1644.598275] Buffer I/O error on device mmcblk1p1, logical block 92101
[ 1644.598882] Buffer I/O error on device mmcblk1p1, logical block 92102
[ 1644.599467] Buffer I/O error on device mmcblk1p1, logical block 92103
[ 1644.600051] Buffer I/O error on device mmcblk1p1, logical block 92104
[ 1644.600636] Buffer I/O error on device mmcblk1p1, logical block 92105

MidG971, 2023-08-11:
> > Segmentation fault      apt-get clean
> 
> > dpkg: error: unable to sync new file '/var/lib/dpkg/status-new': Input/output error`
> 
> Quite some data corruption. If possible, start with a fresh clean image and don't try to repair the old one.

That's what I did...

MidG971, 2023-08-11:
After another shutdown / check :  

`root@DietPi:~# dmesg -l 0,1,2,3
[    2.030692] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    3.592963] rk_gmac-dwmac fe300000.ethernet: cannot get clock clk_mac_speed
[    6.600679] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.718642] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.738288] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 25 2015 18:58:57 version 7.45.69 (r581703) FWID 01-24037f6e
root@DietPi:~#
`
When I google brclfmac , I get these two first results :

https://github.com/reMarkable/brcmfmac-firmware/blob/master/brcmfmac43455-sdio.txt

https://github.com/armbian/firmware/blob/master/brcm/brcmfmac43455-sdio.txt

MichaIng, 2023-08-11:
The `brcmfmac` errors are unrelated. It tells you which firmware the driver uses. The unexpected/problematic errors are the ones starting at 1644s of your previous log.

> I think the RockPi 4 / 4SE may be too different

Unlikely, but I cannot rule it out since I do not remember someone else explicitly using a ROCK 4 SE. However, at least Armbian explicitly states the ROCK 4B image to be compatible with SE, and they all use the same kernel package anyway. However, let's see whether it picked the correct device tree:
```sh
cat /proc/device-tree/model
```

You said you tested it with two different eMMC modules already? And did you even ran into the same error with an SD card already? I hope there is not a hardware issue with the eMMC slot or MMC controller. Just to rule it out, do you have an alternative/stronger PSU? Probably it is a little low on voltage when being on load (package installs etc). As alternative, you could try to switch to `powersave` CPU governor via `dietpi-config` > Performance Options?

MidG971, 2023-08-11:
Yes, I tried with two differents eMMC modules, one was loaded with Debian , the other one with Ubuntu.. till now I had no problems with either of them. I can't test it with a SD card right now. I once copied the distro from the SD card to the eMMC before I bought a USB adapter.

Here's the output. I can't change the CPU governor. I changed the PSU, both QC compatible, if it fails I'll use my last one : VOOC ratings.

root@DietPi:~# cat /proc/device-tree/model
Radxa ROCK Pi 4Broot@DietPi:~# dietpi-config
grep: program error
/boot/dietpi/dietpi-config: line 1268:  3107 Aborted                 grep -qi 'genuineintel' /proc/cpuinfo
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3140 Aborted                 grep -qi 'schedutil' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3141 Aborted                 grep -qi 'ondemand' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3142 Aborted                 grep -qi 'interactive' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3143 Aborted                 grep -qi 'conservative' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3144 Aborted                 grep -qi 'powersave' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3145 Aborted                 grep -qi 'performance' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3146 Aborted                 grep -qi 'userspace' "$fp_governor"
grep: stack overflow
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3199 Aborted                 grep -qi 'schedutil' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3200 Aborted                 grep -qi 'ondemand' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3201 Aborted                 grep -qi 'interactive' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3202 Aborted                 grep -qi 'conservative' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3203 Aborted                 grep -qi 'powersave' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3204 Aborted                 grep -qi 'performance' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3205 Aborted                 grep -qi 'userspace' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1268:  3230 Aborted                 grep -qi 'genuineintel' /proc/cpuinfo
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3251 Aborted                 grep -qi 'schedutil' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3252 Aborted                 grep -qi 'ondemand' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3253 Aborted                 grep -qi 'interactive' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3254 Aborted                 grep -qi 'conservative' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3255 Aborted                 grep -qi 'powersave' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3256 Aborted                 grep -qi 'performance' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  3257 Aborted                 grep -qi 'userspace' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1268:  3276 Aborted                 grep -qi 'genuineintel' /proc/cpuinfo
grep: program error
/boot/dietpi/dietpi-config: line 1268:  3390 Aborted                 grep -qi 'genuineintel' /proc/cpuinfo
root@DietPi:~#

MichaIng, 2023-08-11:
Okay, `grep` is already broken. Best is probably to test with SD card first. If you want to try with a fresh image on eMMC again, before you start the first run installs, go to `dietpi-config` (at the top of the menu) and advanced settings, to change the CPU governor to powersave before starting installs.

The ROCK (Pi) 4B device tree was selected correctly at least. Just to rule out that there is really an issue with the kernel, I could recreate the image with a newer one. Armbian did not update their APT repository since February for unknown reasons. But their images ship with recent kernel builds + packages can be built easily with their tools (that's what Armbian is actually about, the build system). So newer packages can be either scraped from their images or built. But I'd need some days for this, as I'm busy with some other things.

MidG971, 2023-08-11:
I managed to use the powersave setting. I'll do another  clean install. 
It's ok take your time to do your things, I can still use one of my eMMC with another distro.

`[  OK  ] DietPi-Config | Setting in /boot/dietpi.txt adjusted: CONFIG_CPU_GOVERNOR=powersave
[ SUB1 ] DietPi-CPU_set > Applying CPU governor settings: powersave
[ INFO ] DietPi-CPU_set | Setting CPU frequency limits : Max = Disabled MHz | Min = Disabled MHz
[  OK  ] DietPi-CPU_set | Applied CPU governor settings: powersave
grep: program error
/boot/dietpi/dietpi-config: line 1268:  1882 Aborted                 grep -qi 'genuineintel' /proc/cpuinfo
grep: program error
/boot/dietpi/dietpi-config: line 1379:  1915 Aborted                 grep -qi 'schedutil' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  1916 Aborted                 grep -qi 'ondemand' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  1917 Aborted                 grep -qi 'interactive' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  1918 Aborted                 grep -qi 'conservative' "$fp_governor"
grep: stack overflow
grep: program error
/boot/dietpi/dietpi-config: line 1379:  1920 Aborted                 grep -qi 'performance' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  1921 Aborted                 grep -qi 'userspace' "$fp_governor"
[  OK  ] DietPi-Config | Desired setting in /boot/dietpi.txt was already set: CONFIG_CPU_GOVERNOR=powersave
[ SUB1 ] DietPi-CPU_set > Applying CPU governor settings: powersave
[ INFO ] DietPi-CPU_set | Setting CPU frequency limits : Max = Disabled MHz | Min = Disabled MHz
[  OK  ] DietPi-CPU_set | Applied CPU governor settings: powersave
grep: program error
/boot/dietpi/dietpi-config: line 1268:  1974 Aborted                 grep -qi 'genuineintel' /proc/cpuinfo
grep: program error
/boot/dietpi/dietpi-config: line 1379:  2007 Aborted                 grep -qi 'schedutil' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  2008 Aborted                 grep -qi 'ondemand' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  2009 Aborted                 grep -qi 'interactive' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  2010 Aborted                 grep -qi 'conservative' "$fp_governor"
grep: stack overflow
grep: program error
/boot/dietpi/dietpi-config: line 1379:  2012 Aborted                 grep -qi 'performance' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1379:  2013 Aborted                 grep -qi 'userspace' "$fp_governor"
grep: program error
/boot/dietpi/dietpi-config: line 1268:  2053 Aborted                 grep -qi 'genuineintel' /proc/cpuinfo
root@DietPi:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     49 °C / 120 °F : Optimal temperature
 Governor     |     powersave

                 Current Freq    Min Freq   Max Freq
 CPU0         |      408 MHz      408 MHz    1416 MHz
 CPU1         |      408 MHz      408 MHz    1416 MHz
 CPU2         |      408 MHz      408 MHz    1416 MHz
 CPU3         |      408 MHz      408 MHz    1416 MHz
 CPU4         |      408 MHz      408 MHz    1800 MHz
 CPU5         |      408 MHz      408 MHz    1800 MHz

[ INFO ] DietPi-CPU_info | The current CPU frequency may be affected by processing this script itself.

root@DietPi:~# cat /run/initramfs/fsck.log
Log of fsck -C -a -V -t ext4 /dev/mmcblk1p1
Fri Aug 11 21:15:48 2023

fsck from util-linux 2.38.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk1p1] fsck.ext4 -a -C0 /dev/mmcblk1p1
/dev/mmcblk1p1: clean, 19557/1871456 files, 265355/7629824 blocks
[... 28 lines cut ...]
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
/boot/dietpi/func/dietpi-globals: line 1572:  2963 Segmentation fault      apt-get clean
[ INFO ] DietPi-Software | APT update, please wait...
[FAILED] DietPi-Software | APT update
 - Command: apt-get -y -eany update
Segmentation fault
root@DietPi:~# dmesg -l 0,1,2,3
[    2.026689] rockchip-pcie f8000000.pcie: PCIe link training gen1 timeout!
[    3.624527] rk_gmac-dwmac fe300000.ethernet: cannot get clock clk_mac_speed
[    6.674797] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.802095] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.811853] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Aug 25 2015 18:58:57 version 7.45.69 (r581703) FWID 01-24037f6e
root@DietPi:~# cat /run/initramfs/fsck.log
Log of fsck -C -a -V -t ext4 /dev/mmcblk1p1
Fri Aug 11 21:15:48 2023

fsck from util-linux 2.38.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk1p1] fsck.ext4 -a -C0 /dev/mmcblk1p1
/dev/mmcblk1p1: clean, 19557/1871456 files, 265355/7629824 blocks

Fri Aug 11 21:15:48 2023
----------------
root@DietPi:~#
`

MichaIng, 2023-08-11:
> I'll do another clean install.

Jep, the current system has `grep` and `apt-get` already broken, so this won't be easy or even possible to repair.

MidG971, 2023-08-11:
So, clean install, new PSU, powersave governor. Still fails because presumably because of the eMMC. I'll try to get my hands on a SD card to test. I'll keep you posted. 

Here's the log file : 
[[ssh 192.168.1.116]  (2023-08-11_225414)  192.168.1.116.log](https://github.com/MichaIng/DietPi/files/12325566/ssh.192.168.1.116.2023-08-11_225414.192.168.1.116.log)

I'll keep you posted. Thanks for what you've done so far 😀

MidG971, 2023-08-12:
So clean install, I used your link [https://dietpi.com/docs/hardware/#make-your-own-distribution](url) to pass the RockPi as a generic board. It worked fine until I got to the last part where the board doesn't detect the eth0 or wlan0 card. I'll retry later this day, so that you can have the log file.  I used the uart / gpio to access the board.

MidG971, 2024-02-28:
There's now a  custom ROCK 4SE image 😎

closed 2024-02-28 by MidG971

MichaIng, 2024-02-28:
Ah right, I hope this solves the nasty `grep` and `apt-get` errors for you, respectively supports eMMC properly.