MichaIng/DietPi #6537: Odd permission problem with backup on NFS

surtarso, 2023-08-09:
#### Details:
- Date           | Wed Aug  9 11:27:59 -03 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux SurServer 6.2.6-surserver-xeon #1 SMP PREEMPT_DYNAMIC Tue Mar 14 13:42:23 -03 2023 x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `mkdir -p /mnt/nfs_client/dietpi-backup`
- Exit code      | 1
- Software title | DietPi-Backup
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- It should be able to create the folder it needs to start the backup
#### Actual behaviour:

- mkdir command fails. 
#### Extra details:

- If I quit dietpi-backup and use the same command manually it works (as dietpi user), but when I try to add the DIR to the dietpi-backup UI it cannot create the directory as you can see bellow. I simply run 'sudo dietpi-backup' as dietpi user.
#### Additional logs:
SurServer ~$ mkdir -p /mnt/nfs_client/test
drwxr-xr-x   2 dietpi dietpi    6 Aug  9  2023 test

mkdir: cannot create directory ‘/mnt/nfs_client/dietpi-backup’: Permission denied
```

Joulinar, 2023-08-09:
I guess you need to add `no_root_squash` option to your NFS export on server side.

surtarso, 2023-08-09:
ahh... I see.. so it was trying to mkdir with nouser:nogroup.
That did the trick, thanks!