MichaIng/DietPi #6540: Pi-hole Update Failed - Failed to connect to install.pi-hole.net port 443: No route to host

LazerDickMcCheese, 2023-08-11:
This is a crosspost of this report: https://github.com/pi-hole/FTL/issues/1619#issue-1837909225

#### Details:
- Date           | Fri Aug 11 02:19:24 BST 2023
- DietPi version | v8.13.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | from scratch
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `ping -4nc 1 -W 10 9.9.9.9`
- Exit code      | 1
- Software title | DietPi-Update
- 
#### Steps to reproduce:
1. Install Pi-hole
2. Wait 3 years
3. Try to update

#### Expected behaviour:
-Use an update command to update the software
-"Nice"

#### Actual behaviour:
-Use every update command variant you can find on forums
-No updates
-"Failed to connect to install.pi-hole.net port 443: No route to host"

#### Extra details:
If any command lines could fix this, please let me know. I'm very close to nuking this whole install and starting over (even though it's very frustrating as I recall).

Joulinar, 2023-08-11:
Do you use static or DHCP IP address? If static, change DietPi configuration and set local DNS to a global public one. This has no impact to Pihole or how you LAN clients will resolve their DNS request. It's only for DietPi to be able to connect to internet.

LazerDickMcCheese, 2023-08-11:
Thank you for the response. Forgive my ignorance, but is this something I can change from the web dashboard? If so, can I get dummy-proof instructions?

Joulinar, 2023-08-11:
No this setting has nothing to do with Pihole. Please go to command line and open `dietpi-config` > `7  : Network Options: Adapters` > `Ethernet :`. What are the actual settings?

LazerDickMcCheese, 2023-08-11:
The IP is set to static, I changed it to DHCP. Updating still failed: "Retrieval of supported OS list failed. dig failed with return code 10. Unable to determine if the detected OS (Debian 11) is supported"

I used the handy skip OS check, but it still failed: "Error: URL https//..." It says FTL Engine not installed, but isn't that a required part of the base install?

Joulinar, 2023-08-11:
keep it on STATIC and change DNS entry to something like `8.8.8.8`, `9.9.9.9` or `1.1.1.1`. Should looks like this

```
               ●─ DHCP/STATIC IP ──────────────────────●
Change Mode    : [STATIC]                               
Copy           : Copy current address to "Static"       
Static IP      : [192.168.x.x]                        
Static Mask    : [255.255.255.0]                        
Static Gateway : [192.168.x.1]                        
Static DNS     : [9.9.9.9 149.112.112.112]              
               ●─ Additional Options ──────────────────●
Link Speed     : [auto (default)]                       
Disable        : Disable Ethernet adapter               
               ●─ Apply ───────────────────────────────●
Apply          : Save all changes and restart networking
                                                        
                                                        
   <Ok>                                      <Back>     
```

LazerDickMcCheese, 2023-08-11:
I tried to set the static DNS, but it keeps removing my selection...I know the Pi-hole team is doing everything they can for the user, but this is the opposite of fun haha. I'm running v5.15.3, and thank you again for your help.

The command you listed pumped out a lot of "failure resolving" and "failed to fetch" errors. Anything else I should try, or should I just burn it to the ground and try to start from scratch?

Joulinar, 2023-08-11:
are you sure you set STATIC IP?? Can you share following

```
ip a
cat /etc/resolv.conf
```

LazerDickMcCheese, 2023-08-11:
[image]

Joulinar, 2023-08-11:
ok you have `dhcpcd` installed. Means your DNS server is still obtained via DHCP. Pls removed `dhcpcd`,

```
G_AGP dhcpcd5
```

LazerDickMcCheese, 2023-08-11:
"command not found" is there another command?

This won't affect my ability to force all traffic through the pi-hole, correct?

Joulinar, 2023-08-11:
`apt purge dhcpcd5`

LazerDickMcCheese, 2023-08-11:
I removed dhcpcd5, but pi-hole still fails to update. It detects that there is an update to the FTL, but it erroneously says the core and web UI are up to date.

Joulinar, 2023-08-11:
can you share the entire error message from running `pihole -up`

Best is to connect via SSH and copy/past the output. Way better compare to screen prints 😉

LazerDickMcCheese, 2023-08-11:
So I've unsuccessfully tried getting a copy/paste of my SSH many times, but I accidentally just figured it out. Here ya go: 

dietpi@DietPi:~$ sudo pihole -up
  [✓] Update local cache of available packages
  [i] Existing PHP installation detected : PHP version 7.4.33
  [✓] Checking for git
  [✓] Checking for iproute2
  [✓] Checking for dialog
  [✓] Checking for ca-certificates

  [i] Checking for updates...
fatal: unable to access 'https://github.com/pi-hole/pi-hole.git/': Could not resolve host: github.com
  [i] Pi-hole Core:     up to date
fatal: unable to access 'https://github.com/pi-hole/AdminLTE.git/': Could not resolve host: github.com
  [i] Web Interface:    up to date
  [i] FTL:              update available

  [i] FTL out of date, it will be updated by the installer.

  [✓] Root user check

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
             .;cll.;;looo:.
                 .. ','.
                .',,,,,,'.
              .',,,,,,,,,,.
            .',,,,,,,,,,,,....
          ....''',,,,,,,'.......
        .........  ....  .........
        ..........      ..........
        ..........      ..........
        .........  ....  .........
          ........,,,,,,,'......
            ....',,,,,,,,,,,,.
               .',,,,,,,,,'.
                .',,,,,,'.
                  ..'''.

  [i] SELinux not detected
  [✓] Update local cache of available packages
  [i] Existing PHP installation detected : PHP version 7.4.33

  [✓] Checking apt-get for upgraded packages... 59 updates available
  [i] It is recommended to update your OS after installing the Pi-hole!

  [i] Checking for / installing Required dependencies for OS Check...
  [✓] Checking for grep
  [✓] Checking for dnsutils

  [✗] Retrieval of supported OS list failed. dig failed with return code 10.
      Unable to determine if the detected OS (Debian 11) is supported
      Possible causes for this include:
        - Firewall blocking certain DNS lookups from Pi-hole device
        - ns1.pi-hole.net being blocked (required to obtain TXT record from versions.pi-hole.net containing supported operating systems)
        - Other internet connectivity issues

      https://docs.pi-hole.net/main/prerequisites/#supported-operating-systems

      If you wish to attempt to continue anyway, you can try one of the following commands to skip this check:

      e.g: If you are seeing this message on a fresh install, you can run:
             curl -sSL https://install.pi-hole.net | sudo PIHOLE_SKIP_OS_CHECK=true bash

           If you are seeing this message after having run pihole -up:
             sudo PIHOLE_SKIP_OS_CHECK=true pihole -r
           (In this case, your previous run of pihole -up will have already updated the local repository)

      It is possible that the installation will still fail at this stage due to an unsupported configuration.
      If that is the case, you can feel free to ask the community on Discourse with the Community Help category:
      https://discourse.pi-hole.net/c/bugs-problems-issues/community-help/

  Unable to complete update, please contact Pi-hole Support

Joulinar, 2023-08-11:
can you try following as stated on the error message

`sudo PIHOLE_SKIP_OS_CHECK=true pihole -r` > select `Repair`

MichaIng, 2023-08-11:
There still seems to be an issue with the host DNS resolution. What which DNS provider is now used?
```sh
cat /etc/resolv.conf
```

LazerDickMcCheese, 2023-08-11:
Hello, MichaIng! Per your request:

dietpi@DietPi:~$ cat /etc/resolv.conf
# Generated by dhcpcd
# /etc/resolv.conf.head can replace this line
# /etc/resolv.conf.tail can replace this line

MichaIng, 2023-08-11:
Okay none at all, looks like dhcpcd leaves it empty on uninstall 😄. So either as Joulinar said, you can re-apply static IP settings in `dietpi-config`, which also re-sets the DNS server, or just:
```sh
echo 'nameserver 9.9.9.9' > /etc/resolv.conf
```

LazerDickMcCheese, 2023-08-11:
So the command came back with permission denied despite using sudo, but I could set it through the config. The update still fails however

dietpi@DietPi:~$ sudo dietpi-config
[.     ] DietPi-WiFiDB | eval > /var/lib/dietpi/dietpi-wifi.db
[  OK  ] DietPi-WiFiDB | eval > /var/lib/dietpi/dietpi-wifi.db
[ INFO ] DietPi-WiFiDB | Applied WiFi DB slot 0 with SSID ""
[ INFO ] DietPi-WiFiDB | Applied WiFi DB slot 1 with SSID ""
[ INFO ] DietPi-WiFiDB | Applied WiFi DB slot 2 with SSID ""
[ INFO ] DietPi-WiFiDB | Applied WiFi DB slot 3 with SSID ""
[ INFO ] DietPi-WiFiDB | Applied WiFi DB slot 4 with SSID ""
[  OK  ] DietPi-WiFiDB | chown root:root /var/lib/dietpi/dietpi-wifi.db /etc/wpa             _supplicant/wpa_supplicant.conf
[  OK  ] DietPi-WiFiDB | chmod 600 /var/lib/dietpi/dietpi-wifi.db /etc/wpa_suppl             icant/wpa_supplicant.conf
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.4-fpm
[ SUB1 ] DietPi-Set_hardware > wifimodules (disable)
[  OK  ] DietPi-Set_hardware | Desired setting in /boot/config.txt was already s             et: dtoverlay=disable-wifi
[  OK  ] wifimodules disable | Completed
[ INFO ] DietPi-Config | Dropping network connections, please wait...
RTNETLINK answers: No such process
[ INFO ] DietPi-Config | Restarting network connections, please wait...
[  OK  ] DietPi-Config | systemctl daemon-reload
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : php7.4-fpm
[  OK  ] DietPi-Services | start : lighttpd
[  OK  ] DietPi-Services | start : cron
[ INFO ] DietPi-Config | Reloading networking data, please wait...
[  OK  ] DietPi-Config | Network restarted
Device "link" does not exist.
dietpi@DietPi:~$ pihole -up
  [✓] Update local cache of available packages
  [i] Existing PHP installation detected : PHP version 7.4.33
  [✓] Checking for git
  [✓] Checking for iproute2
  [✓] Checking for dialog
  [✓] Checking for ca-certificates

  [i] Checking for updates...
fatal: unable to access 'https://github.com/pi-hole/pi-hole.git/': Could not res             olve host: github.com
  [i] Pi-hole Core:     up to date
fatal: unable to access 'https://github.com/pi-hole/AdminLTE.git/': Could not re             solve host: github.com
  [i] Web Interface:    up to date
  [i] FTL:              update available

  [i] FTL out of date, it will be updated by the installer.

  [✓] Root user check

        .;;,.
        .ccccc:,.
         :cccclll:.      ..,,
          :ccccclll.   ;ooodc
           'ccll:;ll .oooodc
             .;cll.;;looo:.
                 .. ','.
                .',,,,,,'.
              .',,,,,,,,,,.
            .',,,,,,,,,,,,....
          ....''',,,,,,,'.......
        .........  ....  .........
        ..........      ..........
        ..........      ..........
        .........  ....  .........
          ........,,,,,,,'......
            ....',,,,,,,,,,,,.
               .',,,,,,,,,'.
                .',,,,,,'.
                  ..'''.

  [i] SELinux not detected
  [✓] Update local cache of available packages
  [i] Existing PHP installation detected : PHP version 7.4.33

  [✓] Checking apt-get for upgraded packages... 59 updates available
  [i] It is recommended to update your OS after installing the Pi-hole!

  [i] Checking for / installing Required dependencies for OS Check...
  [✓] Checking for grep
  [✓] Checking for dnsutils

  [✗] Retrieval of supported OS list failed. dig failed with return code 10.
      Unable to determine if the detected OS (Debian 11) is supported
      Possible causes for this include:
        - Firewall blocking certain DNS lookups from Pi-hole device
        - ns1.pi-hole.net being blocked (required to obtain TXT record from vers             ions.pi-hole.net containing supported operating systems)
        - Other internet connectivity issues

      https://docs.pi-hole.net/main/prerequisites/#supported-operating-systems

      If you wish to attempt to continue anyway, you can try one of the followin             g commands to skip this check:

      e.g: If you are seeing this message on a fresh install, you can run:
             curl -sSL https://install.pi-hole.net | sudo PIHOLE_SKIP_OS_CHECK=t             rue bash

           If you are seeing this message after having run pihole -up:
             sudo PIHOLE_SKIP_OS_CHECK=true pihole -r
           (In this case, your previous run of pihole -up will have already upda             ted the local repository)

      It is possible that the installation will still fail at this stage due to              an unsupported configuration.
      If that is the case, you can feel free to ask the community on Discourse w             ith the Community Help category:
      https://discourse.pi-hole.net/c/bugs-problems-issues/community-help/

  Unable to complete update, please contact Pi-hole Support

Joulinar, 2023-08-11:
You need to use `root` user. Before running any update, we need to fix the initial DNS issue. Can you poste again `cat /etc/resolv.conf`

LazerDickMcCheese, 2023-08-11:
dietpi@DietPi:~$ cat /etc/resolv.conf
nameserver 9.9.9.9
nameserver 149.112.112.112

Joulinar, 2023-08-11:
ok this is fine, how about following to check DNS resolution

```
dig github.com
```

LazerDickMcCheese, 2023-08-11:
dietpi@DietPi:~$ dig github.com

; <<>> DiG 9.16.37-Debian <<>> github.com
;; global options: +cmd
;; connection timed out; no servers could be reached

Joulinar, 2023-08-11:
> connection timed out

Hmm Quad9 DNS is not reachable for you. This is indicating a local network issue or some blocking by your ISP. Can you try Google DNS

```
dig @8.8.8.8 github.com
```

LazerDickMcCheese, 2023-08-11:
dietpi@DietPi:~$ dig @8.8.8.8 github.com

; <<>> DiG 9.16.37-Debian <<>> @8.8.8.8 github.com
; (1 server found)
;; global options: +cmd
;; connection timed out; no servers could be reached

Joulinar, 2023-08-11:
ok why kind of DNS servers you have set inside PiHole GUI as upstream DNS?

LazerDickMcCheese, 2023-08-11:
I disabled my upstream DNSs, but I get the same error.

Joulinar, 2023-08-11:
But how does your local clients (e.g. your computer) do DNS resolution? I thought they use PiHole?

MichaIng, 2023-08-11:
> So the command came back with permission denied despite using sudo, but I could set it through the config.

Ah right, something like `sudo command > file` does not work, as only the command gets sudo permissions while the redirect of its output to the file has not. As non-root user, this works instead:
```sh
echo 'nameserver 9.9.9.9' | tee /etc/resolv.conf
```
However, the `/etc/resolv.conf` content is fine now. Indeed it seems that any (other) DNS requests from the host are blocked. Probably you added a related firewall rule to your router to prevent clients on your network from bypassing Pi-hole?

... Joulinar is asking the right questions. If Pi-hole is successfully using an upstream DNS, then the host system should be able to use the same directly.

LazerDickMcCheese, 2023-08-11:
So my goal with the Pi-hole is to:
-force all traffic though it (which it does)
-block as much unnecessary stuff as possible (don't get me started about the adlists) (which it does)
-keep itself updated (which I recently found out it does not anymore)
-remote access for my phone (which I've tried at least six time since 2019, I think...pivpn has never worked for me, I'll be trying Tailscale if this is figured out)

But this is confusing at best, demoralizing at worst. Wouldn't I have saved time at this point if I just scrapped this and tried to find a Windows alternative?

Joulinar, 2023-08-11:
We understood your aim and I'm using the exact same setup. BUT there is a huge BUT: there is something on your network blocking DNS access for the DietPi system. Therefore, our question (and it would be helpful if you can answer them, as it is frustrating for us not getting answers on our questions), how does your local clients like your Windows computer, do DNS resolution? Which DNS server they use?

LazerDickMcCheese, 2023-08-11:
Joulinar, how do I check that?

Joulinar, 2023-08-11:
open a Windows `cmd` and run `ipconfig -all`

LazerDickMcCheese, 2023-08-11:
The DNS server listed is my pi-hole's IP

Joulinar, 2023-08-11:
and what DNS you have set inside PiHole GUI? (A question I already ask once)

LazerDickMcCheese, 2023-08-11:
I'm pretty sure I was using Cloudflare, but they are all blank now

ravenclaw900, 2023-08-11:
Sorry to interrupt, but judging from the fact that `ping` failed in the issue description, is it possible that this could be more than just a DNS issue? Does the RPi have access to the internet at all?
```
ping -c 5 1.1.1.1
```

Joulinar, 2023-08-11:
> but they are all blank now

That's not possible (usually). You need to have something selected. Otherwise, it would trough an error inside PiHole Gui.

Inside PiHole, can go to `Query Log` and check on `Status` column. There should be an indication on the DNS server used.

MichaIng, 2023-08-11:
@ravenclaw900 
You are absolutely right. Even that the ping target is a DNS provider, it is not a DNS request.

@LazerDickMcCheese 
This fails as well?
```sh
curl -IL 173.194.76.138
```
It is a Google IP.

I see the `tun0` interface. Is this from your PiVPN installation? And did you already install Tailscale? Probably something is messing with the routes:
```sh
ip r
```

LazerDickMcCheese, 2023-08-11:
dietpi@DietPi:~$ ping -c 5 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
From 169.254.223.131 icmp_seq=1 Destination Host Unreachable
From 169.254.223.131 icmp_seq=2 Destination Host Unreachable
From 169.254.223.131 icmp_seq=3 Destination Host Unreachable
From 169.254.223.131 icmp_seq=4 Destination Host Unreachable
From 169.254.223.131 icmp_seq=5 Destination Host Unreachable

--- 1.1.1.1 ping statistics ---
5 packets transmitted, 0 received, +5 errors, 100% packet loss, time 4075ms
pipe 3

dietpi@DietPi:~$ curl -IL 173.194.76.138
curl: (7) Failed to connect to 173.194.76.138 port 80: No route to host

I have not installed Tailscale yet, one nightmarish issue at a time haha

Joulinar, 2023-08-11:
Can you share `ip r` as well. Do you use a VPN? What about PiHole Query log?

LazerDickMcCheese, 2023-08-11:
Joulinar, sorry for not doing that sooner, I missed your message.

dietpi@DietPi:~$ ip r
default dev eth0 scope link src 169.254.223.131 metric 202
10.86.211.0/24 dev tun0 proto kernel scope link src 10.86.211.1
169.254.0.0/16 dev eth0 scope link src 169.254.223.131 metric 202
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.5

And I misspoke, I had the upstream DNS previously set as my VPN

Joulinar, 2023-08-11:
can you try following as your default route is some incorrect

```
sudo ip route del default via 169.254.223.131
sudo ip route add default via 192.168.1.1
sudo ip r
ping -c 5 1.1.1.1
```

MichaIng, 2023-08-11:
Looks like something assigned a local-link address network + route to this interface. Very strange. Probably some leftover from dhcpcd, and that it started doing something was due to a package update or so?

After you verified that Joulinar's steps help, can you show the output of this:
```sh
cat /etc/network/interfaces /etc/network/interfaces.d/*
systemctl | grep '\.service '
```
This is to see all loaded services, to see whether there are any other network tools interfering, and the network configuration, to see whether there is any unexpected content.

LazerDickMcCheese, 2023-08-11:
@Joulinar I tried those commands, 100% packet loss. Please see the console output below;

dietpi@DietPi:~$ sudo ip route del default via 169.254.223.131
RTNETLINK answers: No such process
dietpi@DietPi:~$ sudo ip route add default via 192.168.1.1
dietpi@DietPi:~$ sudo ip r
default via 192.168.1.1 dev eth0
default dev eth0 scope link src 169.254.223.131 metric 202
10.86.211.0/24 dev tun0 proto kernel scope link src 10.86.211.1
169.254.0.0/16 dev eth0 scope link src 169.254.223.131 metric 202
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.5
dietpi@DietPi:~$ ping -c 5 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
From 192.168.1.5 icmp_seq=1 Destination Host Unreachable
From 192.168.1.5 icmp_seq=2 Destination Host Unreachable
From 192.168.1.5 icmp_seq=3 Destination Host Unreachable
From 192.168.1.5 icmp_seq=4 Destination Host Unreachable
From 192.168.1.5 icmp_seq=5 Destination Host Unreachable

--- 1.1.1.1 ping statistics ---
5 packets transmitted, 0 received, +5 errors, 100% packet loss, time 4066ms
pipe 3

@MichaIng Please see the console output to your commands below:

dietpi@DietPi:~$ cat /etc/network/interfaces /etc/network/interfaces.d/*
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.1.5
netmask 255.255.255.0
gateway 192.168.1.254
#dns-nameservers 9.9.9.9 149.112.112.112

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
cat: '/etc/network/interfaces.d/*': No such file or directory
dietpi@DietPi:~$ systemctl | grep '\.service'
Failed to connect to bus: No such file or directory

PS: your team has insanely good patience

Joulinar, 2023-08-11:
Ah ok your gateway IP address is different than I expected. Do you know the IP address of your internet router? Is it 192.168.1.254? On your Windows system, open a `cmd` and run `ipconfig -all` to check the gateway address.

MichaIng, 2023-08-12:
Also the other default route was not correctly removed, as the IP was the source address, not the gateway. This should work:
```sh
sudo ip r del default dev eth0 src 169.254.223.131
```
I forgot `sudo` for the second command
```sh
sudo systemctl | grep '\.service '
```
If there is not interfering other network tool left and `192.168.1.254` indeed is the correct gateway, easiest is probably to reboot the system to restore intended interface settings. Or bringing down and up the interface once:
```sh
sudo ifdown eth0
sudo ifup eth0
```

LazerDickMcCheese, 2023-08-12:
@MichaIng Here's the results of that one;

dietpi@DietPi:~$ systemctl | grep '\.service'
Failed to connect to bus: No such file or directory
-bash: dietpi@DietPi:~$: command not found
-bash: interfaces.d/*: No such file or directory
-bash: allow-hotplug: command not found
-bash: iface: command not found
-bash: address: command not found
-bash: netmask: command not found
-bash: gateway: command not found
-bash: iface: command not found
-bash: address: command not found
-bash: netmask: command not found
-bash: gateway: command not found
-bash: wpa-conf: command not found
-bash: cat:: command not found
-bash: dietpi@DietPi:~$: command not found
-bash: Failed: command not found
dietpi@DietPi:~$ sudo ip r del default dev eth0 src 169.254.223.131
dietpi@DietPi:~$ sudo systemctl | grep '\.service '
  console-setup.service                                                                        loaded    active exited    Set console font and keymap
  cron.service                                                                                 loaded    active running   Regular background program processing daemon
● dhcpcd.service                                                                               not-found active running   dhcpcd.service
  dietpi-postboot.service                                                                      loaded    active exited    DietPi-PostBoot
  dietpi-preboot.service                                                                       loaded    active exited    DietPi-PreBoot
  dietpi-ramlog.service                                                                        loaded    active exited    DietPi-RAMlog
  dietpi-vpn.service                                                                           loaded    active running   VPN Client (DietPi)
  dropbear.service                                                                             loaded    active running   LSB: Lightweight SSH server
  fake-hwclock.service                                                                         loaded    active exited    Restore / save the current clock
  getty@tty1.service                                                                           loaded    active running   Getty on tty1
  ifup@eth0.service                                                                            loaded    active exited    ifup for eth0
  ifupdown-pre.service                                                                         loaded    active exited    Helper to synchronize boot up for ifupdown
  keyboard-setup.service                                                                       loaded    active exited    Set the console keyboard layout
  kmod-static-nodes.service                                                                    loaded    active exited    Create list of static device nodes for the current kernel
  lighttpd.service                                                                             loaded    active running   Lighttpd Daemon
  netfilter-persistent.service                                                                 loaded    active exited    netfilter persistent configuration
  networking.service                                                                           loaded    active exited    Raise network interfaces
  openvpn.service                                                                              loaded    active exited    OpenVPN service
  openvpn@server.service                                                                       loaded    active running   OpenVPN connection to server
  php7.4-fpm.service                                                                           loaded    active running   The PHP 7.4 FastCGI Process Manager
  pihole-FTL.service                                                                           loaded    active running   Pi-hole FTL
  rngd.service                                                                                 loaded    active running   Start entropy gathering daemon (rngd)
  systemd-fsck-root.service                                                                    loaded    active exited    File System Check on Root Device
  systemd-fsck@dev-disk-by\x2dpartuuid-a97c7839\x2d01.service                                  loaded    active exited    File System Check on /dev/disk/by-partuuid/a97c7839-01
  systemd-journal-flush.service                                                                loaded    active exited    Flush Journal to Persistent Storage
  systemd-journald.service                                                                     loaded    active running   Journal Service
  systemd-modules-load.service                                                                 loaded    active exited    Load Kernel Modules
  systemd-random-seed.service                                                                  loaded    active exited    Load/Save Random Seed
  systemd-remount-fs.service                                                                   loaded    active exited    Remount Root and Kernel File Systems
  systemd-sysctl.service                                                                       loaded    active exited    Apply Kernel Variables
  systemd-sysusers.service                                                                     loaded    active exited    Create System Users
[... cut ...]
  systemd-udevd.service                                                                        loaded    active running   Rule-based Manager for Device Events and Files
  systemd-update-utmp.service                                                                  loaded    active exited    Update UTMP about System Boot/Shutdown
  systemd-user-sessions.service                                                                loaded    active exited    Permit User Sessions
● unbound.service                                                                              loaded    failed failed    Unbound DNS server

sudo ifdown eth0 caused a fatal SSH error

Joulinar, 2023-08-12:
We are getting closer 😊 Pls reboot your system and share following again

```
ip r
cat /etc/resolv.conf
sudo systemctl | grep '\.service '
```

LazerDickMcCheese, 2023-08-12:
dietpi@DietPi:~$ ip r
default via 192.168.1.254 dev eth0 onlink
10.86.211.0/24 dev tun0 proto kernel scope link src 10.86.211.1
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.5
dietpi@DietPi:~$ cat /etc/resolv.conf
nameserver 9.9.9.9
nameserver 149.112.112.112
dietpi@DietPi:~$ sudo systemctl | grep '\.service '
  console-setup.service                                                                        loaded active exited    Set console font and keymap
  cron.service                                                                                 loaded active running   Regular background program processing daemon
  dietpi-postboot.service                                                                      loaded active exited    DietPi-PostBoot
  dietpi-preboot.service                                                                       loaded active exited    DietPi-PreBoot
  dietpi-ramlog.service                                                                        loaded active exited    DietPi-RAMlog
  dropbear.service                                                                             loaded active running   LSB: Lightweight SSH server
  fake-hwclock.service                                                                         loaded active exited    Restore / save the current clock
  getty@tty1.service                                                                           loaded active running   Getty on tty1
  ifup@eth0.service                                                                            loaded active exited    ifup for eth0
  ifupdown-pre.service                                                                         loaded active exited    Helper to synchronize boot up for ifupdown
  keyboard-setup.service                                                                       loaded active exited    Set the console keyboard layout
  kmod-static-nodes.service                                                                    loaded active exited    Create list of static device nodes for the current kernel
  lighttpd.service                                                                             loaded active running   Lighttpd Daemon
  netfilter-persistent.service                                                                 loaded active exited    netfilter persistent configuration
  networking.service                                                                           loaded active exited    Raise network interfaces
  openvpn.service                                                                              loaded active exited    OpenVPN service
  openvpn@server.service                                                                       loaded active running   OpenVPN connection to server
  php7.4-fpm.service                                                                           loaded active running   The PHP 7.4 FastCGI Process Manager
  pihole-FTL.service                                                                           loaded active running   Pi-hole FTL
  rngd.service                                                                                 loaded active running   Start entropy gathering daemon (rngd)
  systemd-fsck-root.service                                                                    loaded active exited    File System Check on Root Device
  systemd-fsck@dev-disk-by\x2dpartuuid-a97c7839\x2d01.service                                  loaded active exited    File System Check on /dev/disk/by-partuuid/a97c7839-01
  systemd-journal-flush.service                                                                loaded active exited    Flush Journal to Persistent Storage
  systemd-journald.service                                                                     loaded active running   Journal Service
  systemd-modules-load.service                                                                 loaded active exited    Load Kernel Modules
  systemd-random-seed.service                                                                  loaded active exited    Load/Save Random Seed
  systemd-remount-fs.service                                                                   loaded active exited    Remount Root and Kernel File Systems
  systemd-sysctl.service                                                                       loaded active exited    Apply Kernel Variables
  systemd-sysusers.service                                                                     loaded active exited    Create System Users
  systemd-tmpfiles-setup-dev.service                                                           loaded active exited    Create Static Device Nodes in /dev
  systemd-tmpfiles-setup.service                                                               loaded active exited    Create Volatile Files and Directories
  systemd-udev-trigger.service                                                                 loaded active exited    Coldplug All udev Devices
  systemd-udevd.service                                                                        loaded active running   Rule-based Manager for Device Events and Files
  systemd-update-utmp.service                                                                  loaded active exited    Update UTMP about System Boot/Shutdown
  systemd-user-sessions.service                                                                loaded active exited    Permit User Sessions
● unbound.service                                                                              loaded failed failed    Unbound DNS server

Joulinar, 2023-08-12:
Ok way better now. Let's try to reach the web

```
ping -c 5 1.1.1.1
dig github.com
```

LazerDickMcCheese, 2023-08-12:
dietpi@DietPi:~$ ping -c 5 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=57 time=11.3 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=57 time=11.0 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=57 time=11.1 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=57 time=11.4 ms

--- 1.1.1.1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4006ms
rtt min/avg/max/mdev = 10.987/11.167/11.405/0.152 ms
dietpi@DietPi:~$ dig github.com

; <<>> DiG 9.16.37-Debian <<>> github.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 40765
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;github.com.                    IN      A

;; ANSWER SECTION:
github.com.             10      IN      A       140.82.113.4

;; Query time: 28 msec
;; SERVER: 9.9.9.9#53(9.9.9.9)
;; WHEN: Sat Aug 12 05:50:04 BST 2023
;; MSG SIZE  rcvd: 55

Joulinar, 2023-08-12:
Can you try `pihole -up` now?

LazerDickMcCheese, 2023-08-12:
To whom it may concern: we have success!

Before I close this, I have questions like what happened in the first place? What did those commands do to fix it? How can I better configure my install for my needs? etc. Can someone help me out with that as well? I feel like a github report probably isn't the best avenue.

Joulinar, 2023-08-12:
> I have questions like what happened in the first place?

In the past, the package `dhcpcd5` was part of PiHole's own installation. But since a while it is no longer needed and can be removed. Unfortunately, the package sometimes causes the DNS resolution not to work correctly in the worst case, as it overwrites the static DNS entry. In addition, in your case the IP address routes were somehow set incorrectly. Whether this was caused by `dhcpcd5` is not entirely clear.

> What did those commands do to fix it?

We have removed the `dhcpcd5` package and set the routes correctly. The rest of the commands were used for debugging.

> How can I better configure my install for my needs?

Always keep your system up to date. I see an old DietPi version and lots of apt software packages that need to be updated. 

> I feel like a github report probably isn't the best avenue.

GitHub is perfect for this kind of thing. You are also welcome to use our forum as well.

---

Another hint. I see in your services that you have Unbound installed. Is there a reason for this? Normally Unbound is used as an upstream DNS server entry in PiHole. Does the service still work?

```
systemctl status unbound.service
journalctl -u unbound.service
```

If you no longer wish to use the service, you can remove it. 😉

LazerDickMcCheese, 2023-08-12:
I appreciate the help, truly. I was under the impression that I was doing myself a disservice by not including Unbound. But honestly, I have a hard time telling if anything in Linux is working...so it may have been functional at one point, but I have no idea.

Fun new quirk: after updating and rebooting, the Pi-hole (both web UI and SSH) are extremely slow. For example, PuTTY has a fluctuating 15-20 second delay.

Joulinar, 2023-08-12:
You updated DietPi to latest version? How does CPU usage looks like? Can you check `htop`?

LazerDickMcCheese, 2023-08-12:
Yes, I updated every piece of Pi-hole then Diet-pi when I saw there was an update ready; <1% CPU usage

MichaIng, 2023-08-12:
I do not like to admit, but it seems not only on Windows a reboot can be a generic solution, in this case at least after purging dhcpcd 😄. Sorry for suggesting to kill your SSH session via `ifdown` 😅.

Does the Pi-hole query show an unusual high amount of queries or failures? As not the CPU load causes the slow UI, probably it's some loops in network traffic.

LazerDickMcCheese, 2023-08-12:
It gets weirder: the speed is back to normal (but to answer your question, the traffic was less than usual with 0 failures), however my password for the SSH login doesn't work anymore? Any idea what could've caused that and how to fix it?

[2 more replies]

closed 2023-08-13 by LazerDickMcCheese