MichaIng/DietPi #6543: DietPi-Software | Kroki Server

SjoerdV, 2023-08-12:
### Creating a software request

#### Formal software information
- Software title | Kroki Server
- Short description | Server software to convert plain text diagrams to images !
- Official URL (if available) | https://docs.kroki.io
- GitHub URL (if available) | https://github.com/yuzutech/kroki
- (Official) install documentation | https://docs.kroki.io/kroki/setup/manual-install/

#### Are there similar/alternative software titles available with DietPi-Software?
- No

#### What makes your requested software better than the above solutions, if available?
- It is fully open source, without any (feature) limitation to free users. 
- It has a large impact for Software Development and would prevent all kinds of diagrams to be generated over the internet
- I would like an alternative for the Docker method they are offering

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
- downloading the latest JAR file with the webserver software, creating a separate user account to run the java web server, installing all supported diagram binaries, configuring the firewall etc. are tasks that can easily take more than an hour. So it would be nice to have a secure baseline for this 'Development & Programming' tool offered in DietPi.
- It has great install + configuration documentation, so it will probably be not very hard to add to the DietPi Sofware Catalog
- the Jar file could be subject of the updating mechanism in DietPi perhaps? As it is not in the default package repos.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
- The manual install process is very well written, but here is the summary:
- install Java 11+
- downloading the latest JAR file with the webserver software in /opt (with a custom port and environment variable)
- generate a self signed certificate (for internal use)
- creating a separate user account to run the java web server (not in the documentation, but shouldn't be hard to add this step)
- create a service that will run the webserver on start (not in the documentation, but shouldn't be hard to add this step)
- installing ALL supported diagram binaries (mainly through apt-get): https://docs.kroki.io/kroki/setup/configuration/#_diagram_binary_paths
- optional: configuring the firewall to allow incoming port (not in the documentation, but shouldn't be hard to add this step)

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
- I am not really opting for a maintainer position, but I am willing to work closely with a regular maintainer/developer during initial development for TESTING purposes (on AMD64 hardware) and we'll see what might come after that

SjoerdV, 2023-08-13:
Just found out this is way more complex than initially anticipated, because of all the different diagram binaries. So I'll close the case

closed 2023-08-13 by SjoerdV

MichaIng, 2023-08-19:
Is there more need than just installing the diagram binaries e.g. via APT? IMO we also do not necessarily install all supported ones, but only the most commonly needed types, and those which really can be simply installed via APT.

SjoerdV, 2023-08-19:
A lot of `pip` as well, dependency hell is waiting for this one, so better not get burned ;-)

MichaIng, 2023-08-19:
Sounds like a `venv` or even `pyenv` would help ti have an isolated Python environment.

SjoerdV, 2023-08-23:
> Sounds like a `venv` or even `pyenv` would help ti have an isolated Python environment.

Sure, but those packages will be updated frequently and things might break, some binaries need to get built from a repo as well...
I also wouldn't know what the demand for something like this would be.
Anyway, if you think this would be a good addition to the dietpi-software suite you might consider re-opening.