MichaIng/DietPi #6544: DietPI firstboot customization

msanaullahsahar, 2023-08-13:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

1. Unfortunately, there is no way to customize the dietPi OS during the first boot, nothing like cloud-init or userdata.
2. Without a display and keyboard, complex "headless" deployments are impossible.

#### Describe the solution you'd like:

- Modifying first boot customization similar to proposed by @nmcclain at [nmcclain solution](https://github.com/nmcclain/raspberian-firstboot#what-changes-were-made-to-the-standard-raspbian-lite-image)

#### Describe alternatives you've considered:

- considering to tinker with dietPi OS but realised that this method is too dificult for me to implement as i am not a developer. [Possible Solution That can be implemented by DietPi developer](https://github.com/nmcclain/raspberian-firstboot#how-to-reproduce-this-image-yourself)

Joulinar, 2023-08-13:
related to https://dietpi.com/forum/t/how-to-run-a-command-on-first-boot/17641

closed 2023-08-13 by msanaullahsahar

MichaIng, 2023-08-13:
I see you found a way already. For completeness and others who land here, this is what [dietpi.txt](https://github.com/MichaIng/DietPi/blob/master/dietpi.txt) is for, which can be found on the FAT partition of RPi images.

Most other images do not come with a dedicated FAT partition, but a single ext4 filesystem only where the file can be found in the `/boot` directory (like on the running system itself). Sadly, on Windows and macOS it is not trivial to mount and access Linux filesystems like ext4, not even with WSL2. Best is probably to use a VM to mount it and exit the file, when needed.