MichaIng/DietPi #6549: DietPi-Bookworm_Python version

ndg2806, 2023-08-14:
Hello DietPi.
how to downgrade Python version to < 3.10 in Bookworm installation.
Reason: Python 3.11 or greater version creates issues during Radaxa Quad SATA HAT firmware -
( curl -sL [https://rock.sh/get-rockpi-sata 4](https://rock.sh/get-rockpi-sata) | sudo -E bash - ).

Please advise.

Thanks & regards,
Deepak

MichaIng, 2023-08-14:
Which SBC do you try this on? On ROCK 4 or another non-RPi device this should already fail as we do use Radxa's own APT repo anymore and hence there is no `rockpi4-dtbo` package, at least not when the image was flashed within the 1.5 years or so.

On RPi it looks like the installation should go through nicely, or is it this "Adafruit_SSD1306" Python project which just does not run well with Python 3.11?

Generally there is no trivial way to install older Python versions. Debian Bookworm distributes only Python 3.11 via package repository, like every Debian version ships only a single Python version. But it is possible to install it into an isolated [`pyenv`](https://github.com/pyenv/pyenv) environment. `pyenv` allows you to relatively easily compile and install any Python version into a directory structure, which can then be loaded similar to how `venv` and `virtualenv` works. Within the environment you would need to install the required Python modules and the HAT project manually, i.e. Radxa's installer cannot be used. But the needed steps can be derived from it: https://s3.setq.io/rockpi/sh/quad-sata/raspbian.sh

Better would be of course if Radxa or Adafruit or whoever is responsible would update the software to be compatible with Python 3.11 (if this really is the problem). This and Debian Bullseye are stable releases for a while now, and in case of Debian there were 6 months freeze before the release where it was already clear that Python 3.11 will be the final version shipped, hence now 8 months time to add support for any project which shall run on Debian.

ndg2806, 2023-08-14:
I ran curl -sL https://s3.setq.io/rockpi/sh/quad-sata/raspbian.sh | sudo -E bash -,
and got following

Package(s)  python3-setuptools python3-pip python3-pil python3-spidev pigpio python3-pigpio net-tools is required.

Would you like to apt-get install the packages? [Y/n] y
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-spidev : Depends: python3 (< 3.10) but 3.11.2-1+b1 is to be installed
E: Unable to correct problems, you have held broken packages.
Traceback (most recent call last):
  File "/tmp/tmp.4aZ3ZoLIAP/Adafruit_SSD1306-v1.6.2/setup.py", line 3, in <module>
    from setuptools import setup, find_packages
ModuleNotFoundError: No module named 'setuptools'
(Reading database ... 42389 files and directories currently installed.)
Preparing to unpack /tmp/tmp.4HY9SUyi7a ...
Unpacking rockpi-sata (0.14) over (0.14) ...
Setting up rockpi-sata (0.14) ...
Failed to enable unit: Unit file pigpiod.service does not exist.
dpkg: error processing package rockpi-sata (--install):
 installed rockpi-sata package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 rockpi-sata
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory
sh: 0: getcwd() failed: No such file or directory

Joulinar, 2023-08-14:
@MichaIng I guess this is a little bit similar to FFMPEG 🙄 

> ```
> The following packages have unmet dependencies:
>  python3-spidev : Depends: python3 (< 3.10) but 3.11.2-1+b1 is to be installed
> ```

Package `python3-spidev` is pulled from RPI repository and not from global Debian one. https://packages.debian.org/bookworm/python3-spidev

```
root@DietPi4:~# apt show python3-spidev
Package: python3-spidev
Version: 20200602~200721-1
Priority: optional
Section: python
Source: spidev
Maintainer: Serge Schneider <serge@raspberrypi.com>
Installed-Size: 49.2 kB
Depends: python3 (<< 3.10), python3 (>= 3.9~), libc6 (>= 2.17)
Suggests: python-spidev-doc
Homepage: https://github.com/doceme/py-spidev
Download-Size: 13.0 kB
APT-Sources: https://archive.raspberrypi.org/debian bullseye/main arm64 Packages
Description: Bindings for Linux SPI access through spidev (Python 3)
 Contains a module for interfacing with SPI devices from user space via the
 spidev linux kernel driver
 .
 This package installs the library for Python 3.

N: There is 1 additional record. Please use the '-a' switch to see it
root@DietPi4:~#
```

MichaIng, 2023-08-14:
Ah right, because the RPi repo ships this package with a date-based version string while on Debian it is `3.6-1`. Try this:
```sh
cat << '_EOF_' > /etc/apt/preferences.d/python3-spidev
Package: python3-spidev
Pin: origin archive.raspberrypi.org
Pin-Priority: -1
apt install python3-spidev
_EOF_

ndg2806, 2023-08-15:
### **SOLUTION**:
**i performed**;
```
**root@DietPi:~# cat << '_EOF_' > /etc/apt/preferences.d/python3-spidev
> Package: python3-spidev
> Pin: origin archive.raspberrypi.org
> Pin-Priority: -1
> _EOF_
root@DietPi:~# apt install python3-spidev**
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  python3-spidev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 13.5 kB of archives.
After this operation, 94.2 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bookworm/main arm64 python3-spidev arm64 3.6-1+b1 [13.5 kB]
Fetched 13.5 kB in 0s (30.7 kB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package python3-spidev.
(Reading database ... 44462 files and directories currently installed.)
Preparing to unpack .../python3-spidev_3.6-1+b1_arm64.deb ...
Unpacking python3-spidev (3.6-1+b1) ...
Setting up python3-spidev (3.6-1+b1) ...
Setting up rockpi-sata (0.14) ...
Failed to enable unit: Unit file pigpiod.service does not exist.
dpkg: error processing package rockpi-sata (--configure):
 installed rockpi-sata package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 rockpi-sata
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
**then, i performed**;
```
root@DietPi:~# apt update
Hit:1 https://deb.debian.org/debian bookworm InRelease
Hit:2 https://deb.debian.org/debian bookworm-updates InRelease
Hit:3 https://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 https://deb.debian.org/debian bookworm-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
root@DietPi:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up rockpi-sata (0.14) ...
Failed to enable unit: Unit file pigpiod.service does not exist.
dpkg: error processing package rockpi-sata (--configure):
 installed rockpi-sata package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 rockpi-sata
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

**then I realized broken installation of Quad SATA might have caused above errors. so without thinking much further I repeated installation**;
`root@DietPi:~# curl -sL https://s3.setq.io/rockpi/sh/quad-sata/raspbian.sh | sudo -E bash -`

installation went well, RasPi recognizes SATA HAT and disks are kicking loud 😄, thanks lot for the quick support  👍

ndg2806, 2023-08-15:
> @MichaIng I guess this is a little bit similar to FFMPEG 🙄
>
>
> Package `python3-spidev` is pulled from RPI repository and not from global Debian one. [...]

is it an issue to pull from Debian repo? if yes can you fix it?

Joulinar, 2023-08-15:
> is it an issue to pull from Debian repo? if yes can you fix it?

That's fine for time being. At the moment RPI guys did not provide a Bookworm repository. Therefore, we need to pull some files from Debian Bookworm side.

closed 2023-08-15 by ndg2806