MichaIng/DietPi #6550: Add console screen rotation to dietpi-config

owendelong, 2023-08-14:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
I'm using dietpi on a NanoPi R6s as a termina/console server in a cabinet in a datacenter. Mostly, it is accessed via SSH. However, when on-site, I have a monitor and keyboard that are used for local access. I would like the monitor in portrait mode (for a variety of reasons).

#### Describe the solution you'd like:
An option in dietpi-config to control the console screen orientation on the NanoPI R6S (or any platform, ideally)

#### Describe alternatives you've considered:
So far, short of going to an R.Pi and using config.txt, I have been unable to find a viable alternative. Workable alternatives would be very much appreciated.

#### Additional context
DietPi V6.20.1
Linux DietPi 5.10.160 #1 SMP Tue Jul 4 16:02:09 CST 2023 aarch64 GNU/Linux
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

MichaIng, 2023-08-14:
Even setting just the console mode resolution is not trivial with modern mainline kernel. There is this `video=` Linux command-line parameter, but on NanoPi 6 series we have no command-line access with our current images, as the kernel and all boot parameters are buried somewhere in raw on the first 6 partitions.

For reference: https://github.com/MichaIng/DietPi/issues/5641

So I am quite sure that there is no possibility for this until we migrate our images to use proper kernel packages with Linux and all configs directly accessible in `/boot`.
____________
But I had a little review: The problem I had when trying to chance the resolution via [`fbset`](https://manpages.debian.org/fbset) on my NanoPi R6S when it did not match the monitor resolution, was that this command just changed the size of the console on the screen, but not the resolution of the screen. Now I just found the [`-match`](https://manpages.debian.org/bookworm/fbset#match) flag:

> Display geometry:
> 
> - [-xres <value>](https://manpages.debian.org/fbset#xres)
> set visible horizontal resolution (in pixels)
> - [-yres <value>](https://manpages.debian.org/fbset#yres)
> set visible vertical resolution (in pixels)
> - [-vxres <value>](https://manpages.debian.org/fbset#vxres)
> set virtual horizontal resolution (in pixels)
> - [-vyres <value>](https://manpages.debian.org/fbset#vyres)
> set virtual vertical resolution (in pixels)
> - [-depth <value>](https://manpages.debian.org/fbset#depth)
> set display depth (in bits per pixel)
> - [-nonstd <value>](https://manpages.debian.org/fbset#nonstd)
> select nonstandard video mode
> - [--geometry, -g ...](https://manpages.debian.org/fbset#geometry)
> set all geometry parameters at once in the order <xres> <yres> <vxres> <vyres> <depth>, e.g. -g 640 400 640 400 4
> - [-match ](https://manpages.debian.org/bookworm/fbset#match)     
> make the physical resolution match the virtual resolution

This is actually what I was looking for that time, at least if I understand it correctly (needs testing). But while it changes the resolution, AFAIK it does not change the rotation. This does not help, right?
```sh
apt install fbset
# Show current modes
fbset -fb /dev/fb0
# And then switch x and y values, e.g. for 1080x1920x16
fbset -fb /dev/fb0 -g 1080 1920 1080 1920 16 -match
```
But it might not work at all if KMS is used. I still failed to understand whether pure console mode always uses the framebuffer driver and hence a device like `/dev/fb0`, or whether KMS can be used as well to show the console itself, so that `fbset` has no effect, at least not for the actual screen resolution/mode.

owendelong, 2023-08-21:
I'll give it a try and let you know what happens. Not sure if resolution applies or helps here or not since I'm dealing with text-only console, not starting an X server or any graphical application.

MichaIng, 2023-09-01:
Yeah, within X it is trivial, but for console mode it is a challenge for both: resolution and orientation, if default/EDID data is not right/what one wants.

MichaIng, 2025-02-09:
We added a new tool to control console modes and rotation via DRM/KMS sysfs and `video` kernel command-line parameter: #7390
I'll close this issue in favour of the new one.

closed 2025-02-09 by MichaIng