MichaIng/DietPi #6552: Raid5 and Superblocks

ndg2806, 2023-08-15:
**### How to reset /dev/sd[abc] to factory settings?**

currently I face issue during execution `mdadm` for **raid5, with 3 devices**.
mdadm stops after couple of minutes, and mdstat shows no results.

then I stopped` md0`, removed the same, and did `--zero-superblock` as well. I also tried `wipefs -a`.

but following is the overview;
```
root@DietPi:~# sudo mke2fs -n /dev/sdc
mke2fs 1.47.0 (5-Feb-2023)
Creating filesystem with 244190646 4k blocks and 61054976 inodes
Filesystem UUID: 86255c99-0be3-4da4-ada3-24fc87a809e1
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
        4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968,
        102400000, 214990848

root@DietPi:~# sudo mke2fs -n /dev/sda
mke2fs 1.47.0 (5-Feb-2023)
Creating filesystem with 244190646 4k blocks and 61054976 inodes
Filesystem UUID: a34905d8-5743-4b52-b634-4083fe0168b1
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
        4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968,
        102400000, 214990848

root@DietPi:~# sudo mke2fs -n /dev/sdb
mke2fs 1.47.0 (5-Feb-2023)
Creating filesystem with 244190646 4k blocks and 61054976 inodes
Filesystem UUID: 6bf97366-7d7a-4c48-889e-c3376bd7af80
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208,
        4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968,
        102400000, 214990848

```
**so, how to get rid of this and have disks new as factory settings?**

Joulinar, 2023-08-15:
Did you tried using Drive Manager to format the drives?

ndg2806, 2023-08-15:
I have used Drive-manager and Terminal as well.

ndg2806, 2023-08-16:
**current status**;
```
root@DietPi:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
sda           8:0    0 931.5G  0 disk
└─md0         9:0    0     0B  0 raid5
sdb           8:16   0 931.5G  0 disk
└─md0         9:0    0     0B  0 raid5
sdc           8:32   0 931.5G  0 disk
└─md0         9:0    0     0B  0 raid5
mmcblk0     179:0    0 119.1G  0 disk
├─mmcblk0p1 179:1    0   128M  0 part  /boot
└─mmcblk0p2 179:2    0   119G  0 part  /
```

```
root@DietPi:~# mdadm --detail /dev/md0
/dev/md0:
           Version : 1.2
     Creation Time : Wed Aug 16 12:48:25 2023
        Raid Level : raid5
     Used Dev Size : 976630272 (931.39 GiB 1000.07 GB)
      Raid Devices : 3
     Total Devices : 3
       Persistence : Superblock is persistent

       Update Time : Wed Aug 16 12:49:49 2023
             State : active, FAILED, Not Started
    Active Devices : 2
   Working Devices : 3
    Failed Devices : 0
     Spare Devices : 1

            Layout : left-symmetric
        Chunk Size : 512K

Consistency Policy : unknown

              Name : DietPi:0  (local to host DietPi)
              UUID : fbb06b1a:175d6fa9:cfaa5cba:107d55a3
            Events : 17

    Number   Major   Minor   RaidDevice State
       -       0        0        0      removed
       -       0        0        1      removed
       -       0        0        2      removed

       -       8       32        -      spare   /dev/sdc
       -       8        0        0      sync   /dev/sda
       -       8       16        1      sync   /dev/sdb
```

MichaIng, 2023-08-19:
Not sure what you derive from the `mke2fs` output and why you think it is related to the `mdadm` issue? `mke2fs -n` shows you what it _would_ do if you did run it without `-n`. So it always shows you a random UUID and backup superblock positions according to the partition (or drive) size, regardless on which block device you run it.

I have not much experience with RAIDs, but heard that they are quite unreliable on USB drives, so probably with your combination of drives it just won't work. @StephanStS might be able to tell you more or faced such a failure as well by times?

Btw, how do you power those USB drives? Based on partitioning, I assume that it is an RPi? Are there any kernel errors?
```sh
dmesg -l 0,1,2,3
```

StephanStS, 2023-08-19:
Looks that it seems to be similar to https://dietpi.com/blog/?p=3208#Sticking_clean_degraded.
Could you execute the measures given there?

closed 2024-01-16 by MichaIng