MichaIng/DietPi #6554: Command "shutdown" throws an error

Gill-Bates, 2023-08-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
  - G_DIETPI_VERSION_CORE=8
  G_DIETPI_VERSION_SUB=20
  G_DIETPI_VERSION_RC=1
  G_GITBRANCH='master'
  G_GITOWNER='MichaIng'
  G_LIVE_PATCH_STATUS[0]='applied'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
  - bookworm
- Kernel version | `uname -a`
  - Linux dietpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
  - RPi 3 Model B (aarch64)

#### Additional Information (if applicable)

#### Steps to reproduce

1. Login as root
2. enter `shutdown`

#### Expected behaviour
Schedule the shutdown process

#### Actual behaviour
Error Message
```
Failed to set wall message, ignoring: Unit dbus-org.freedesktop.login1.service failed to load properly, please adjust/correct and reload service manager: File exists
Failed to schedule shutdown: Unit dbus-org.freedesktop.login1.service failed to load properly, please adjust/correct and reload service manager: File exists
```

Joulinar, 2023-08-16:
This is known. A cosmic issue only not doing any harm to your system.

MichaIng, 2023-08-19:
Reported to the Debian bug tracker: https://bugs.debian.org/1042015

But I think I need to report this at the systemd repo instead: https://github.com/systemd/systemd/issues