MichaIng/DietPi #6555: Android emulation?

privacyguy123, 2023-08-16:
Any chance of supporting one of the Android emulators so we can run .apk files? My head is pounding reading the workarounds/hacks that would need to be applied to get either Waydroid or Anbox working. :|

MichaIng, 2023-08-19:
How did you try to install those and on which SBC/hardware did you try it?

Waydroid seems to be pretty simple to install with their installer, and their own APT repository: https://docs.waydro.id/usage/install-on-desktops#ubuntu-debian-and-derivatives

Anbox is not actively developed anymore, and only states to support Ubuntu Bionic and Focal. However, it is available as snap, which can be generally installed on Debian/DietPi as well:
```sh
sudo apt install snap
sudo snap install --devmode --edge anbox
```
But I would concentrate on getting Waydroid up, which is also the suggested successor by the Anbox maintainer(s): https://github.com/anbox/anbox#important---hi-there-

anarkia47, 2026-09-18:
Hi, just following up on this request as Waydroid has matured quite a bit since the original issue was opened.

Would there be any interest in adding official Waydroid support to DietPi, possibly through dietpi-software, at least for compatible systems?

Devices such as the ROCK 5B, Orange Pi 5/5 Plus, Radxa ROCK 5A, Khadas Edge2, and other RK3588/RK3588S boards seem particularly well suited to this use case thanks to their performance and ARM64 support.

Having DietPi handle or verify the required Binder/BinderFS support, Wayland dependencies and basic setup would make the installation much more straightforward.

Even if support had to be limited to selected architectures/devices rather than all DietPi systems, I think it could be a useful addition.

Thanks for considering it.

MichaIng, 2026-09-18:
Installing Waydroid since Debian Trixie is as simple as:
```sh
apt install waydroid
```
But it requires a Wayland session, while `dietpi-software` installs and sets up all desktop environments with X11. So settings up the desktop to use Wayland is the harder part, which seems to be individual for every desktop environment, not a generic switch.

I just talked with @StephanStS about that it would be nice to implement Wayland as an alternative to X11 for desktop environments to `dietpi-software`, but at least for LDXE and LXQt we skip custom configuration files, some of which target X11. So it requires some work. Also, Wayland support in LXDE does not seem to be that straight forward to enable. If it gets too complicated, we could also limit the Wayland choice to those desktop environments, where it is easy to enable. When installing Waydroid, we might then need to force users switching to another desktop environment, or starting Wayland with an own Wayland server session from console, it that works.