MichaIng/DietPi #6557: RAXDA ROCK Pi 4: eMMC becomes read-only after sustained writes when running at HS400

Epic-Busta, 2023-08-17:
#### Required Information
- DietPi version: G_DIETPI_VERSION_CORE=8                                                                                                                                                                               G_DIETPI_VERSION_SUB=20                                                                                                                                                                                                         G_DIETPI_VERSION_RC=1   
- Distro version: bookworm
- Kernel version: Linux DietPi 5.15.93-rockchip64 # 23.02.2 SMP PREEMPT Fri Feb 17 23:48:36 UTC 2023 aarch64 GNU/Linux
- SBC model: ROCK Pi 4B v1.5
- Power supply used: Belkin 30W USB-PD USB-C
- SD card used | 16GB Raxda eMMC (add on socket) 

#### Steps to reproduce
1. Run a write intensive process (apt install for example)
2. Will eventually fail due to file system becoming read-only

Rebooting lifts the read-only state.

#### Expected behaviour
System should run as normal. Similar behaviour as if run from SD or other storage media

#### Actual behaviour
System enters a read-only state, essentially soft-locking the system until reboot.
dmesg often (but not always) has this after some write operations:
`mmc1: running CQE recovery`
There is usually one reported every 3-10 seconds during an apt install
Also, if write operations are intensive enough, a dump of what looks like the mmc driver is reported in dmesg. (See raxda forum post at the bottom)

Due to this, it's close to impossible to install anything on the system as apt will instantly lock up due to the RO state. 

#### Extra details
This issue has appeared on the Raxda forums in January and in June a workaround was added to Raxda distributed linux images. The workaround appears to be reducing the speed of the eMMC controller from HS400-es to HS200. I haven't tested the workaround myself using their images, but I assume it works since nothing has been posted about this issue since the workaround was released.

Forum post: https://forum.radxa.com/t/rock4-se-emmc-input-output-errors-read-only-fs/14406
Git pull on raxda kernel for workaround: https://github.com/radxa/kernel/pull/127

Epic-Busta, 2023-08-18:
Looks like #6536 is related to this. The I/O errors they experienced look pretty similar to mine. Also after rebooting several times, the board no longer boots and isn't accessible via SSH. Best guess is that the file system is corrupted.

I did try to use raxda's kernel release, but dpkg threw an I/O error and terminated. Using `powersave` governor doesn't appear to make a difference.

I'll admit, I'm very far out of my league here and talking about anything makes me think I'm [very high on mount stupid.](https://www.smbc-comics.com/comic/2011-12-28) I've just done my best to collate what I've seen about this issue. I'd be happy to do any further troubleshooting and I'll post part of the output of `dmesg` during an `apt install`. In the meantime I'll probably just go back to using an SD card since that works just fine.

Epic-Busta, 2023-08-18:
In this case, the fs didn't enter a RO state, but dpkg errored out due to I/O errors. Any attempts to repeat the installation are fruitless.
```
[  209.090959] mmc1: running CQE recovery
[  209.563136] mmc1: running CQE recovery
[  210.812277] mmc1: running CQE recovery
[  210.842707] mmc1: running CQE recovery
[  210.847838] mmc1: running CQE recovery
[  210.875998] mmc1: running CQE recovery
[  210.883147] mmc1: running CQE recovery
[  210.910576] mmc1: running CQE recovery
[  210.922066] mmc1: running CQE recovery
[  210.943918] mmc1: running CQE recovery
[  210.948894] mmc1: running CQE recovery
[  210.986280] mmc1: running CQE recovery
[  212.699541] mmc1: running CQE recovery
[  212.705554] mmc1: running CQE recovery
[  212.712457] mmc1: running CQE recovery
[  212.713717] blk_update_request: I/O error, dev mmcblk1, sector 1913856 op 0x1:(WRITE) flags 0x4000 phys_seg 9 prio class 0
[  212.761081] EXT4-fs warning (device mmcblk1p1): ext4_end_bio:348: I/O error 10 writing to inode 3099 starting block 239616)
[  212.761277] Buffer I/O error on device mmcblk1p1, logical block 233472
[  212.761983] Buffer I/O error on device mmcblk1p1, logical block 233473
[  212.762608] Buffer I/O error on device mmcblk1p1, logical block 233474
[  212.763224] Buffer I/O error on device mmcblk1p1, logical block 233475
[  212.763837] Buffer I/O error on device mmcblk1p1, logical block 233476
[  212.764452] Buffer I/O error on device mmcblk1p1, logical block 233477
[  212.765066] Buffer I/O error on device mmcblk1p1, logical block 233478
[  212.765681] Buffer I/O error on device mmcblk1p1, logical block 233479
[  212.766336] Buffer I/O error on device mmcblk1p1, logical block 233480
[  212.766960] Buffer I/O error on device mmcblk1p1, logical block 233481
[  213.572421] mmc1: running CQE recovery
[  213.619948] mmc1: running CQE recovery
[  213.730563] mmc1: running CQE recovery
[  213.872340] mmc1: running CQE recovery
[  213.879169] mmc1: running CQE recovery
[  213.992361] mmc1: running CQE recovery
[  214.187480] mmc1: running CQE recovery
[  214.203443] mmc1: running CQE recovery
[  214.345048] mmc1: running CQE recovery
[  214.377660] mmc1: running CQE recovery
[  214.385873] mmc1: running CQE recovery
[  214.396785] mmc1: running CQE recovery
[  214.397282] blk_update_request: I/O error, dev mmcblk1, sector 1987584 op 0x1:(WRITE) flags 0x4000 phys_seg 8 prio class 0
[  214.406890] EXT4-fs warning (device mmcblk1p1): ext4_end_bio:348: I/O error 10 writing to inode 22492 starting block 248764)
[  214.730775] mmc1: running CQE recovery
[  214.803020] mmc1: running CQE recovery
[  214.888657] mmc1: running CQE recovery
[  215.811407] mmc1: running CQE recovery
[  215.833115] mmc1: running CQE recovery
[  215.839689] mmc1: running CQE recovery
[  215.859596] mmc1: running CQE recovery
[  215.872168] mmc1: running CQE recovery
[  215.880326] mmc1: running CQE recovery
[  215.886617] mmc1: running CQE recovery
[  215.888014] blk_update_request: I/O error, dev mmcblk1, sector 580968 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 0
[  215.889068] blk_update_request: I/O error, dev mmcblk1, sector 580968 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 0
[  215.890136] Aborting journal on device mmcblk1p1-8.
[  215.903457] mmc1: running CQE recovery
[  215.915248] mmc1: running CQE recovery
[  215.934097] mmc1: running CQE recovery
[  215.943581] mmc1: running CQE recovery
[  215.949267] mmc1: running CQE recovery
```

MichaIng, 2023-08-18:
Many thanks for your report and investigation. Indeed the symptoms look pretty similar to the other issue you linked, where also an eMMC module was used. I wonder why this issue appears just now, since the kernel did not receive an update since February.

The problem with Radxa's kernel (fix) is that we use (Armbian's) mainline Linux builds. So we'd need a fix applied upstream. Probably this has already happened and we just need a recent build.

I always wanted to test building kernel packages via Armbian's build tools. So far we just used their APT repo, but for strange reasons they did not update it since February, which is already a critical security problem due to missing kernel CVE fixes. Will try that now and keep you updated with a new kernel package. If you cannot install it, I'll generate a new image to flash, with the new kernel.

MichaIng, 2023-08-19:
Test these:
```sh
cd /tmp
curl -fO https://dietpi.com/downloads/binaries/linux-dtb-current-rockchip64.deb
curl -fO https://dietpi.com/downloads/binaries/linux-image-current-rockchip64.deb
apt install ./linux-*
rm ./linux-*
reboot
```

Epic-Busta, 2023-08-19:
Looks like we might need to flash with the new kernel. I just tried twice and dpkg throws an error, even when using `powersave` to slow things down. Attempting to install again afterwards leads dpkg terminating in the same manner.

```
root@DietPi:/tmp# apt install ./armbian-firmware.deb ./linux-*
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'armbian-firmware' instead of './armbian-firmware.deb'
Note, selecting 'linux-dtb-current-rockchip64' instead of './linux-dtb-current-rockchip64.deb'
Note, selecting 'linux-image-current-rockchip64' instead of './linux-image-current-rockchip64.deb'
The following packages will be upgraded:
  linux-dtb-current-rockchip64 linux-image-current-rockchip64
The following packages will be DOWNGRADED:
  armbian-firmware
2 upgraded, 0 newly installed, 1 downgraded, 0 to remove and 0 not upgraded.
Need to get 0 B/56.1 MB of archives.
After this operation, 122 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 /tmp/armbian-firmware.deb armbian-firmware all 23.08.0-trunk [14.1 MB]
Get:2 /tmp/linux-dtb-current-rockchip64.deb linux-dtb-current-rockchip64 arm64 23.08.0-trunk [316 kB]
Get:3 /tmp/linux-image-current-rockchip64.deb linux-image-current-rockchip64 arm64 23.08.0-trunk [41.7 MB]
debconf: delaying package configuration, since apt-utils is not installed
dpkg: warning: downgrading armbian-firmware from 23.08.0-trunk--1-SA1251-B226d to 23.08.0-trunk
(Reading database ... 17596 files and directories currently installed.)
Preparing to unpack /tmp/armbian-firmware.deb ...
Unpacking armbian-firmware (23.08.0-trunk) over (23.08.0-trunk--1-SA1251-B226d) ...
Preparing to unpack .../linux-dtb-current-rockchip64.deb ...
Armbian 'linux-dtb-current-rockchip64' for '6.1.46-current-rockchip64': 'preinst' starting.
Armbian 'linux-dtb-current-rockchip64' for '6.1.46-current-rockchip64': 'preinst' finishing.
Unpacking linux-dtb-current-rockchip64 (23.08.0-trunk) over (23.02.2) ...
Preparing to unpack .../linux-image-current-rockchip64.deb ...
Armbian 'linux-image-current-rockchip64' for '6.1.46-current-rockchip64': 'preinst' starting.
Armbian 'linux-image-current-rockchip64' for '6.1.46-current-rockchip64': 'preinst' finishing.
Unpacking linux-image-current-rockchip64 (23.08.0-trunk) over (23.02.2) ...
dpkg: unrecoverable fatal error, aborting:
 unable to fsync updated status of 'linux-image-current-rockchip64': Input/output error
E: Sub-process /usr/bin/dpkg returned an error code (2)

MichaIng, 2023-08-26:
New image is available here: https://dietpi.com/downloads/images/testing/
You can also wait until tomorrow, where I'll rebuild all images based on DietPi v8.21.

Epic-Busta, 2023-08-27:
Ah, no joy. Looks like I still got issues. It booted fine and was able to both install OpenSSH and run a 1000MiB benchmark on the eMMC, But as soon as I tried to install more stuff (I think it was Avahi and Java JRE. Program selection probably unrelated) the FS entered read only during install and there were a bunch of Buffer I/O errors like before.

I'm not sure if my eMMC is defective, but I'm able to write to it at full speed when connected to a windows PC and write images (~35-40MB/s) so I'm doubtful. Perhaps my Rock Pi 4 is defective? But there's no way to really prove that without another one. Feels a bit dead end here unfortunately.

Epic-Busta, 2023-08-27:
This was in the forum post, but I'll note it here too. There looks to be some kind of process trace in dmesg with the cqhci-core driver (I think?). From a quick google search, cqhci appears to be some kind of thing for queueing commands to eMMC. 

I think the trace happened during initial setup after SSHing in. There's no way to really see when during setup it happened to my knowledge. 

```
[  284.460966] ------------[ cut here ]------------
[  284.460989] mmc1: cqhci: spurious TCN for tag 11
[  284.461147] WARNING: CPU: 0 PID: 275 at drivers/mmc/host/cqhci-core.c:786 cqhci_irq+0x408/0x67c
[  284.461202] Modules linked in: snd_soc_hdmi_codec snd_soc_spdif_tx snd_soc_audio_graph_card snd_soc_simple_card snd_soc_simple_card_utils dw_hdmi_cec snd_soc_es8316 snd_soc_rockchip_i2s panfrost dw_hdmi_i2s_audio gpu_sched btsdio drm_shmem_helper bluetooth rockchip_rng rng_core snd_soc_core brcmfmac hantro_vpu rockchip_vdec(C) brcmutil v4l2_vp9 snd_compress videobuf2_dma_contig snd_pcm_dmaengine rockchip_rga v4l2_h264 v4l2_mem2mem cfg80211 videobuf2_dma_sg snd_pcm videobuf2_memops videobuf2_v4l2 snd_timer rfkill videobuf2_common snd videodev soundcore mc cpufreq_dt dm_mod ip_tables x_tables autofs4 realtek dwmac_rk stmmac_platform stmmac pcs_xpcs
[  284.461717] CPU: 0 PID: 275 Comm: kworker/0:1H Tainted: G         C         6.1.46-current-rockchip64 #3
[  284.461743] Hardware name: Radxa ROCK Pi 4B (DT)
[  284.461760] Workqueue: kblockd blk_mq_run_work_fn
[  284.461805] pstate: 600000c5 (nZCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  284.461831] pc : cqhci_irq+0x408/0x67c
[  284.461861] lr : cqhci_irq+0x408/0x67c
[  284.461887] sp : ffff800008003da0
[  284.461900] x29: ffff800008003da0 x28: 0000000000004000 x27: ffff000000ce2480
[  284.461946] x26: ffff0000015f0580 x25: ffff000000ce2498 x24: ffff800009571c28
[  284.461991] x23: ffff800009c3d8f0 x22: 0000000000000002 x21: ffff000000ce2480
[  284.462035] x20: ffff0000015f0000 x19: 000000000000000b x18: ffffffffffffffff
[  284.462078] x17: ffff80007636e000 x16: ffff800008000000 x15: ffff800009c84066
[  284.462122] x14: 0000000000000000 x13: 0000000000000239 x12: 00000000ffffffea
[  284.462165] x11: 00000000ffffefff x10: 00000000ffffefff x9 : ffff800009ab8908
[  284.462208] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
[  284.462251] x5 : 0000000000000fff x4 : 0000000000000000 x3 : 0000000000000027
[  284.462293] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff0000040f2d00
[  284.462337] Call trace:
[  284.462349]  cqhci_irq+0x408/0x67c
[  284.462379]  sdhci_arasan_cqhci_irq+0x54/0x84
[  284.462408]  sdhci_irq+0xc4/0x1030
[  284.462429]  __handle_irq_event_percpu+0x58/0x240
[  284.462466]  handle_irq_event+0x4c/0xc0
[  284.462497]  handle_fasteoi_irq+0xc0/0x220
[  284.462524]  generic_handle_domain_irq+0x2c/0x44
[  284.462556]  gic_handle_irq+0x50/0x130
[  284.462586]  call_on_irq_stack+0x24/0x4c
[  284.462615]  do_interrupt_handler+0xd4/0xe0
[  284.462644]  el1_interrupt+0x34/0x6c
[  284.462671]  el1h_64_irq_handler+0x18/0x2c
[  284.462698]  el1h_64_irq+0x64/0x68
[  284.462720]  preempt_count_sub+0xc/0xd4
[  284.462743]  sdhci_cqe_enable+0x13c/0x2a0
[  284.462774]  sdhci_arasan_cqe_enable+0xa4/0xf0
[  284.462803]  cqhci_request+0xd4/0x690
[  284.462830]  mmc_cqe_start_req+0xb4/0x1b0
[  284.462861]  mmc_blk_mq_issue_rq+0x448/0x880
[  284.462888]  mmc_mq_queue_rq+0x118/0x2ac
[  284.462914]  blk_mq_dispatch_rq_list+0x1a4/0x8a0
[  284.462939]  __blk_mq_sched_dispatch_requests+0xb4/0x160
[  284.462967]  blk_mq_sched_dispatch_requests+0x3c/0x80
[  284.462994]  __blk_mq_run_hw_queue+0x80/0xc0
[  284.463026]  blk_mq_run_work_fn+0x20/0x30
[  284.463056]  process_one_work+0x1fc/0x480
[  284.463079]  worker_thread+0x14c/0x444
[  284.463100]  kthread+0xf8/0xfc
[  284.463129]  ret_from_fork+0x10/0x20
[  284.463156] ---[ end trace 0000000000000000 ]---
```

MichaIng, 2023-08-27:
I'm also not sure how to know/prove whether its the ROCK 4 or the eMMC module. You could run a bad block check e.g. via Rufus from Windows, but that also does not always see actual issues. In theory, even that you can write well from Windows, it still could be the eMMC and just a certain block/area being broken that is written to after installing some software. Although, the stack trace happened earlier? `dmesg -T` shows the absolute time that can be compared with/timestamps. Also `journalctl` contains kernel errors and at least service start/finish logs.

Probably you can test one of Radxa's images and see whether you face the same issue when installing larger APT packages. If so, it might be at least worth it to contact Radxa support?

Epic-Busta, 2023-09-12:
Apologies for the delay. Got a little busy. 

I ran a block check using Rufus (2 pass) and it didn't show any errors. Then I installed the latest DietPi image and ran it. It first did a dist-upgrade (I think) It had to reboot. Then SSHing in again I installed a minimal image. There doesn't appear to be any issues thus far. I'm going to install OpenSSH (my test so far) and see what happens.

`dmesg -T` output:

``` [Tue Sep 12 21:08:04 2023] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[Tue Sep 12 21:08:04 2023] Linux version 6.1.50-current-rockchip64 (armbian@next) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #3 SMP PREEMPT Wed Aug 30 14:11:13 UTC 2023
[Tue Sep 12 21:08:04 2023] Machine model: Radxa ROCK Pi 4B
[Tue Sep 12 21:08:04 2023] efi: UEFI not found.
[Tue Sep 12 21:08:04 2023] NUMA: No NUMA configuration found
[Tue Sep 12 21:08:04 2023] NUMA: Faking a node at [mem 0x0000000000200000-0x000000007fffffff]
[Tue Sep 12 21:08:04 2023] NUMA: NODE_DATA [mem 0x7fb8ea80-0x7fb90fff]
[Tue Sep 12 21:08:04 2023] Zone ranges:
[Tue Sep 12 21:08:04 2023]   DMA      [mem 0x0000000000200000-0x000000007fffffff]
[Tue Sep 12 21:08:04 2023]   DMA32    empty
[Tue Sep 12 21:08:04 2023]   Normal   empty
[Tue Sep 12 21:08:04 2023] Movable zone start for each node
[Tue Sep 12 21:08:04 2023] Early memory node ranges
[Tue Sep 12 21:08:04 2023]   node   0: [mem 0x0000000000200000-0x000000007fffffff]
[Tue Sep 12 21:08:04 2023] Initmem setup node 0 [mem 0x0000000000200000-0x000000007fffffff]
[Tue Sep 12 21:08:04 2023] On node 0, zone DMA: 512 pages in unavailable ranges
[Tue Sep 12 21:08:04 2023] cma: Reserved 128 MiB at 0x0000000071200000
[Tue Sep 12 21:08:04 2023] psci: probing for conduit method from DT.
[Tue Sep 12 21:08:04 2023] psci: PSCIv1.0 detected in firmware.
[Tue Sep 12 21:08:04 2023] psci: Using standard PSCI v0.2 function IDs
[Tue Sep 12 21:08:04 2023] psci: MIGRATE_INFO_TYPE not supported.
[Tue Sep 12 21:08:04 2023] psci: SMC Calling Convention v1.0
[Tue Sep 12 21:08:04 2023] percpu: Embedded 30 pages/cpu s82408 r8192 d32280 u122880
[Tue Sep 12 21:08:04 2023] pcpu-alloc: s82408 r8192 d32280 u122880 alloc=30*4096
[Tue Sep 12 21:08:04 2023] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5
[Tue Sep 12 21:08:04 2023] Detected VIPT I-cache on CPU0
[Tue Sep 12 21:08:04 2023] CPU features: detected: GIC system register CPU interface
[Tue Sep 12 21:08:04 2023] CPU features: detected: ARM erratum 845719
[Tue Sep 12 21:08:04 2023] alternatives: applying boot alternatives
[Tue Sep 12 21:08:04 2023] Fallback order for Node 0: 0
[Tue Sep 12 21:08:04 2023] Built 1 zonelists, mobility grouping on.  Total pages: 515592
[Tue Sep 12 21:08:04 2023] Policy zone: DMA
[Tue Sep 12 21:08:04 2023] Kernel command line: root=UUID=9d0e871a-f90d-4f76-9023-fa008ebfade6 rootfstype=ext4 rootwait console=ttyS2,1500000 console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0
[Tue Sep 12 21:08:04 2023] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[Tue Sep 12 21:08:04 2023] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[Tue Sep 12 21:08:04 2023] mem auto-init: stack:off, heap alloc:on, heap free:off
[Tue Sep 12 21:08:04 2023] Memory: 1880732K/2095104K available (16512K kernel code, 2278K rwdata, 6172K rodata, 3968K init, 585K bss, 83300K reserved, 131072K cma-reserved)
[Tue Sep 12 21:08:04 2023] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=6, Nodes=1
[Tue Sep 12 21:08:04 2023] trace event string verifier disabled
[Tue Sep 12 21:08:04 2023] rcu: Preemptible hierarchical RCU implementation.
[Tue Sep 12 21:08:04 2023] rcu:         RCU event tracing is enabled.
[Tue Sep 12 21:08:04 2023] rcu:         RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=6.
[Tue Sep 12 21:08:04 2023]      Trampoline variant of Tasks RCU enabled.
[Tue Sep 12 21:08:04 2023]      Tracing variant of Tasks RCU enabled.
[Tue Sep 12 21:08:04 2023] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[Tue Sep 12 21:08:04 2023] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=6
[Tue Sep 12 21:08:04 2023] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[Tue Sep 12 21:08:04 2023] GICv3: GIC: Using split EOI/Deactivate mode
[Tue Sep 12 21:08:04 2023] GICv3: 256 SPIs implemented
[Tue Sep 12 21:08:04 2023] GICv3: 0 Extended SPIs implemented
[Tue Sep 12 21:08:04 2023] Root IRQ handler: gic_handle_irq
[Tue Sep 12 21:08:04 2023] GICv3: GICv3 features: 16 PPIs
[Tue Sep 12 21:08:04 2023] GICv3: CPU0: found redistributor 0 region 0:0x00000000fef00000
[Tue Sep 12 21:08:04 2023] ITS [mem 0xfee20000-0xfee3ffff]
[Tue Sep 12 21:08:04 2023] ITS@0x00000000fee20000: allocated 65536 Devices @500000 (flat, esz 8, psz 64K, shr 0)
[Tue Sep 12 21:08:04 2023] ITS: using cache flushing for cmd queue
[Tue Sep 12 21:08:04 2023] GICv3: using LPI property table @0x00000000004e0000
[Tue Sep 12 21:08:04 2023] GIC: using cache flushing for LPI property table
[Tue Sep 12 21:08:04 2023] GICv3: CPU0: using allocated LPI pending table @0x00000000004f0000
[Tue Sep 12 21:08:04 2023] GICv3: GIC: PPI partition interrupt-partition-0[0] { /cpus/cpu@0[0] /cpus/cpu@1[1] /cpus/cpu@2[2] /cpus/cpu@3[3] }
[Tue Sep 12 21:08:04 2023] GICv3: GIC: PPI partition interrupt-partition-1[1] { /cpus/cpu@100[4] /cpus/cpu@101[5] }
[Tue Sep 12 21:08:04 2023] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[... cut ...]
[Tue Sep 12 21:08:10 2023] rk_gmac-dwmac fe300000.ethernet eth0: configuring for phy/rgmii link mode
[Tue Sep 12 21:08:11 2023] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[Tue Sep 12 21:08:12 2023] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
[Tue Sep 12 21:08:14 2023] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[Tue Sep 12 21:08:14 2023] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[Tue Sep 12 21:08:21 2023] platform sound-dit: deferred probe pending

MichaIng, 2023-09-12:
So far no kernel errors, let's hope it stays like that.

Epic-Busta, 2023-09-12:
It just.. _works_?

I installed OpenSSH, Avahi-Daemon and Java JRE. There was not a single hiccup, nor any new lines in dmesg.
I also ran a 2000MiB test using the built in benchmark in drive-manager. Had a write speed of 52 MiB/s and a read speed of 84 MiB/s. No new lines in dmesg regarding mmc.

I don't know if it was something you did or an update upstream, but it works now.

Epic-Busta, 2024-01-06:
My ROCK PI 4 has had no issues for over a year now. It's been running a tailscale client and a wordpress website with basically no downtime. 

I'll be closing this issue then. Thanks for looking into this :)

closed 2024-01-06 by Epic-Busta