MichaIng/DietPi #6559: PHP 7.4 autoremove error in bookworm update script

rtglobal, 2023-08-18:
│
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Fri Aug 18 17:37:59 CEST 2023
- DietPi version | v8.21.-1 (MichaIng/dev)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages purge php7.4 php7.4-bcmath php7.4-bz2 php7.4-cgi php7.4-cli
php7.4-common php7.4-curl php7.4-dba php7.4-dev php7.4-enchant php7.4-fpm php7.4-gd php7.4-gmp php7.4-imap php7.4-interb
ase php7.4-intl php7.4-json php7.4-ldap php7.4-mbstring php7.4-mysql php7.4-odbc php7.4-opcache php7.4-pgsql php7.4-phpd
bg php7.4-pspell php7.4-readline php7.4-snmp php7.4-soap php7.4-sqlite3 php7.4-sybase php7.4-tidy php7.4-xml php7.4-xmlr
pc php7.4-xsl php7.4-zip`
- Exit code      | 100
- Software title | dietpi-bookworm-upgrade
#### Steps to reproduce:

1. Running your dietpi bookworm update script, coming from bullseye
2. Script stops at this point.
#### Expected behaviour:

- The script should do all the wizarding things until the normal end and then should activate or restart my dietpi services (nextcloud, redis, nginx, mumble, etc.).
#### Actual behaviour:

- Script stops at this point. A second start oft this script errors because I am already on bookworm.
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Package 'php7.4' is not installed, so not removed
Package 'php7.4-bcmath' is not installed, so not removed
Package 'php7.4-bz2' is not installed, so not removed
Package 'php7.4-cgi' is not installed, so not removed
Package 'php7.4-dba' is not installed, so not removed
Package 'php7.4-dev' is not installed, so not removed
Package 'php7.4-enchant' is not installed, so not removed
Package 'php7.4-gmp' is not installed, so not removed
Package 'php7.4-imap' is not installed, so not removed
Package 'php7.4-interbase' is not installed, so not removed
Package 'php7.4-ldap' is not installed, so not removed
Package 'php7.4-odbc' is not installed, so not removed
Package 'php7.4-pgsql' is not installed, so not removed
Package 'php7.4-phpdbg' is not installed, so not removed
Package 'php7.4-pspell' is not installed, so not removed
Package 'php7.4-snmp' is not installed, so not removed
Package 'php7.4-soap' is not installed, so not removed
Package 'php7.4-sqlite3' is not installed, so not removed
Package 'php7.4-sybase' is not installed, so not removed
Package 'php7.4-tidy' is not installed, so not removed
Package 'php7.4-xmlrpc' is not installed, so not removed
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved

rtglobal, 2023-08-18:
root@DietPi:~# php -v
PHP Warning:  IcePHP_Properties::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  Ice\ObjectPrx::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Logger::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Endpoint::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Connection::__toString() implemented without string return type in Unknown on line 0
PHP 8.2.7 (cli) (built: Jun  9 2023 19:37:27) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.2.7, Copyright (c) Zend Technologies
    with Zend OPcache v8.2.7, Copyright (c), by Zend Technologies

MichaIng, 2023-08-18:
Many thanks for your report.

Strangely it tries to install packages while it should purge them. Probably there is one explicitly depending on one of those PHP 7.4 packages. And then it looks like PHP 8.2 is already installed, but broken, while it should actually be installed afterwards. And the APT purge command actually checks for and attempts to purge only installed packages, while on your case or tried to purge a lot more which are not installed. It should even adjust the list on every retry, in case some have been purged already prior to the error.

Can you show the output of the following commands:
```sh
dpkg -l | grep 'php7.4'
apt upgrade
apt purge php7.4-common # in case, confirm
dpkg -l | grep 'php7.4'
dmesg -l 0,1,2,3
```

rtglobal, 2023-08-19:
root@DietPi:~# dpkg -l | grep 'php'
hi  php-apcu                        5.1.19+4.0.11-3
  arm64        APC User Cache for PHP
hi  php-common                      2:76
  all          Common files for PHP packages
hi  php-igbinary                    3.2.1+2.0.8-2
  arm64        igbinary PHP serializer
hi  php-redis                       5.3.2+4.3.0-2+deb11u1
  arm64        PHP extension for interfacing with Redis
ii  php-zeroc-ice                   3.7.8-2.1+b1
  arm64        PHP extension for Ice
hi  php7.4-cli                      7.4.33-1+deb11u1
  arm64        command-line interpreter for the PHP scripting langu
age
hi  php7.4-common                   7.4.33-1+deb11u1
  arm64        documentation, examples and common module for PHP
hi  php7.4-curl                     7.4.33-1+deb11u1
  arm64        CURL module for PHP
hi  php7.4-fpm                      7.4.33-1+deb11u1
  arm64        server-side, HTML-embedded scripting language (FPM-C
GI binary)
hi  php7.4-gd                       7.4.33-1+deb11u1
  arm64        GD module for PHP
hi  php7.4-intl                     7.4.33-1+deb11u1
  arm64        Internationalisation module for PHP
hi  php7.4-json                     7.4.33-1+deb11u1
  arm64        JSON module for PHP
hi  php7.4-mbstring                 7.4.33-1+deb11u1
  arm64        MBSTRING module for PHP
hi  php7.4-mysql                    7.4.33-1+deb11u1
  arm64        MySQL module for PHP
hi  php7.4-opcache                  7.4.33-1+deb11u1
  arm64        Zend OpCache module for PHP
hi  php7.4-readline                 7.4.33-1+deb11u1
  arm64        readline module for PHP
hi  php7.4-xml                      7.4.33-1+deb11u1
  arm64        DOM, SimpleXML, XML, and XSL module for PHP
hi  php7.4-zip                      7.4.33-1+deb11u1
  arm64        Zip module for PHP
ii  php8.2-cli                      8.2.7-1~deb12u1
  arm64        command-line interpreter for the PHP scripting langu
age
ii  php8.2-common                   8.2.7-1~deb12u1
  arm64        documentation, examples and common module for PHP
ii  php8.2-opcache                  8.2.7-1~deb12u1
  arm64        Zend OpCache module for PHP
ii  php8.2-phpdbg                   8.2.7-1~deb12u1
  arm64        server-side, HTML-embedded scripting language (PHPDB
G binary)
ii  php8.2-readline                 8.2.7-1~deb12u1
  arm64        readline module for PHP
                                                                   

rtglobal, 2023-08-19:
Thanks for your help. Here comes the requested output:
`root@DietPi:~# dpkg -l | grep 'php'
hi  php-apcu                        5.1.19+4.0.11-3                  arm64        APC User Cache for PHP
hi  php-common                      2:76                             all          Common files for PHP packages
hi  php-igbinary                    3.2.1+2.0.8-2                    arm64        igbinary PHP serializer
hi  php-redis                       5.3.2+4.3.0-2+deb11u1            arm64        PHP extension for interfacing with Redis
ii  php-zeroc-ice                   3.7.8-2.1+b1                     arm64        PHP extension for Ice
hi  php7.4-cli                      7.4.33-1+deb11u1                 arm64        command-line interpreter for the PHP scripting language
hi  php7.4-common                   7.4.33-1+deb11u1                 arm64        documentation, examples and common module for PHP
hi  php7.4-curl                     7.4.33-1+deb11u1                 arm64        CURL module for PHP
hi  php7.4-fpm                      7.4.33-1+deb11u1                 arm64        server-side, HTML-embedded scripting language (FPM-CGI binary)
hi  php7.4-gd                       7.4.33-1+deb11u1                 arm64        GD module for PHP
hi  php7.4-intl                     7.4.33-1+deb11u1                 arm64        Internationalisation module for PHP
hi  php7.4-json                     7.4.33-1+deb11u1                 arm64        JSON module for PHP
hi  php7.4-mbstring                 7.4.33-1+deb11u1                 arm64        MBSTRING module for PHP
hi  php7.4-mysql                    7.4.33-1+deb11u1                 arm64        MySQL module for PHP
hi  php7.4-opcache                  7.4.33-1+deb11u1                 arm64        Zend OpCache module for PHP
hi  php7.4-readline                 7.4.33-1+deb11u1                 arm64        readline module for PHP
hi  php7.4-xml                      7.4.33-1+deb11u1                 arm64        DOM, SimpleXML, XML, and XSL module for PHP
hi  php7.4-zip                      7.4.33-1+deb11u1                 arm64        Zip module for PHP
ii  php8.2-cli                      8.2.7-1~deb12u1                  arm64        command-line interpreter for the PHP scripting language
ii  php8.2-common                   8.2.7-1~deb12u1                  arm64        documentation, examples and common module for PHP
ii  php8.2-opcache                  8.2.7-1~deb12u1                  arm64        Zend OpCache module for PHP
ii  php8.2-phpdbg                   8.2.7-1~deb12u1                  arm64        server-side, HTML-embedded scripting language (PHPDBG binary)
ii  php8.2-readline                 8.2.7-1~deb12u1                  arm64        readline module for PHP
`

`root@DietPi:~# apt upgrade
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paketaktualisierung (Upgrade) wird berechnet… Fertig
Die folgenden Pakete sind zurückgehalten worden:
  libnginx-mod-http-echo mariadb-common mariadb-server mysql-common nginx-common nginx-light php-apcu php-common php-igbinary php-redis redis-server
  redis-tools
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 12 nicht aktualisiert.
`
`root@DietPi:~# apt purge php7.4-common
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 php7.4-cli : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-curl : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-fpm : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-gd : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-intl : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-json : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-mbstring : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-mysql : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-opcache : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-readline : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-xml : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
 php7.4-zip : Hängt ab von: php7.4-common (= 7.4.33-1+deb11u1) soll aber nicht installiert werden
E: Fehler: Unterbrechungen durch pkgProblemResolver::Resolve hervorgerufen; dies könnte durch zurückgehaltene Pakete verursacht worden sein.
`
`root@DietPi:~# dpkg -l | grep 'php'
hi  php-apcu                        5.1.19+4.0.11-3                  arm64        APC User Cache for PHP
hi  php-common                      2:76                             all          Common files for PHP packages
[... cut ...]
ii  php8.2-common                   8.2.7-1~deb12u1                  arm64        documentation, examples and common module for PHP
ii  php8.2-opcache                  8.2.7-1~deb12u1                  arm64        Zend OpCache module for PHP
ii  php8.2-phpdbg                   8.2.7-1~deb12u1                  arm64        server-side, HTML-embedded scripting language (PHPDBG binary)
ii  php8.2-readline                 8.2.7-1~deb12u1                  arm64        readline module for PHP
`
`root@DietPi:~# dmesg -l 0,1,2,3
[    0.597690] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
`

MichaIng, 2023-08-19:
Ah this is the reason: You set all PHP 7.4 packages on hold. This way of course they cannot be upgraded or removed. For packages from the Debian repo of a stable Debian version, it should never be required to set any package on hold, and it is harmful, since it prevents security upgrades. Please unhold all packages:
```sh
apt-mark unhold $(apt-mark showhold)
```
Please use the `dev` branch of the script, as the `master` one does not allow to run on a Bookworm system. But it makes sense to allow repeating it, in case of a failure between package upgrades and software migrations, hence I just patched it:
```sh
bash -c "$(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-bookworm-upgrade')"
```

rtglobal, 2023-08-19:
Thanks for your help. I went through this commands. In between there was another issue with mariadb, but I could jumped over this point. Next issue was known warning to IcePHP. The script runs until the end including `restart` and `apt autoremove`.

My problem is still this following php-error, which blocks all php-related commands.

`root@DietPi:~# php -v`
`PHP Warning:  IcePHP_Properties::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  Ice\ObjectPrx::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  IcePHP_Logger::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  IcePHP_Endpoint::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  IcePHP_Connection::__toString() implemented without string return type in Unknown on line 0`
`PHP 8.2.7 (cli) (built: Jun  9 2023 19:37:27) (NTS)`
`Copyright (c) The PHP Group`
`Zend Engine v4.2.7, Copyright (c) Zend Technologies`
`    with Zend OPcache v8.2.7, Copyright (c), by Zend Technologies
` 

`root@DietPi:~# sudo -u www-data php /var/www/nextcloud/ occ db:add-missing-indices`
`PHP Warning:  IcePHP_Properties::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  Ice\ObjectPrx::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  IcePHP_Logger::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  IcePHP_Endpoint::__toString() implemented without string return type in Unknown on line 0`
`PHP Warning:  IcePHP_Connection::__toString() implemented without string return type in Unknown on line 0`
`

MichaIng, 2023-08-19:
Is this an additional module or somehow a wrapper for the PHP CLI?

I found the IcePHP project with a matching issue here: https://github.com/zeroc-ice/ice/issues/1405

Can show:
```sh
which php
php -m
```

So whatever it is exactly, and why it is invoked when you just call the PHP CLI, it may require an update to include this patch: https://github.com/zeroc-ice/ice/pull/1423

Btw, you can create nicer code and log blocks like this:
````console
```console
root@DietPi:~# sudo -u www-data php /var/www/nextcloud/ occ db:add-missing-indices
PHP Warning:  IcePHP_Properties::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  Ice\ObjectPrx::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Logger::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Endpoint::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Connection::__toString() implemented without string return type in Unknown on line 0
```
````
The "console" behind the open syntax highlights part of the console logs a way that the command prompt, command and output can be nicely seen. Skipping this will remove the colouring, but still have the nicely readable multi-line monotype block.

rtglobal, 2023-08-19:
I do not know why I have or why I need this IcePHP. If there is a possible way without this, I would kick it away.  ;o)
Actually I tried to call my nextcloud server after this update procedure, but the browser still shows nothing. Maybe something with nginx or with mariadb? 
I could start the update from nextcloud 25.0.10 to 26.x and this this nextcloud update seems to look fine. But after all, my nextcloud instance is not present.

When I try to reinstall mariadb I run into this error:

`[ INFO ] DietPi-Software | APT install mariadb-server php8.2-mysql, please wait...`
`Reading package lists...`
`Building dependency tree...`
`Reading state information...`
`php8.2-mysql is already the newest version (8.2.7-1~deb12u1).`
`Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.`
`The following information may help to resolve the situation:`

`The following packages have unmet dependencies:`
` mariadb-server : Depends: mariadb-client (>= 1:10.11.3-1) but it is not going to be installed`
`E: Unable to correct problems, you have held broken packages.`
`[FAILED] DietPi-Software | APT install mariadb-server php8.2-mysql`
` - Command: apt-get -y --allow-change-held-packages install mariadb-server php8.2-mysql`
`[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...`
`[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...`
`[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT`
`[  OK  ] DietPi-BugReport | Packing upload archive`
`[  OK  ] DietPi-BugReport | Sending bug report`
`[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.`
`- Reference code: 6f733a96-c184-47d8-b4e2-7f3d35bdb963`

`Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1
`

MichaIng, 2023-08-19:
Please show the output of the commands I posted above, and additionally:
```sh
apt install mariadb-server
```

rtglobal, 2023-08-19:
```
root@DietPi:~# which php
/usr/bin/php
```

`root@DietPi:~# php -m
PHP Warning:  IcePHP_Properties::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  Ice\ObjectPrx::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Logger::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Endpoint::__toString() implemented without string return type in Unknown on line 0
PHP Warning:  IcePHP_Connection::__toString() implemented without string return type in Unknown on line 0
[PHP Modules]
apcu
calendar
Core
ctype
curl
date
dom
exif
FFI
fileinfo
filter
ftp
gd
gettext
hash
ice
iconv
igbinary
intl
json
libxml
mbstring
mysqli
mysqlnd
openssl
pcntl
pcre
PDO
pdo_mysql
Phar
posix
random
readline
redis
Reflection
session
shmop
SimpleXML
sockets
sodium
SPL
standard
sysvmsg
sysvsem
sysvshm
tokenizer
xml
xmlreader
xmlwriter
xsl
Zend OPcache
zip
zlib

[Zend Modules]
Zend OPcache

`

rtglobal, 2023-08-19:
```
root@DietPi:~# apt install mariadb-server
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 mariadb-server : Hängt ab von: mariadb-client (>= 1:10.11.3-1) soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
```

MichaIng, 2023-08-19:
There it is, the `ice` module. Is there any package providing this?
```sh
dpkg -l | grep ice
```
Disable it for now. For Nextcloud it is definitely not required. If you have another PHP application, which requires it, it needs to be updated first anyway:
```sh
phpdismod ice
php -v
```

And lets see why `mariadb-client` cannot be upgraded:
```sh
apt install mariadb-client
```

rtglobal, 2023-08-19:
```
root@DietPi:~# dpkg -l | grep ice
ii  device-tree-compiler            1.6.1-4+b1                       arm64        Device Tree Compiler for Flat Device Trees
ii  dirmngr                         2.2.40-1.1                       arm64        GNU privacy guard - network certificate management service
ii  dmsetup                         2:1.02.185-2                     arm64        Linux Kernel Device Mapper userspace library
ii  ethtool                         1:6.1-1                          arm64        display or change Ethernet device settings
ii  gpg-wks-client                  2.2.40-1.1                       arm64        GNU privacy guard - Web Key Service client
ii  gpg-wks-server                  2.2.40-1.1                       arm64        GNU privacy guard - Web Key Service server
ii  iw                              5.19-1                           arm64        tool for configuring Linux wireless devices
ii  libblkid1:arm64                 2.38.1-5+b1                      arm64        block device ID library
ii  libdevmapper1.02.1:arm64        2:1.02.185-2                     arm64        Linux Kernel Device Mapper userspace library
ii  libfdt1:arm64                   1.6.1-4+b1                       arm64        Flat Device Trees manipulation library
ii  libmount1:arm64                 2.38.1-5+b1                      arm64        device mounting library
ii  libzeroc-ice3.7:arm64           3.7.8-2.1+b1                     arm64        C++ run-time libraries for the Ice framework
ii  php-zeroc-ice                   3.7.8-2.1+b1                     arm64        PHP extension for Ice
ii  systemd                         252.12-1~deb12u1                 arm64        system and service manager
ii  systemd-sysv                    252.12-1~deb12u1                 arm64        system and service manager - SysV compatibility symlinks
ii  systemd-timesyncd               252.12-1~deb12u1                 arm64        minimalistic service to synchronize local time with NTP servers
```

```
root@DietPi:~# phpdismod ice
root@DietPi:~# php -v
PHP 8.2.7 (cli) (built: Jun  9 2023 19:37:27) (NTS)
Copyright (c) The PHP Group
Zend Engine v4.2.7, Copyright (c) Zend Technologies
    with Zend OPcache v8.2.7, Copyright (c), by Zend Technologies

```

root@DietPi:~# apt install mariadb-client
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  libaio1 libsnappy1v5 libssl1.1
Verwenden Sie »apt autoremove«, um sie zu entfernen.
Die folgenden zusätzlichen Pakete werden installiert:
  libdaxctl1 libndctl6 libpmem1 liburing2 mariadb-client-core mariadb-server mariadb-server-core
Vorgeschlagene Pakete:
  mailx mariadb-test netcat-openbsd
Empfohlene Pakete:
  libdbd-mariadb-perl | libdbd-mysql-perl libterm-readkey-perl libhtml-template-perl mariadb-plugin-provider-bzip2 mariadb-plugin-provider-lz4
  mariadb-plugin-provider-lzma mariadb-plugin-provider-lzo mariadb-plugin-provider-snappy pv
Die folgenden Pakete werden ENTFERNT:
  mariadb-client-10.5 mariadb-client-core-10.5 mariadb-server-10.5 mariadb-server-core-10.5
Die folgenden NEUEN Pakete werden installiert:
  libdaxctl1 libndctl6 libpmem1 liburing2 mariadb-client mariadb-client-core mariadb-server-core
Die folgenden zurückgehaltenen Pakete werden verändert:
  mariadb-client
Die folgenden Pakete werden aktualisiert (Upgrade):
  mariadb-server
1 aktualisiert, 7 neu installiert, 4 zu entfernen und 0 nicht aktualisiert.
Es müssen 14,0 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 37,3 MB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] j
Holen:1 https://deb.debian.org/debian bookworm/main arm64 mariadb-server arm64 1:10.11.3-1 [3.439 kB]
Holen:2 https://deb.debian.org/debian bookworm/main arm64 mariadb-client-core arm64 1:10.11.3-1 [845 kB]
Holen:3 https://deb.debian.org/debian bookworm/main arm64 mariadb-client arm64 1:10.11.3-1 [2.867 kB]
Holen:4 https://deb.debian.org/debian bookworm/main arm64 libdaxctl1 arm64 76.1-1 [19,8 kB]
Holen:5 https://deb.debian.org/debian bookworm/main arm64 libndctl6 arm64 76.1-1 [57,4 kB]
Holen:6 https://deb.debian.org/debian bookworm/main arm64 libpmem1 arm64 1.12.1-2 [45,7 kB]
Holen:7 https://deb.debian.org/debian bookworm/main arm64 liburing2 arm64 2.3-3 [12,6 kB]
Holen:8 https://deb.debian.org/debian bookworm/main arm64 mariadb-server-core arm64 1:10.11.3-1 [6.691 kB]
Es wurden 14,0 MB in 2 s geholt (5.749 kB/s).     
debconf: delaying package configuration, since apt-utils is not installed
dpkg: mariadb-server-10.5: Abhängigkeitsprobleme, wird aber wie gefordert dennoch entfernt:
 mariadb-server hängt ab von mariadb-server-10.5 (>= 1:10.5.15-0+deb11u1).

(Lese Datenbank ... 33504 Dateien und Verzeichnisse sind derzeit installiert.)
[... 20 lines cut ...]
Vorbereitung zum Entpacken von .../3-libndctl6_76.1-1_arm64.deb ...
Entpacken von libndctl6:arm64 (76.1-1) ...
Vormals nicht ausgewähltes Paket libpmem1:arm64 wird gewählt.
Vorbereitung zum Entpacken von .../4-libpmem1_1.12.1-2_arm64.deb ...
Entpacken von libpmem1:arm64 (1.12.1-2) ...
Vormals nicht ausgewähltes Paket liburing2:arm64 wird gewählt.
Vorbereitung zum Entpacken von .../5-liburing2_2.3-3_arm64.deb ...
Entpacken von liburing2:arm64 (2.3-3) ...
Vormals nicht ausgewähltes Paket mariadb-server-core wird gewählt.
[... cut ...]
Neue Version der Konfigurationsdatei /etc/init.d/mariadb wird installiert ...
Neue Version der Konfigurationsdatei /etc/logrotate.d/mariadb wird installiert ...
Neue Version der Konfigurationsdatei /etc/mysql/debian-start wird installiert ...
Neue Version der Konfigurationsdatei /etc/mysql/mariadb.conf.d/50-server.cnf wird installiert ...
Warning: The unit file, source configuration file or drop-ins of mariadb.service changed on disk. Run 'systemctl daemon-reload' to reload units.
Trigger für libc-bin (2.36-9+deb12u1) werden verarbeitet ...

rtglobal, 2023-08-19:
```
root@DietPi:~# apt install mariadb-server
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
mariadb-server ist schon die neueste Version (1:10.11.3-1).
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  libaio1 libsnappy1v5 libssl1.1
Verwenden Sie »apt autoremove«, um sie zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
```

MichaIng, 2023-08-19:
Okay, so this PHP ice module is a Debian package. Indeed it has last been updated (January) on the Debian repo before the fix for these warnings has been merged (February):
- https://metadata.ftp-master.debian.org/changelogs//main/z/zeroc-ice/zeroc-ice_3.7.8-2.1_changelog

The update is however on the way, at least Trixie has v3.7.9 from March already: https://packages.debian.org/trixie/php-zeroc-ice
We could ask Ondrej (the Debian PHP maintainer) whether he can backport this fixed version for Bullseye, or even ship it as regular update, or merge the fix/patch only, as having these bunch of warnings on every PHP call is of course not acceptable for a package shipped on a stable Debian release.

What is disturbing is that even we ran `apt-mark unhold` on all held packages, `mariadb-client` again was set on hold, which was the reason why again an upgrade was not possible non-interactively.:
```
Die folgenden zurückgehaltenen Pakete werden verändert:
  mariadb-client
```
I assume you do not do this manually, so what the hack marks these packages on hold? Are there more?
```sh
apt-mark showhold
```

And to check for Nextcloud, let's see whether all required services are running successfully:
```sh
journalctl -u mariadb-server -u php8.2-fpm -u nginx -u redis
```

rtglobal, 2023-08-19:
```
root@DietPi:~# apt-mark showhold
nginx-core
php-apcu-bc
php-doctrine-bundle
php-facedetect
php-geoip
php-log
php-lua
php-mapscript
php-net-ipv6
php-nrk-predis
php-propro
php-radius
php-raphf-dev
php-sass
php-symfony-debug
php-symfony-security
php-symfony-web-server-bundle
php-xajax
php-xml-htmlsax3
php-xmlrpc
php7.4
php7.4-bcmath
php7.4-bz2
php7.4-cgi
php7.4-dba
php7.4-dev
php7.4-enchant
php7.4-gmp
php7.4-imap
php7.4-interbase
php7.4-ldap
php7.4-odbc
php7.4-pgsql
php7.4-phpdbg
php7.4-pspell
php7.4-snmp
php7.4-soap
php7.4-sqlite3
php7.4-sybase
php7.4-tidy
php7.4-xmlrpc
php7.4-xsl

```

In between I reinstalled nginx (`dietpi-software reinstall 79`)  and nextcloud (`dietpi-software reinstall 114`). But still no luck with the nextcloud instance.

```
root@DietPi:~# journalctl -u mariadb-server -u php8.2-fpm -u nginx -u redis
Aug 19 21:06:20 DietPi systemd[1]: Starting php8.2-fpm.service - The PHP 8.2 FastCGI Process Manager...
Aug 19 21:06:26 DietPi php-fpm8.2[868]: [19-Aug-2023 21:06:26] NOTICE: PHP message: PHP Warning:  IcePHP_Properties::__toString() implemented without string return type in Unknown on line 0
Aug 19 21:06:26 DietPi php-fpm8.2[868]: [19-Aug-2023 21:06:26] NOTICE: PHP message: PHP Warning:  Ice\ObjectPrx::__toString() implemented without string return type in Unknown on line 0
Aug 19 21:06:26 DietPi php-fpm8.2[868]: [19-Aug-2023 21:06:26] NOTICE: PHP message: PHP Warning:  IcePHP_Logger::__toString() implemented without string return type in Unknown on line 0
Aug 19 21:06:26 DietPi php-fpm8.2[868]: [19-Aug-2023 21:06:26] NOTICE: PHP message: PHP Warning:  IcePHP_Endpoint::__toString() implemented without string return type in Unknown on line 0
Aug 19 21:06:26 DietPi php-fpm8.2[868]: [19-Aug-2023 21:06:26] NOTICE: PHP message: PHP Warning:  IcePHP_Connection::__toString() implemented without string return type in Unknown on line 0
Aug 19 21:06:26 DietPi systemd[1]: Started php8.2-fpm.service - The PHP 8.2 FastCGI Process Manager.
Aug 19 21:06:26 DietPi systemd[1]: Starting nginx.service - A high performance web server and a reverse proxy server...
Aug 19 21:06:26 DietPi systemd[1]: Started nginx.service - A high performance web server and a reverse proxy server.

```

MichaIng, 2023-08-19:
All PHP 7.4 packages are still there. I thought you purged them previously? Did one of those commands throw an error?
```sh
apt-mark unhold $(apt-mark showhold)
G_AGP '*php7.4*'
```

rtglobal, 2023-08-19:
```
root@DietPi:~# apt-mark unhold $(apt-mark showhold)
dpkg: Warnung: Paket nicht in der »status«- oder »available«-Datenbank in Zeile 1: nginx-core:all
dpkg: Warnung: Paket nicht in der »status«- oder »available«-Datenbank in Zeile 2: php-nrk-predis:all
dpkg: Warnung: Paket nicht in der »status«- oder »available«-Datenbank in Zeile 3: php-xmlrpc:arm64
dpkg: Warnung: unbekannte Pakete wurden gefunden; dies könnte bedeuten, dass die
»available«-Datenbank veraltet ist und durch eine Frontend-Methode
aktualisiert werden muss; bitte ziehen Sie die FAQ
<https://wiki.debian.org/Teams/Dpkg/FAQ#set-selections> zu Rate.
Halten-Markierung für nginx-core entfernt.
Halten-Markierung für php-nrk-predis entfernt.
Halten-Markierung für php-xmlrpc entfernt.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-apcu-bc« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-doctrine-bundle« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-facedetect« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-geoip« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-log« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-lua« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-mapscript« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-net-ipv6« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-propro« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-radius« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-raphf-dev« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-sass« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-symfony-debug« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-symfony-security« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-symfony-web-server-bundle« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-xajax« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-xml-htmlsax3« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-bcmath« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-bz2« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-cgi« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-dba« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-dev« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-enchant« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-gmp« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-imap« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-interbase« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-ldap« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-odbc« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-pgsql« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-phpdbg« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-pspell« ausgewählt werden, da beide nicht existieren.
[... cut ...]
Aug 19 21:06:20 DietPi mariadbd[744]: Version: '10.11.3-MariaDB-1'  socket: '/run/mysqld/mysqld.sock'  port: 3306  Debian 12
Aug 19 21:06:20 DietPi systemd[1]: Started mariadb.service - MariaDB 10.11.3 database server.
Aug 19 21:06:20 DietPi /etc/mysql/debian-start[856]: Upgrading MySQL tables if necessary.
Aug 19 21:06:22 DietPi /etc/mysql/debian-start[910]: Triggering myisam-recover for all MyISAM tables and aria-recover for all Aria tables
Aug 19 21:06:26 DietPi mariadbd[744]: 2023-08-19 21:06:26 0 [Note] InnoDB: Buffer pool(s) load completed at 230819 21:06:26

```

rtglobal, 2023-08-19:
In between I updated my nextcloud server to the newest and highest available version. But still no luck when I try to call the local ip or global URL in my browser. 

```
root@DietPi:~# sudo -u www-data php /var/www/nextcloud/updater/updater.phar 
Nextcloud Updater - version: v26.0.0beta2-9-g64e2e4c dirty

Current version is 26.0.5.

Update to Nextcloud 27.0.2 available. (channel: "stable")
.
.
.
Update successful

```

MichaIng, 2023-08-19:
The services seem to be all running. I have never seen this kind of APT database mismatch/error. The link from the error message contains some info: https://wiki.debian.org/Teams/Dpkg/FAQ#set-selections
But lets try to clear the cache first, do a fresh update and see whether it is still required:
```sh
/boot/dietpi/func/dietpi-set_software apt clean
apt update
apt-mark showhold
apt-mark unhold $(apt-mark showhold)
```
If the same errors are thrown, let's try the first solution from the guide:
```sh
apt-cache dumpavail | dpkg --merge-avail
apt-mark unhold $(apt-mark showhold)
```
The `nginx-core` package is also in this unexpected list, and it is actually not available anymore since Bookworm (has been merged with `nginx`). Probably it causes issues, so let's clear these strange APT/dpkg errors first.

rtglobal, 2023-08-19:
```
root@DietPi:~# /boot/dietpi/func/dietpi-set_software apt clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt (clean)

[  OK  ] DietPi-Set_software | rm -Rf /tmp/apt /var/cache/apt /var/lib/apt/lists /var/lib/apt/mirrors /var/lib/apt/periodic /var/lib/apt/cdroms.list
[  OK  ] DietPi-Set_software | mkdir -p /var/cache/apt/archives
[  OK  ] apt clean | Completed
root@DietPi:~# apt update
Holen:1 https://repos.influxdata.com/debian bullseye InRelease [7.040 B]
Holen:2 https://deb.debian.org/debian bookworm InRelease [151 kB]                                                                                    
Holen:3 https://apt.grafana.com stable InRelease [5.984 B]                                                                                           
Holen:4 https://deb.debian.org/debian bookworm-updates InRelease [52,1 kB]     
Holen:5 https://archive.raspberrypi.org/debian bullseye InRelease [23,6 kB]
Holen:6 https://deb.debian.org/debian-security bookworm-security InRelease [48,0 kB]
Holen:7 https://deb.debian.org/debian bookworm-backports InRelease [54,7 kB]
Holen:8 https://repos.influxdata.com/debian bullseye/stable arm64 Packages [5.573 B]
Holen:9 https://deb.debian.org/debian bookworm/contrib arm64 Packages [45,1 kB]
Holen:10 https://deb.debian.org/debian bookworm/non-free arm64 Packages [76,1 kB]
Holen:11 https://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages [5.856 B]
Holen:12 https://deb.debian.org/debian bookworm/main arm64 Packages [8.803 kB]
Holen:13 https://apt.grafana.com stable/main arm64 Packages [98,6 kB]
Holen:14 https://deb.debian.org/debian bookworm-updates/main arm64 Packages [4.736 B]
Holen:15 https://deb.debian.org/debian-security bookworm-security/main arm64 Packages [57,0 kB]
Holen:16 https://archive.raspberrypi.org/debian bullseye/main arm64 Packages [308 kB]
Holen:17 https://deb.debian.org/debian bookworm-backports/contrib arm64 Packages [1.580 B]
Holen:18 https://deb.debian.org/debian bookworm-backports/main arm64 Packages [81,1 kB]
Es wurden 9.829 kB in 6 s geholt (1.637 kB/s).                                                                                                       
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Alle Pakete sind aktuell.
root@DietPi:~# apt-mark showhold
nginx-core
php-apcu-bc
php-doctrine-bundle
php-facedetect
php-geoip
php-log
php-lua
php-mapscript
php-net-ipv6
php-nrk-predis
php-propro
php-radius
php-raphf-dev
php-sass
php-symfony-debug
php-symfony-security
php-symfony-web-server-bundle
php-xajax
php-xml-htmlsax3
php-xmlrpc
php7.4
php7.4-bcmath
php7.4-bz2
php7.4-cgi
php7.4-dba
php7.4-dev
php7.4-enchant
php7.4-gmp
php7.4-imap
php7.4-interbase
php7.4-ldap
php7.4-odbc
php7.4-pgsql
php7.4-phpdbg
php7.4-pspell
[... 34 lines cut ...]
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php-xml-htmlsax3« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-bcmath« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-bz2« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-cgi« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-dba« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-dev« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-enchant« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-gmp« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-imap« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-interbase« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-ldap« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-odbc« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-pgsql« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-phpdbg« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-pspell« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-snmp« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-soap« ausgewählt werden, da beide nicht existieren.
[... cut ...]
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-sybase« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-tidy« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-xmlrpc« ausgewählt werden, da beide nicht existieren.
E: Es kann weder eine installierte Version noch ein Installationskandidat von Paket »php7.4-xsl« ausgewählt werden, da beide nicht existieren.

```

rtglobal, 2023-08-19:
When I try to connect to my nextcloud server via mobile apps like "Nextcloud" and "Talk"-App on Android, then there is also no connection: "Server unavailable".

MichaIng, 2023-08-19:
Okay, there are three databases with different information. Let's check where exactly those packages are present:
```sh
sed -n '/^Package: php7.4$/,/^$/p' /var/lib/dpkg/status
sed -n '/^Package: php7.4$/,/^$/p' /var/lib/dpkg/available
sed -n '/^Package: php7.4$/,/^$/p' /var/lib/apt/extended_states
```
The first contains the hold status. But it contains (should so) only installed packages, hence I have no idea how it is possible that an uninstalled package can have a hold state saved in any of them.

rtglobal, 2023-08-19:
```
root@DietPi:~# sed -n '/^Package: php7.4$/,/^$/p' /var/lib/dpkg/status
Package: php7.4
Status: hold ok not-installed
Architecture: all

root@DietPi:~# sed -n '/^Package: php7.4$/,/^$/p' /var/lib/dpkg/available
Package: php7.4
Maintainer: Dummy Example <dummy@example.org>
Architecture: all
Version: 0~
Description: dummy package record
 A record is needed to put a package on hold, so here it is.

root@DietPi:~# sed -n '/^Package: php7.4$/,/^$/p' /var/lib/apt/extended_states
```

MichaIng, 2023-08-19:
Okay, these look like intended dummy entries to store the hold state for not-installed packages. And at the same time, these hold states cannot be removed if the package is not installed and not available anymore. Why exactly is it possible to cause such a situation, i.e. why does APT just do what it is asked for and remove those entries, regardless whether they are still in any repo or not, or better especially if they are not available in any repo anyway 😄.

Let's temporarily re-add the Bullseye repo, which should allow unholding the packages:
```sh
echo 'deb https://deb.debian.org/debian bullseye main' > /etc/apt/sources.list.d/bullseye.list
apt update
apt-mark unhold $(apt-mark showhold)
apt-mark showhold
```

rtglobal, 2023-08-19:
```
root@DietPi:~# echo 'deb https://deb.debian.org/debian bullseye main' > /etc/apt/sources.list.d/bullseye.list
root@DietPi:~# apt update
OK:1 https://deb.debian.org/debian bookworm InRelease
OK:2 https://deb.debian.org/debian bookworm-updates InRelease                                                                                        
OK:3 https://repos.influxdata.com/debian bullseye InRelease                                                                                          
OK:4 https://deb.debian.org/debian-security bookworm-security InRelease                                                                              
OK:5 https://deb.debian.org/debian bookworm-backports InRelease                                                                             
Holen:6 https://deb.debian.org/debian bullseye InRelease [116 kB]                                                                           
OK:7 https://apt.grafana.com stable InRelease                                                                                               
OK:8 https://archive.raspberrypi.org/debian bullseye InRelease                
Holen:9 https://deb.debian.org/debian bullseye/main arm64 Packages [8.071 kB]
Es wurden 8.187 kB in 6 s geholt (1.284 kB/s).                                                                                                       
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Alle Pakete sind aktuell.
root@DietPi:~# apt-mark unhold $(apt-mark showhold)
Halten-Markierung für nginx-core entfernt.
Halten-Markierung für php-apcu-bc entfernt.
Halten-Markierung für php-doctrine-bundle entfernt.
Halten-Markierung für php-facedetect entfernt.
Halten-Markierung für php-geoip entfernt.
Halten-Markierung für php-log entfernt.
Halten-Markierung für php-lua entfernt.
Halten-Markierung für php-mapscript entfernt.
Halten-Markierung für php-net-ipv6 entfernt.
Halten-Markierung für php-nrk-predis entfernt.
Halten-Markierung für php-propro entfernt.
Halten-Markierung für php-radius entfernt.
Halten-Markierung für php-raphf-dev entfernt.
Halten-Markierung für php-sass entfernt.
Halten-Markierung für php-symfony-debug entfernt.
Halten-Markierung für php-symfony-security entfernt.
Halten-Markierung für php-symfony-web-server-bundle entfernt.
Halten-Markierung für php-xajax entfernt.
Halten-Markierung für php-xml-htmlsax3 entfernt.
Halten-Markierung für php-xmlrpc entfernt.
Halten-Markierung für php7.4 entfernt.
Halten-Markierung für php7.4-bcmath entfernt.
Halten-Markierung für php7.4-bz2 entfernt.
Halten-Markierung für php7.4-cgi entfernt.
Halten-Markierung für php7.4-dba entfernt.
Halten-Markierung für php7.4-dev entfernt.
Halten-Markierung für php7.4-enchant entfernt.
Halten-Markierung für php7.4-gmp entfernt.
Halten-Markierung für php7.4-imap entfernt.
Halten-Markierung für php7.4-interbase entfernt.
Halten-Markierung für php7.4-ldap entfernt.
Halten-Markierung für php7.4-odbc entfernt.
Halten-Markierung für php7.4-pgsql entfernt.
Halten-Markierung für php7.4-phpdbg entfernt.
Halten-Markierung für php7.4-pspell entfernt.
Halten-Markierung für php7.4-snmp entfernt.
Halten-Markierung für php7.4-soap entfernt.
Halten-Markierung für php7.4-sqlite3 entfernt.
Halten-Markierung für php7.4-sybase entfernt.
Halten-Markierung für php7.4-tidy entfernt.
Halten-Markierung für php7.4-xmlrpc entfernt.
Halten-Markierung für php7.4-xsl entfernt.
root@DietPi:~# apt-mark showhold
root@DietPi:~# 

```

MichaIng, 2023-08-19:
Okay, now removing it again:
```sh
rm /etc/apt/sources.list.d/bullseye.list
/boot/dietpi/func/dietpi-set_software apt clean
```
So at least these cannot be responsible for the Nextcloud access issues. Can you access it locally?
```sh
curl -IL 127.0.0.1/nextcloud
```
If you use HTTPS, try the same after replacing `127.0.0.1` with the actual domain name.

rtglobal, 2023-08-19:
```
root@DietPi:~# rm /etc/apt/sources.list.d/bullseye.list
root@DietPi:~# /boot/dietpi/func/dietpi-set_software apt clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt (clean)

[  OK  ] DietPi-Set_software | rm -Rf /tmp/apt /var/cache/apt /var/lib/apt/lists /var/lib/apt/mirrors /var/lib/apt/periodic /var/lib/apt/cdroms.list
[  OK  ] DietPi-Set_software | mkdir -p /var/cache/apt/archives
[  OK  ] apt clean | Completed
```

rtglobal, 2023-08-20:
Good news from my side!! I renewed the https-certificates with your `dietpi-letsencrypt ` command. And now everything seems to be ok. I can login to my nextcloud, all files and data are there, no errors at the inside system-check. 
I am very happy. Thanks a lot for your extended help. I will donate some coins via patreon for your fantastic work.

Joulinar, 2023-08-20:
@rtglobal thx for sharing. Good you are back online

@MichaIng do you like to keep the issue open for adjustments on the Bookworm update script??

MichaIng, 2023-08-20:
Does Certbot on Bullseye probably create an Nginx SSL config which is not compatible with the Nginx version on Bookworm? The syntax must be correct, as otherwise the server would not start, but probably some format, cipher or any such is not supported (anymore). The connection seems to fail exactly when the first HTTPS request is attempted. If the certificate was expired, there would be a different error message.

About the nasty package hold thing: Looks like we should unhold all packages before adjusting the list files. There are potential strong reasons to keep individual packages on hold, especially with 3rd party repos, but on the other hand, a distro upgrade is a major thing as well, one should have a backup or mentioned hold DEB files available somewhere, and often such mark becomes obsolete when the distro upgrade resolves the underlying reason for the hold state.

MichaIng, 2023-08-26:
I'll mark this as closed. Will keep the Certbot Bullseye vs Bookworm question in mind, in case others report a similar issue, and in case add a prompt for a cert update.

closed 2023-08-26 by MichaIng