MichaIng/DietPi #6562: Amiberry missing in DietPi_RPi-ARMv8-Bullseye_Amiberry

mibosshard, 2023-08-20:
DietPi_RPi-ARMv8-Bullseye_Amiberry.7z from 2023-07-31 is missing Amiberry

MichaIng, 2023-08-20:
Many thanks for your report.

Amiberry should be installed automatically on first boot. Is this not the case?

mibosshard, 2023-08-20:
this is super strange! I moved on to the Bookworm image and it worked right from the start. to verify, I now have tried it again with the Bullseye image and this time it worked flawless. RPi-400 with Kingston industrial µSD cards (these never gave me trouble so far). case closed! sorry for bothering you... and thanks a lot for getting back to me this quick!

closed 2023-08-20 by Joulinar

MichaIng, 2023-08-20:
Okay great. Probably there was just a temporary network or server issue or such, which made the install fail or hang.