MichaIng/DietPi #6564: Open Beta v8.21 | Please help testing and hardening our upcoming release

MichaIng, 2023-08-20:
RC version | v8.21.0
----------- | -------
v8.21.0 PR | #6563
Raw changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release date | **2023-08-26**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av8.21
____
### Beta v8.21.0
_(2023-08-20)_

#### New images
- Orange Pi 5 Plus | Support for the Orange Pi 5 Plus SBC has been added to DietPi with hardware ID 82. Images will be provided on our download page: https://dietpi.com/#download

#### Enhancements
- Orange Pi 5/ROCK 5B | An update of the kernel to Linux 5.10.160 will be applied automatically as part of the DietPi update.
- DietPi-Config | When applying WiFi hotspot settings, manually changed IP addresses in /etc/network/interfaces will now be preserved.
- DietPi-Config | WiFi Hotspot: Toggles for enabling 802.11n/WiFi 4, 802.11ac/WiFi 5 and 5 GHz WiFi respectively have been added to the menu.
- DietPi-Software | WiFi Hotspot: We added new dietpi.txt settings `SOFTWARE_WIFI_HOTSPOT_WIFI4`, `SOFTWARE_WIFI_HOTSPOT_WIFI5` and `SOFTWARE_WIFI_HOTSPOT_5G` to have 802.11n/WiFi 4, 802.11ac/WiFi 5 and 5 GHz support respectively enabled on install.

#### Bug fixes
- DietPi-Globals | G_GET_NET: Resolved an issue where the default route was not detected correctly if learned via OPSF or other routing protocols which change the output field numbers of the `ip route` command. Many thanks to @owendelong for reporting this issue: https://github.com/MichaIng/DietPi/issues/6561
- DietPi-Installer | Resolved an issue where building images for ARMv7 hardware was not possible on a ARMv8 host. Offered/accepted ARM hardware IDs are not based on the userland/OS architecture of the image instead of the CPU/kernel architecture returned via `uname -m`. Many thanks to @dirkhh for reporting this issue: https://github.com/MichaIng/DietPi/discussions/6541
- DietPi-Set_swapfile | Resolved an issue on Bookworm systems where zram swap space was not enabled automatically on boot due to missing syscall permissions, and the swappiness was not changed as intended. Many thanks to @magicfoxt-magicfox for reporting this issue: https://github.com/MichaIng/DietPi/issues/6511
- DietPi-LetsEncrypt | Resolved a v8.20 regression where Lighttpd did not start after applying or renewing the certificate due to a syntax error. Many thanks to @JappeHallunken for fixing this issue: https://github.com/MichaIng/DietPi/pull/6517
- DietPi-Config | Resolved an issue where /etc/network/interfaces was created with missing key values if lines were previously manually removed. Defaults are now applied in this case. Many thanks to `@huettenwirt` for reporting this issue: https://dietpi.com/forum/t/ifup-error-due-to-faulty-etc-network-interfaces/17605
- DietPi-Config | Resolved an issue where the DHCP server failed to start after applying WiFi hotspot settings.
- DietPi-Software | TasmoAdmin: Resolved an issue on Bookworm systems with Nginx and Lighttpd webserver where login and logout did not work anymore as the rewrite directives pointed to a file which does not exist anymore since TasmoAdmin v3. Many thanks to `@TBirth` for reporting this issue: https://dietpi.com/forum/t/tasmoadmin-not-found-after-visiting-login-page/17632

lukaszsobala, 2023-08-22:
This might be an obvious question but why only WiFi4 and WiFi5 are the options for the hotspot? Is a WiFi6 hotspot not possible? A lot of hardware does support it already (although most commonly the dedicated wifi cards connected to compatible boards).

MichaIng, 2023-08-22:
Just because I have no WiFi 6 adapter to test it and I do not see a way to enable the 6 GHz band (WiFi 6E) and it is surely not yet supported by the `hostapd` version on Buster and probably not even Bullseye. However, I'm working on a few smaller enhancements to the hotspot menu, so when I find time I'll add a WiFi 6 toggle as well. Would be great if you could then test it.

lukaszsobala, 2023-08-24:
I could test it on Radxa Rock 5A or 5B.

MichaIng, 2023-08-24:
@lukaszsobala 
I opened a PR to add a WiFi 6 toggle: #6575
Would be great if you could test it. You can quickly switch to this branch with this command:
```sh
G_DEV_BRANCH wifi6
```

MichaIng, 2023-08-26:
#### DietPi Beta v8.21.1 has been merged: #6579

MichaIng, 2023-08-26:
#### DietPi v8.21 has been released: #6580
Many thanks to all testers ❤️ !

closed 2023-08-26 by MichaIng

lukaszsobala, 2023-09-06:
> @lukaszsobala 
> I opened a PR to add a WiFi 6 toggle: #6575
> Would be great if you could test it. You can quickly switch to this branch with this command:
> ```sh
> G_DEV_BRANCH wifi6
> ```

I am trying to test this but the isc-dhcp-server fails to start. I have sent a report, please take a look