MichaIng/DietPi #6566: Make dietpi-config multiple-interface aware

owendelong, 2023-08-21:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
Currently dietpi-config assumes eth0 (ethernet) and wlan0 (wifi) are the only possible network interfaces

#### Describe the solution you'd like:
It would be nice if dietpi-config were expanded to support additional interfaces (such as on NanoPI R6S and/or USB Wired adapters)

#### Describe alternatives you've considered:
I have not considered any alternatives. For now, I use /etc/network/interfaces and vi to do the network configuration and dietpi-config is none the wiser.

#### Additional context
None I can think of.

Joulinar, 2023-08-21:
Basically there is a long term plan to rework the whole network part https://github.com/users/MichaIng/projects/3

owendelong, 2023-08-21:
I think that URL needs to get reworked first... I get a 404.

Joulinar, 2023-08-21:
Ah I see, URL is correct but the Projekt is not public. Therefore, you are not able to see it. Anyway, the rework of the network configuration is already on the list.

MichaIng, 2023-08-22:
I didn't even know that projects are private by default. I just made them all public 😄.

Kevin-Nexio, 2023-12-15:
Hello, I really need to be able to manage the 3 network interfaces of my R5S...
Do you have any idea when these modifications will be implemented?
And in the meantime, is there an alternative so that I can activate the interfaces in DHCP and have them accept hot-plug (no need to reboot the DietPI when connecting an RJ45)?

Joulinar, 2023-12-15:
no ETA from our side. However, you can do all this configuration manually within configuration files directly.

Kevin-Nexio, 2023-12-15:
Ok, where can I find documentation about the manually configuration ?

MichaIng, 2023-12-15:
It is the standard Debian network stack `ifupdown`: https://manpages.debian.org/interfaces

Easiest is probably to setup the internet facing interface via `dietpi-config` and the other interfaces via dedicated config files in `/etc/network/interfaces.d`, which are imported automatically. That way they are preserved even when you change network settings in `dietpi-config`.

Kevin-Nexio, 2023-12-18:
Hi, I've tried several ways but nothing works. I can't get several interfaces to work in DHCP at the same time (without having to restart the service each time I make a change). If someone can help me, I'd love to hear from you.

MichaIng, 2023-12-19:
> without having to restart the service each time I make a change

But that is normal when using the `networking` or `ifup@` services. The alternative is to run `ifup` manually, or directly `dhclient`:
- https://manpages.debian.org/ifup
- https://manpages.debian.org/dhclient

Kevin-Nexio, 2023-12-19:
Ok thanks for the information, if I want to use the dhclient do I need to disable something?

owendelong, 2023-12-20:
Assuming a NanoPi R6S (3 ethernet interfaces eth0, eth1, eth2), it should be as simple as putting the following into /etc/network/interfaces

allow-hotplug eth0
iface eth0 inet6 auto
iface eth0 inet dhcp

allow-hotplug eth1
iface eth1 inet6 auto
iface eth1 inet dhcp

allow-hotplug eth2
iface eth2 inet6 auto
iface eth2 inet dhcp

MichaIng, 2023-12-20:
> it should be as simple as putting the following into /etc/network/interfaces

Better create separate configs into `/etc/network/interfaces.d` for those interfaces you do not use for Internet access. E.g. let `dietpi-config` setup `eth0`, and setup the other two for service as gateway for your LAN, or whatever you want to do with them. 3x DHCP, i.e. different routers/DHCP servers on 3 different networks looks like a pretty uncommon use case? 😄

Ah, and the `inet6` entries are generally redundant. SLAAC is done anyway automatically, no need to explicitly tell the network stack to do that.

> if I want to use the dhclient do I need to disable something?

Only assure that the interfaces you are running the daemon for are not present in `/etc/network/interfaces`, or only as `iface ethX inet manual`. With `manual`, the interfaces is enabled (carrier signal starts), but no address (aside of SLAAC) or route assigned. Not sure if `dhclient` does this as well, before (trying to) sending a DHCP broadcast.

owendelong, 2023-12-20:
> On Dec 20, 2023, at 08:46, MichaIng ***@***.***> wrote:
> 
> 
> it should be as simple as putting the following into /etc/network/interfaces
> 
> Better create separate configs into /etc/network/interfaces.d for those interfaces you do not use for Internet access. E.g. let dietpi-config setup eth0, and setup the other two for service as gateway for your LAN, or whatever you want to do with them. 3x DHCP, i.e. different routers/DHCP servers on 3 different networks looks like a pretty uncommon use case? 😄
> 
> Ah, and the inet6 entries are generally redundant. SLAAC is done anyway automatically, no need to explicitly tell the network stack to do that.
> 
> if I want to use the dhclient do I need to disable something?
> 
> Only assure that the interfaces you are running the daemon for are not present in /etc/network/interfaces, or only as iface ethX inet manual. With manual, the interfaces is enabled (carrier signal starts), but no address (aside of SLAAC) or route assigned. Not sure if dhclient does this as well, before (trying to) sending a DHCP broadcast.
> 

Until recently, dhclient was what happened when you set auto.

Now, systems does the dhcp client function as ISC dhclient is being deprecated (along with ISC DHCP Server).

Replacement for ISC DHCP Server is Kea, which is actually pretty nice. Huge improvement over ISC DHCP server.

I don’t know of a replacement for dhclient other than what’s built into systems and/or dhcpcd, neither of which is super appealing compared to dhclient’s robust featureset.

Owen

MichaIng, 2023-12-20:
> Until recently, dhclient was what happened when you set auto.

Since the years I am dealing with this, "auto" always meant SLAAC for IPv6 only, nothing else. `dhclient` is and was only involved when using "dhcp". Though the "auto" option has another "dhcp" sub-option for stateless DHCP (compared to "dhcp" being stateful  DHCP). Not sure if this invokes `dhclient`. See the man page regarding this: https://manpages.debian.org/ifupdown/interfaces#The_auto_Method

> Now, systems does the dhcp client function as ISC dhclient is being deprecated (along with ISC DHCP Server).

Good hint. The server is not deprecated (yet), but the client and relay: https://www.isc.org/blogs/dhcp-client-relay-eom/

Checking the docs, `dhcpcd` is supported as well by `ifupdown`, and `udhcpc`, the BusyBox DHCP client. For some more (optional) flexibility, and since the rest of BusyBox is not required, `dhcpcd` seems to be the better choice. I will test whether it can be just replaced by removing one and installing the other. However, as long as Debian maintains the `isc-dhcp-client` package, it is no urgent task at least. If someone is getting aware of a popular/trusted fork of the original `dhclient`, that would be interesting. But I would generally prefer a package provided by Debian, and there I do not see an alternative to `dhcpcd`, currently.