MichaIng/DietPi #6567: {Query} is there a way to force a screen resolution with an Odroid C4

MDAR, 2023-08-21:
Hi

So up till now, every Odroid C4 I've ever plugged in has happily configured itself to the highest screen resolution available from a screen 

Last week, I plugged in a new machine to a new a Iiyama 4K screen.

The Odroid read the EDID and saw it as a 4K screen, but setup as 1920x1080

On a reboot, it set to 4K

Next reboot, back to 1080.

Today, I have a replacement Odroid C4 connected to my own Iiyama 4K, the EDID reads correctly and the output is 4K

```
Screen 0: minimum 320 x 200, current 3840 x 2160, maximum 3840 x 2160
HDMI-1 connected primary 3840x2160+0+0 (normal left inverted right x axis y axis) 941mm x 529mm

3840x2160 30.00*+ 29.97
1920x1080 60.00 50.00 59.94
1920x1080i 60.00 50.00 59.94
1680x1050 59.88 
1280x1024 60.02 
1440x900 59.90
1152x864 75.00
1280x720 60.00 50.00 59.94
1024x768 60.00
832x624 74.55
800x600 60.32
720x576 50.00
720x480 59.94
640x480 75.00 66.67
```

I saw that the XU4 has an override option in the boot.ini, is it possible to use the same (type of) command to force the resolution with the C4?

https://github.com/MichaIng/DietPi/blob/17a32753cea78821406556b1e0738eff8c6a526e/.build/images/OdroidXU4/boot.ini#L78

MDAR, 2023-08-22:
Hi
As an update.

There is something very strange going on when an Odroid C4 is connected to a particular Iiyama 4K screen.

When I tested it with my own Iiyama 4K, the Odroid happily set to 4K.

But when it got to the clients and they plugged it in, it set to 1024.

Even though Xrandr sees the EDID as 4K

Is there anything I can do to override this?

MichaIng, 2023-08-22:
The C4 uses mainline Linux, so it does not support those non-standard vendor boot config keys available for XU4.

Probably it works to enforce a mode via kernel command-line. Can you check for the name of the DRM device?
```sh
for i in /sys/class/drm/*; do [[ -e $i/status ]] && echo "$i: $(<$i/status)"; done
```

MDAR, 2023-08-24:
Hi

Thanks for that.

I think I may have stumbled across the real issue.

Based on an observation of my test rig, whereby if I power up my Odroid C4 dev unit "before" I power up my FeelWorld 4K screeen, the Odroid will struggle setting the EDID correctly.

(even through HDMI spec suggests that EDID handshake should happen at any point.)

Setting the resolution at anything between 1024 and 4K.

**********

Today I remotely connected to my customer's machine and using some Node-RED magic, I can see that he Iiyama screen is ON and set to a different HDMI input (The CCTV NVR)

When running any command on the Odroid to query the connected display, it simply suggests that there isn't a screen attached.

Switching input on the Iiyama and waiting, means that the Odroid can see a screen, but even through it sees it as a 4K, it only sends 1024.

As I need a fairly quick but PERFECT switch between inputs, I'm going to put a HDMI splitter and EDID manager in line, so that the Odroid thinks it has a 4K screen attached right from boot up and 24/7.

************

Doesn't address the incorrect EDID handshake issue, but it does resolve my bigger issues.

**********

Would you like me to continue looking into this issue with my dev setup?

MDAR, 2023-08-31:
> As I need a fairly quick but PERFECT switch between inputs, I'm going to put a HDMI splitter and EDID manager in line, so that the Odroid thinks it has a 4K screen attached right from boot up and 24/7.

I know this isn't a "great solution", but it is working perfectly, every time.

For those that are interested

Very specifically this device, set to **Copy EDID** from output 1 and force output 2 to follow.

https://www.amazon.co.uk/OREI-HD-Outputs-Splitter-Certified-Black/dp/B005HXFARS

With a Dummy 4K HDMI plug in output 1, to give the Odroid a 4K EDID to work with, 24/7, regardless of the state of the true monitor connected to output 2.

Something / anything like this

https://www.amazon.co.uk/Headless-Display-Emulator-Headless-1920x1080-generation/dp/B06XT1Z9TF/f=asc_df_B06XT1Z9TF/

MichaIng, 2023-09-07:
Let me reopen this as I generally want to test enforcing screen resolutions via kernel command-line (of userland tool fallback) it EDID detection does not work correctly or native resolution is not wanted, probably including screen rotation.

MDAR, 2023-09-08:
> Let me reopen this as I generally want to test enforcing screen resolutions via kernel command-line (of userland tool fallback) it EDID detection does not work correctly or native resolution is not wanted, probably including screen rotation.

If there's anything you want me to test, please let me know

MDAR, 2023-09-08:
> ```shell
> or i in /sys/class/drm/*; do [[ -e $i/status ]] && echo "$i: $(<$i/status)"; done
> ```

This command didn't work, but...

There is a file called `version` in the /sys/class/drm folder, that only contains this text :-

```drm 1.1.0 20060810```

MichaIng, 2025-02-09:
We added a new tool to control console modes and rotation via DRM/KMS sysfs and `video` kernel command-line parameter: #7390
I'll close this issue in favour of the new one.

closed 2025-02-09 by MichaIng