MichaIng/DietPi #6568: Update to 8.20.1 but problem with Openhub repository

Joulinar, 2023-08-22:
@MichaIng

I try to update tu 8.20.1 but i have problem with Openhub repository

- ...
#### Additional logs:

```
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
Get:3 https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable InRelease [12.8 kB]
Get:4 https://openhab.jfrog.io/openhab/openhab-linuxpkg stable InRelease [12.8 kB]
Err:3 https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable InRelease
  The following signatures were invalid: EXPKEYSIG 075721F6A224060A openHAB Bintray Repositories <owner@openhab.org>
Err:4 https://openhab.jfrog.io/openhab/openhab-linuxpkg stable InRelease
  The following signatures were invalid: EXPKEYSIG 075721F6A224060A openHAB Bintray Repositories <owner@openhab.org>
Fetched 25.6 kB in 2s (15.1 kB/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://openhab.jfrog.io/artifactory/openhab-linuxpkg stable InRelease: The following signatures were invalid: EXPKEYSIG 075721F6A224060A openHAB Bintray Repositories <owner@openhab.org>
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://openhab.jfrog.io/openhab/openhab-linuxpkg stable InRelease: The following signatures were invalid: EXPKEYSIG 075721F6A224060A openHAB Bintray Repositories <owner@openhab.org>
E: Failed to fetch https://openhab.jfrog.io/artifactory/openhab-linuxpkg/dists/stable/InRelease  The following signatures were invalid: EXPKEYSIG 075721F6A224060A openHAB Bintray Repositories <owner@openhab.org>
E: Failed to fetch https://openhab.jfrog.io/openhab/openhab-linuxpkg/dists/stable/InRelease  The following signatures were invalid: EXPKEYSIG 075721F6A224060A openHAB Bintray Repositories <owner@openhab.org>
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

_Originally posted by @maury77 in https://github.com/MichaIng/DietPi/issues/6444#issuecomment-1687773223_

Joulinar, 2023-08-22:
I guess you need to fresh the PGP key for OpenHub repository 

https://next.openhab.org/docs/installation/linux.html#apt-based-systems

Joulinar, 2023-08-22:
Did you install OpenHub via DietPi-Sotware?

maury77, 2023-08-22:
openhub is not installed , there are old installation parameter of the repository

I remove It , now all ok

closed 2023-08-22 by Joulinar