MichaIng/DietPi #6569: Trying to install Python

ISquishWorms, 2023-08-22:
#### Details:
- Date           | Tue Aug 22 12:38:49 BST 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux RaspberryPi-3B 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bookworm (ID=7,RASPBIAN=0)
- Command        | `apt-get -y --allow-change-held-packages install python3-dev`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

Selected to install Python from the dietpi-software browser.
#### Expected behaviour:
Python should be installed.
#### Actual behaviour:
Python is not installed and errors see additional logs.
#### Extra details:
This is on a fresh install of DietPi which was installed one day prior.
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libc-dev-bin libc6-dev libcrypt-dev libexpat1-dev libjs-jquery
  libjs-sphinxdoc libjs-underscore libnsl-dev libpython3-dev libpython3.11
  libpython3.11-dev libtirpc-dev linux-libc-dev python3-distutils
  python3-lib2to3 python3.11-dev rpcsvc-proto zlib1g-dev
Suggested packages:
  glibc-doc manpages-dev
Recommended packages:
  manpages manpages-dev libc-devtools javascript-common
The following NEW packages will be installed:
  libc-dev-bin libc6-dev libcrypt-dev libexpat1-dev libjs-jquery
  libjs-sphinxdoc libjs-underscore libnsl-dev libpython3-dev libpython3.11
  libpython3.11-dev libtirpc-dev linux-libc-dev python3-dev python3-distutils
  python3-lib2to3 python3.11-dev rpcsvc-proto zlib1g-dev
0 upgraded, 19 newly installed, 0 to remove and 0 not upgraded.
Need to get 11.8 MB of archives.
After this operation, 54.4 MB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bookworm/main arm64 libc-dev-bin arm64 2.36-9+deb12u1 [44.1 kB]
Get:2 https://deb.debian.org/debian bookworm/main arm64 libcrypt-dev arm64 1:4.4.33-2 [121 kB]
Get:3 https://archive.raspberrypi.org/debian bullseye/main arm64 linux-libc-dev arm64 1:1.20230405-1 [1135 kB]
Get:4 https://deb.debian.org/debian bookworm/main arm64 libtirpc-dev arm64 1.3.3+ds-1 [194 kB]
Get:5 https://deb.debian.org/debian bookworm/main arm64 libnsl-dev arm64 1.3.0-2 [66.1 kB]
Get:6 https://deb.debian.org/debian bookworm/main arm64 rpcsvc-proto arm64 1.4.3-1 [59.7 kB]
Get:7 https://deb.debian.org/debian bookworm/main arm64 libc6-dev arm64 2.36-9+deb12u1 [1429 kB]
Get:8 https://deb.debian.org/debian bookworm/main arm64 libexpat1-dev arm64 2.5.0-1 [135 kB]
Get:9 https://deb.debian.org/debian bookworm/main arm64 libjs-jquery all 3.6.1+dfsg+~3.5.14-1 [326 kB]
Get:10 https://deb.debian.org/debian bookworm/main arm64 libjs-underscore all 1.13.4~dfsg+~1.11.4-3 [116 kB]
Get:11 https://deb.debian.org/debian bookworm/main arm64 libjs-sphinxdoc all 5.3.0-4 [130 kB]
Get:12 https://deb.debian.org/debian bookworm/main arm64 libpython3.11 arm64 3.11.2-6 [1841 kB]
Get:13 https://deb.debian.org/debian bookworm/main arm64 zlib1g-dev arm64 1:1.2.13.dfsg-1 [913 kB]
Get:14 https://deb.debian.org/debian bookworm/main arm64 libpython3.11-dev arm64 3.11.2-6 [4395 kB]
Get:15 https://deb.debian.org/debian bookworm/main arm64 libpython3-dev arm64 3.11.2-1+b1 [9564 B]
Get:16 https://deb.debian.org/debian bookworm/main arm64 python3.11-dev arm64 3.11.2-6 [617 kB]
Get:17 https://deb.debian.org/debian bookworm/main arm64 python3-lib2to3 all 3.11.2-3 [76.3 kB]
Get:18 https://deb.debian.org/debian bookworm/main arm64 python3-distutils all 3.11.2-3 [131 kB]
Get:19 https://deb.debian.org/debian bookworm/main arm64 python3-dev arm64 3.11.2-1+b1 [26.2 kB]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 11.8 MB in 2s (6693 kB/s)
dpkg: unrecoverable fatal error, aborting:
 too-long line or missing newline in '/var/lib/dpkg/diversions'
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

ISquishWorms, 2023-08-22:
I reinstalled DietPi during the install process I changed the Host name and passwords.  I set the eth0 adaptor to Static, disabled the Wifi and IPV6, I also did a check to make sure that the DietPi was online and could see the internet after the changes.  I changed the SSH from Dropbear to OpenSSH and selected I2C and Python to be installed and the install for them completed fine.  I checked the file '/boot/dietpi/.version' and the contents seemed fine, I then decided to reboot so issued the command 'sudo reboot' (yes I used sudo even though I was logged in as root).  On doing this I got the following message in red 'Failed to connect to bus: No such file or directory' and got booted as I guess the Pi was doing as asked and rebooting, only it never reappeared on the network, I tried pinging it and do not get a reply.  So now I just don't know what is going on.

Joulinar, 2023-08-22:
> Failed to connect to bus: No such file or directory

This is expected and not doing any harm. 

Can you connect a screen to check what is going on?

ISquishWorms, 2023-08-22:
I can but it will mean moving the Pi and therefore I will just have to unplug it as I am not able to access it or connect it to a screen in its current location.  Should I do this?

Joulinar, 2023-08-22:
First thing to try is to power cycle. If this did not solve the issue, a screen would be necessary as we don't know where the system stuck.

ISquishWorms, 2023-08-22:
Ok I am going to just turn it off at the wall and bring it into this room so I can hook it up to a monitor to see what is on the screen.

ISquishWorms, 2023-08-22:
Here is a picture of the screen.

[image]

Joulinar, 2023-08-22:
Looks like file system corruption and the system is not able to mount the root file system. You are using a SD card? You have a spare one for testing?

ISquishWorms, 2023-08-22:
I can probably find one think I have a 64gig around here somewhere, it will mean moving some large files though to a network drive for backup. I do not understand why this is happening only after a reboot though?  I did check this SD card for bad blocks when writting the image and it passed.

ISquishWorms, 2023-08-22:
I just did a slow format of the SD card I was using and also checked for bad blocks here is the report from rufus:

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Bad Blocks: Checking from block 0 to 30491 (1 block = 512 KB)
Bad Blocks: Using offset 182946 for fake device check
Bad Blocks: Writing test pattern 0x55
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.9 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 8192 bytes
Slow format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Using Zeroed MBR
Found volume \\?\Volume{7ed46bdc-40cc-11ee-9fa2-b06ebfcdc8c2}\
Created: E:\autorun.inf
Created: E:\autorun.ico
Successfully remounted \\?\Volume{7ed46bdc-40cc-11ee-9fa2-b06ebfcdc8c2}\ as E:

So the card looks to be ok?

ISquishWorms, 2023-08-22:
I decided to write them image again to the same SD card, I made the same changes but did not install any software only swapped dropbear out for OpenSSH.  The exact same file '/boot/dietpi/.version' has junk on the first line.  If this was an SD card issue what are the chances that the exact same file would be corrupted?

ISquishWorms, 2023-08-22:
I am using an original RPI PSU with nothing connected to the PI other than a network cable.  I tried that command but nothing was displayed.  If this was a power issue again what would be the chances that the very same file is corrupted, it would be more random wouldn't it?  I would have thought any errors writting to files would not always result in the same files being corrupted esp after writting the image to the SD on multiple occasions.

ISquishWorms, 2023-08-22:
I have a PI B+ here and a PI 4 maybe i should try burning the image once more to this SD card and try it in one of those?  That will have to be something for tomorrow though.

Joulinar, 2023-08-22:
The file is written during initial update/install process. Hence it has a chance to get corrupted on r/w operations.

Yes you can try the Rpi4. Would be a good comparison. 

In meantime we can ask @MichaIng as he has some further ideas on your issue.

ISquishWorms, 2023-08-22:
Ok will try this SD card with the same DietPi image tomorrow and report back.  Thank you for the time and help so far. :)

Joulinar, 2023-08-22:
I would recommend to download / flash a new fresh image.

ISquishWorms, 2023-08-22:
I am going to do just that, as I thought I would try'DietPi_RPi-ARMv8-Bullseye.7z' on the same SD card and Pi and see if I get the same issues.

ISquishWorms, 2023-08-22:
Well bullseye seems fine, no file corruption in  '/boot/dietpi/.version'.  I am now going to try and install Python and I2C and see if the installs work as expected.  If so then I beleive this to be an issue with Bookworm and not my SD card or RPI.

Joulinar, 2023-08-22:
If it would be the image, I would expect more people reporting such issues. Actually, you are the only one having these symptoms.

ISquishWorms, 2023-08-22:
I spoke too soon getting errors again. :(

       Selecting previously unselected package libpython3.9-minimal:arm64.
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'firmware-linux-free' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)
[FAILED] DietPi-Software | APT install python3-dev
 - Command: apt-get -y --allow-change-held-packages install python3-dev
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait.                                                                                                 ..
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, ple                                                                                                 ase wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 40f872ea-d985-4c8d-8b94-8ef38c873dd4

Joulinar, 2023-08-22:
can you check again for kernel messages

```
dmesg -l 0,1,2,3
```

BTW: do you have any network security gateway inside your local network that could interrupt or inspect traffic? Something like a firewall device?

ISquishWorms, 2023-08-22:
I tried that command but I get nothing:
root@RaspberryPi-3B:/tmp/DietPi-Software# sudo dmesg -l 0,1,2,3
root@RaspberryPi-3B:/tmp/DietPi-Software#

I know a bit about wireshark although I would have to be running that on the RPI with the issues I think to capture the data? Other than that no, I just have a standard ISP provided router and am using another PI for DNS combined with Unbound.

Joulinar, 2023-08-22:
No, I was just asking. I remember a case where a user had a local firewall device that was doing package inspection, causing issues in files downloaded.

ISquishWorms, 2023-08-22:
Hmm maybe my router is doing something odd dunno if so though I could not do much about it as ISP provided routers are very limited with what settings you can change.  Also my other RPI's have been updating fine so unless it is just an issue today or recently appeared then I would have thought they would also struggle to update.

MichaIng, 2023-08-22:
As first the `/var/lib/dpkg/diversions` was corrupted, I guess it is not the network/downloads but indeed SD card write errors. I would try it with a different one. Not uncommon that SD cards are dying, not really made for 24/7 operation.

Btw, in case it was an isolated issue, there are backups of this file and all other dpkg databases in `/var/backups/`.

ISquishWorms, 2023-08-22:
I was being lazy as it would mean moving files to try a different SD card, but having tried the same SD card on my RPI 4 after writting the image again I am still getting errors.  So this the common factor now seems to be the SD card despite rufus not reporting and bad blocks.  I fetch out another SD card and try again and report back once I have done so.

Joulinar, 2023-08-22:
btw, you should be able to boot of a USB pen stick as well. At least if enabled. 🤣 

https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#usb-mass-storage-boot

On the 3B we have a config option. For the 4B some manuell bootloader configs are needed to enable mass storage boot.

MichaIng, 2023-08-22:
I had a similar case where Rufus did not find bad blocks but the SD card definitely caused filesystem corruption regularly, switching to a new one resolved it immediately.

I am not sure whether bad blocks are always reliably found (e.g. one can define a number of write rounds, so it seems to be not necessarily detected on first write attempt) and whether there are other reasons for write errors than a specific broken block.

Indeed booting from USB is actually much better.

> For the 4B some manuell bootloader configs are needed to enable mass storage boot.

Is that still the case? I though after one of the bootloader updates within the first year after RPi 4 release it is supported OOTB?

ISquishWorms, 2023-08-22:
Are USB sticks not slower than SD cards though?  I also was trying to keep things the same as maybe it is an issue with my SD card port?  Although that is good to know and a good option to have, so thank you for mentioning it.

Joulinar, 2023-08-22:
On a 3B not that much difference between USB and SD card. But on a 4B, USB should be faster. BTW I use RPI eMMC module RaspiKey from UUgear. It is not the cheapest option but works better than an SD card.

MichaIng, 2023-08-22:
> Are USB sticks not slower than SD cards though?

In average they are faster. AFAIK the SD card slot on RPi 4 is limited to about 45 MiB/s, while USB potentially supports native USB 3.0 speeds.

ISquishWorms, 2023-08-22:
Well that will teach me for being lazy, I should have listened to you Joulinar when near the start of this topic you suggested trying a different SD card.  That does seem to have fixed the problem.  Note to self don't be lazy and trust Joulinar not Rufus as that reported no bad blocks.  Now I need this test SD card though and was just using it to test, I need to copy the files back onto it.  So maybe I will go the USB stick route.  I guess using a 3B I will not see any speed improvement and maybe less speed at times if I am busy accessing the network but still it saves me buying a new SD card for this RPI which I will not be using much really.

Thank you both again for your time, patience and help.  DietPi and the team rock they make my RPI's worth using.

Joulinar, 2023-08-23:
Hi, can this be closed? I guess you found the issue 🤣

ISquishWorms, 2023-08-23:
> Hi, can this be closed? I guess you found the issue 🤣

It sure can, yes I found the issue.  It was the SD Card combined with a lazy user who delayed trying an alternative card.

closed 2023-08-23 by ISquishWorms

Joulinar, 2023-08-23:
> combined with a lazy user

This is also part of a learning journey