MichaIng/DietPi #6570: [ROCK5B

dclobato, 2023-08-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2023-08-22:
This is correct and expected as the image contains an ext4 partition only.

Joulinar, 2023-08-24:
That might be an option

MichaIng, 2023-08-26:
This can be also a VM on Windows, like our VirtualBox DietPi appliance.

I don't like the current situation either that even with WSL2 there is no reliable way to mount Linux filesystems on Windows. But as dpkg does not support FAT properly, having a `/boot` FAT filesystem was and it's always a hacky thing with limitations.

dirkhh, 2023-08-29:
I have been playing with this for a while since my similar question stayed unanswered.
`/boot` on FAT does have its issues, but I think I have come up with an idea that might solve this in a clean manner.
Add an optional `/firstboot` FAT partition as second partition after the ext4 partition in the images. And have the `dietpi.txt` and `dietpi-wifi.txt` files in that partition. Make this all optional. The firstboot looks for the partition. If it is there, it mounts it, copies the .txt files over the ones in `/boot/dietpi`, then unmounts the partition and deletes it.
Now the rest of the installation can continue as before.
If the partition isn't there, nothing changes. If it exists, it becomes a lot easier for users to edit those two files (and anything else that might benefit and could be put there).
If this is soething that @MichaIng finds a reasonable idea, I'll be happy to work on it and submit a PR.
(in case it isn't obvious, yes, I tried this in a very very hackish manner and it works - but I'd need to make this clean and more generic)

MichaIng, 2023-08-29:
That's quite a nice idea. Could be all done in https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/fs_partition_resize.sh.

The partition could be created in [dietpi-build](https://github.com/MichaIng/DietPi/blob/dev/.build/images/dietpi-build). But one difficulty is that [dietpi-imager](https://github.com/MichaIng/DietPi/blob/dev/.build/images/dietpi-imager) currently expects the rootfs to be on the last partition, otherwise it wouldn't be able to shrink the image. And it compresses it in the same turn. So actually the only way I currently see is to do this in `dietpi-imager`, probably based on an environment variable and optionally menu option 🤔.

MichaIng, 2023-08-29:
To avoid misunderstanding in the first place: It is not like `rpi-imager` a tool to pre-configure and flash an image, but to create a minimised image (minimised filesystem and partition sizes) from an attached drive or `img` file. Our images are generated with these tools:
- `dietpi-build` as the parent build script to `debootstrap` the rootfs, do partitioning and filesystem creation and call the other scripts.
- `dietpi-installer` to install platform/choice dependent packages, install DietPi files and configure the system, all from within the system started as (`systemd-nspawn`) container.
- `dietpi-imager` to reduce the image size back to a minimum and generate an archive from it.

There are some special cases, like the Clonezilla-based x86_64 installer images and VM appliances, but for those, the extra FAT partition has no use anyway.

dirkhh, 2023-08-29:
Yup... reading the sources has gotten me to that point already... doing what I'm trying to do seems fairly easy, given the structure. Doing it aligned with your way of writing shell code... yeah, that might take a few days 🤣