MichaIng/DietPi #6571: Docker Compose not installing correctly?

privacyguy123, 2023-08-23:
Says it's installed but definitely is not - running pip install docker-compose comes up with some error. Is this known broken on Bookworm perhaps?

Joulinar, 2023-08-23:
Docker compose is not installed via pip. It is a Debian plugin package for Docker and the command is docker compose and not `docker-compose`

https://docs.docker.com/compose/migrate/

dirkhh, 2023-08-29:
since I stumbled over this several times in the past... there is the old `docker-compose` that was its own thing, and then there is the `docker compose` module that is part of docker and is the officially supported version from Docker (and the one that is supported in DietPi and that appears to work well in my testing so far - it's a dependency of the app that I'm trying to add to DietPi, so something I'm interested in)

Some people simple add a small alias `docker-compose=docker compose` in order to keep their muscle memory happy.

MichaIng, 2023-08-29:
And we shipped the standalone `docker-compose` as well in the past. We did note the chance in the release notes, but it was to be expected that these issues/questions appear 😄. I also thought about adding the alias OOTB, but probably it is not bad either to force users learning and using the new way of calling it. Also there might be reasons to install and use the standalone version, which would be masked by the alias.

closed 2025-07-18 by MichaIng