MichaIng/DietPi #6572: At least put vim.tiny back in the default install

owendelong, 2023-08-23:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
It's bad enough that NANO is the default. I understand why it was done, but it's the first package I remove after software install.
However, taking out vim was just wrong-headed, IMHO. I have no interest in learning NANO and vim (even vim.tiny) has a LOT
more functionality than NANO.

#### Describe the solution you'd like:
Put vim.tiny back into the default load. I assume removing nano would be a bridge too far, but it wouldn't bother me at all if you did.

#### Describe alternatives you've considered:
Not a lot of alternatives available. Need editor to modify interfaces file before apt will work, so not like I can "just install vim.tiny" for this purpose.

#### Additional context
I forgot the wordstar keybindings many many years ago and relearning them seems like a giant leap backwards from useful editors.

MichaIng, 2023-08-24:
> However, taking out vim was just wrong-headed, IMHO.

vim was never pre-installed with DietPi, but it was only `nano` as long as I am with this project 🤔. And IMO it is right like this:
- `nano` is much easier/intuitive to use for Linux newbies. vim needs to be learned.
- `nano` is in comparison very small.

When it is about high-efficiency text editors, it starts to become political among the alternatives 😄. Not participating and e.g. not upsetting passionate Emacs users is another reason to use something more neutral like `nano` 😅.

Is it such a big problem to install Vim-Tiny via `dietpi-software` or just `apt install vim-tiny`?

owendelong, 2023-08-24:
I noted the reasons that is an issue starting. eMacs is huge (compared to other editors). I’ll look again, but I thought I remembered the vim.tiny package being smaller than nano. Certainly more feature rich. As I said, I understand the default to nano, but a system lacking vi is awkward for those of us who remember Unix before Linux. Owen

ravenclaw900, 2023-08-24:
`vim-tiny` is available by default in the Debian package repository, so you just need to run `apt install vim-tiny`. No file modification is needed to do this, all you have to do is run the command.

owendelong, 2023-08-25:
If you can explain how to make that work before the system has a network connection, you’re on. As documented in my original message, the concern is for editing the files needed to get the network operational.

Joulinar, 2023-08-25:
> If you can explain how to make that work before the system has a network connection

> concern is for editing the files needed to get the network operational

Normally the network connection is established automatically, and it is not necessary to adjust files. There are very rare cases where people use USB-to-Ethernet adapters where this might be necessary. But `nano` is a good choice for that too.

owendelong, 2023-08-25:
Normally the network connection is established automatically, and it is not necessary to adjust files. There are very rare cases where people use USB-to-Ethernet adapters where this might be necessary. But nano is a good choice for that too.There are less rare cases for routers (nano pi R6S) where dhcp isn’t available and the first connection established isn’t eth0. As to “nano is a good choice for that”, agree to disagree. Owen

MichaIng, 2023-08-26:
I think we agree that a text editor is required, and that `nano` is good to have as it is mostly self-explaining and sufficiently easy to use also for newbies.

We could now start to pre-install `vim-tiny` for those who like Vim, and then others ask for other Vim variants and then Emacs fans ask for that, and finally the idea of a minimal "diet" image is broken. We ship one small (indeed `vim-tiny` is overall similar size) text editor that everyone is able to use without hassle, and I do not see a reason to ship a second one, as it can be so easily installed via APT. Editing one or two text files with `nano` to bring up Internet access shouldn't be a problem, before you can install Vim for all future work.

MichaIng, 2023-08-26:
> However, taking out vim was just wrong-headed, IMHO.

And as this wording implies that Vim was there at some point:
- DietPi never shipped with Vim pre-installed.
- Also neither `debootstrap` nor a minimal Debian via netinst ISOs ships with Vim.
- So we did not "take it out", but it was just never there.

One thing I aim for is replacing the `nano` calls in our code with `editor`, so that at the few places where our scripts open text files Vim is used, if installed, respectively and other higher priority editor installed or explicitly selected via:
```sh
update-alternatives --config editor
```

owendelong, 2023-08-27:
The first thing I do on any debi-oid system is install vim. Removing nano comes next. Nano is not hassle free. Its keybindings come from wordstar (yes from the cp/m 8080A days). It’s a giant leap backwards. Rest assured, if I encounter anything with nano hard coded instead of editor, I’ll assist in that effort by calling them out.

dirkhh, 2023-08-29:
you will never manage to find an editor that makes everyone happy.
`nano` really unifies your users in that basically everyone dislikes it 🤣
Joking aside, `nano` gets the job done to get you to the point where you can install your editor of choice. And if that's too hard, then it's really easy to create custom images that include `vim` for you.
(and yes, `apt install -y vim` is typically my first command once the network is up as well - but I do realize that the vast majority of humans on the planet would describe `vi` and derivatives as about the most newbie hostile editor still in active use)

MichaIng, 2023-08-29:
Even I do not use `vi` 🙈. I read a tutorial, like the concept and see the potential for massive efficiency increase, but I never found time and mood to actually practice and getting used to it. In the end for the smaller text file searches and edits I do not Linux, `nano` just works, also not too bad with search+page/line/word/char forward/backwards shortcuts. I do coding on Windows with notepad++ 🙈.

dirkhh, 2023-08-29:
🤣
I think a big part of this is **when** you started working on Unix systems. For me that's the 1980s... so `vi` was The Editor. I mean, sure Escape-Meta-Alt-Ctrl-Shift (often referred to by its initials) existed (and, ironically, was the first open source project I contributed to), but pretty much everyone simply ended up using `vi`. So I can do more or less anything I want in `vi` without having to think about it. Complex search and replace? Automated edits on hundreds of lines? Easy...
`nano` on the other hand is maddening.
But still. It is the right choice as the only editor included in DietPi by default. No argument.

MichaIng, 2023-08-29:
> Complex search and replace? Automated edits on hundreds of lines? Easy...

Indeed, I'm not aware that `nano` is capable of such. For those cases I use `sed`, but switching back and forth between editor and command-line is not great. However, rare cases for me.

So I am definitely too young then, and also I entered the server/Linux/open source world quite late while studying in a different field, when searching for a solution to better organise myself with calendar, tasks, contacts, notes synced across PC, laptop and mobile, disliking the only prominent big G cloud solution 😄.