MichaIng/DietPi #6576: Tailscale installed via dietpi-software does not update cleanly

mstgrv, 2023-08-24:
### Creating a bug report/issue

#### Required Information
- DietPi version | 8.20.1
- Distro version | bullseye
- Kernel version | Linux fountain 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | RPi
- SD card used | Samsung

#### Additional Information (if applicable)
As of v1.36 and above, Tailscale now includes a functionality for self-updating (`sudo tailscale update`). I've confirmed this works fine on another machine running Raspbian, but on Dietpi 8.20.1 it fails with the following message: 

> open /etc/apt/sources.list.d/tailscale.list: no such file or directory

Looking at /etc/apt/sources.list.d on my dietpi machine, there is a file called "dietpi-tailscale.list" which I assume is named that way to denote that it was installed via dietpi-software, but it seems to be causing an issue with Tailscale's self-update functionality that there's no tailscale.list file present in /etc/apt

#### Expected behaviour
Tailscale should self-update cleanly

#### Actual behaviour
The command fails with the above error message

#### Extra details

- ...

MichaIng, 2023-08-24:
Strange, the self-updater checks for a hardcoded APT list name? What exactly does it actually do? Since it is an APT package, updates are easily done via `apt upgrade`. What does `tailscale update` different than that?

mstgrv, 2023-08-27:
I guess I'll have to wait until the next Tailscale update to investigate further, will post back with more info after the next update drops.

MichaIng, 2023-08-27:
Probably you can downgrade it. Check for available versions via:
```sh
apt show -a tailscale
```
Then downgrade to a specific version via:
```sh
apt install tailscale=a.b.c
```
Replace "a.b.c" with the version string.

mstgrv, 2023-08-28:
0. So after installing Tailscale from dietpi-software, it seems that only [b]dietpi-tailscale.list[/b] is created in /etc/apt/sources.list.d/ - there is no "tailscale.list" file created when installing Tailscale via dietpi-software

1. I just ran 'apt install tailscale=1.46.1' to downgrade to that version, and it gives me the following error message:

> W: Target Packages (main/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-tailscale.list:1 and /etc/apt/sources.list.d/tailscale.list:2
> 
> W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/dietpi-tailscale.list:1 and /etc/apt/sources.list.d/tailscale.list:2
> 
> Do you want to continue? [Y/n]

So installing / downgrading via apt (rather than dietpi-software) now appears to create tailscale.list, but there's a clash with the dietpi-tailscale.list file that was already there.

2. I entered Y and the downgrade proceeded. There are now 2 tailscale files in /etc/apt/sources.list.d - dietpi-tailscale.list, and tailscale.list

3. I ran 'tailscale update' again to retry the self-update functionality, and it worked without issue, because there is now a tailscale.list file in /etc/apt/sources.list.d.

So it's fixable when using apt directly, but the root issue appears to be that installing Tailscale via dietpi-software in the first place does not create a tailscale.list file, but installing it via apt does create it.

MichaIng, 2023-08-28:
> So after installing Tailscale from dietpi-software, it seems that only dietpi-tailscale.list is created in /etc/apt/sources.list.d/ - there is no tailscale.list file created when installing Tailscale via dietpi-software

Yes this is intentional for all APT lists we installed in `dietpi-software`, to have them unique and identifiable so that we can e.g. safely apply patches to them without risking to overwrite custom admin files, or generally such we cannot be certain about what the content/intention is.

> So installing / downgrading via apt (rather than dietpi-software) now appears to create tailscale.list

No, the APT package does not install a list file. That would be a chicken and egg issue 😉. I could imagine that `tailscale update` created it 🤔.

I just tried it myself and neither installing nor downgrading the package creates this list file. Are you sure that you did not manually create it in attempt to solve the issue?

I also renamed the file, and `tailscale update` does nothing else than `apt update && apt install tailscale`, quite unnecessary IMO. It has a `--track` option to switch to the `stable`/`unstable` suite, in which case it edits the `tailscale.list`. But the implementation is broken and it additionally passes `unstable` as version string, which of course does not exist:
```console
root@VM-Bookworm:/etc/apt/sources.list.d# tailscale update --track unstable
This will update Tailscale from 1.48.1 to unstable. Continue? [y/n] y
Updated /etc/apt/sources.list.d/tailscale.list to use the unstable track
Get:1 https://pkgs.tailscale.com/unstable/debian bookworm InRelease
Get:2 https://pkgs.tailscale.com/unstable/debian bookworm/main amd64 Packages [34.7 kB]
Get:3 https://pkgs.tailscale.com/unstable/debian bookworm/main all Packages [354 B]
Fetched 41.7 kB in 2s (19.2 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package tailscale is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Version 'unstable' for 'tailscale' was not found
exit status 100
```
I have no idea why someone would create and maintain an APT CLI wrapper for the own software, just with limited options and enough complexity that above bugs can happen 😄.

Forget about this, just use `apt update` (or wait for the DietPi banner to show/offer you upgrades) and `apt upgrade` like for every other APT package.

mstgrv, 2023-08-28:
OK thanks for that, I've tried it another way but I'm still getting an issue that doesn't exist on other machines running Raspbian:

1. Install Tailscale via dietpi-software - this installs the latest version (currently 1.48.1)
2. Run 'tailscale up' to authenticate the machine and add it to my Tailnet
3. SSH into the machine over the Tailscale connection
4. Run 'apt install tailscale=1.46.1' to downgrade to the previous version (so I can test updating)
5. Run 'dietpi-update' - this runs 'apt update' and correctly identifies that a new version of Tailscale is available.
6. Run 'apt upgrade' - this fails with the following message: 
> E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem. 

7. Run 'dpkg --configure -a' (per the error message) and the following happens:
> Setting up tailscale (1.46.1) ...
> Connection to dietpi closed by remote host.
> Connection to dietpi closed.

8. SSHing back in over Tailscale, I still get a notification that there's an update to install. Running 'apt upgrade' just produces the same error from step 6 again.
9. On the machine directly (not via SSH over Tailscale), run 'apt upgrade' - I receive the same error from step 6.
10. Again on the machine directly, run 'dpkg --configure -a' and it appears to work correctly, with the below output:
> Setting up tailscale (1.46.1) ...

11. Run 'tailscale version' and it's confirm it's only 1.46.1 - it hasn't been upgraded to 1.48.1 despite having run 'apt upgrade'
12. Run 'apt upgrade' again (on the machine directly), and it works OK

So trying the 'apt update && apt upgrade' method fails when you do it while SSH'd into the machine over the Tailscale connection itself. On other Raspbian machines, you can update Tailscale this way without issues - you do lose the SSH connection at the "Setting up tailscale (version)..." step, but when you SSH back in again over the Tailscale connection, the setup has completed.

MichaIng, 2023-08-28:
Are you trying to upgrade Tailscale while you are connected through Tailscale? It seems like the package upgrade aborts the connection which aborts the upgrade, causing this error.

It is normal that aborted SSH connections abort any running processes from this session. This is done via SIGTERM, AFAIK, so whether this aborts an APT package install in a bad state, or whether it even finishes is just a question of timing, and e.g. how far the package install/configuration went with previous attempts. So likely it finally worked because previous `dpkg --configure -a` run did already configure some stuff so that the last attempt just finished quickly enough, before the process is killed by the system.

Raspbian uses OpenSSH by default instead of Dropbear. It might give processes a little more grace time before killing them, but generally behaves the same. So if/as a Tailscale package upgrade does abort all open connections, you simply should not upgrade it while being connected through Tailscale.

If there is really no other way, better use something like GNU/screen (`apt install screen`) to have the shell session within a dedicated process that is detached from the SSH session. This way, when network connection or SSH session is aborted for any reason, the shell with all its processes just continues detached and when you connect back you can re-attach to this shell.

mstgrv, 2023-08-28:
Yes I was trying to upgrade Tailscale while connected through Tailscale. On Raspbian this works fine - the SSH connection drops during the process, but once you log back in everything has completed successfully and there's none of the issues that I'm running into with Dietpi, regardless of whether you use 'tailscale update' or 'apt update && apt upgrade'.

Not sure how else to dig into finding the issue at this stage.

Joulinar, 2023-08-28:
as stated by @MichaIng , on DietPi we use a different SSH on a default configuration. Simply switch to OpenSSH and check again. However way better to use GNU/screen (apt install screen) to have the shell session within a dedicated process that is detached from the SSH session.

MichaIng, 2023-08-28:
And again: This is not an issue but perfectly expected behaviour with all SSH servers. It is pure luck and timing whether a task is able to finish before being killed or not, and it is pure luck whether a killed task leaves its work in a graceful state or broken. Especially with APT upgrades this is quite risky: Think of a kernel upgrade where the large kernel image is not fully written, or essential modules missing, or the initramfs not fully created. Do not rely on this working, even if it did work well a few times on RPi OS.

mstgrv, 2023-08-29:
> as stated by @MichaIng , on DietPi we use a different SSH on a default configuration. Simply switch to OpenSSH and check again. However way better to use GNU/screen (apt install screen) to have the shell session within a dedicated process that is detached from the SSH session.

Having the same issues with OpenSSH selected as the SSH server, but if screen is gonna be the better way to do it I'll try that on the next update for sure.

mstgrv, 2023-09-12:
A new Tailscale version (1.48.2) just dropped so I tried updating via screen and there are still issues:

1. SSH into my dietpi machine via Tailscale (the SSH server running on this machine is OpenSSH)
2. run 'screen'
3. run 'sudo apt update && sudo apt upgrade' -  it eventually asks me if I want to apply the tailscale upgrade and I select Y.
4. The SSH session eventually drops, as expected
5. SSH back into the machine over Tailscale - it works correctly.
6. run 'screen -rd' to reconnect to the screen session from earlier - it says "There is no screen to be detached.", which is unexpected 
7. Check the Tailscale update applied correctly ('tailscale version') - it did (it reports version 1.48.2)
8. Run 'sudo apt update && sudo apt upgrade' again - it gives the following error:
>E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
9. run 'sudo dpkg --configure -a' - it says "Setting up tailscale (1.48.2) ..." but then the SSH connection drops
10. re-connect to the machine via Tailscale, run 'screen', and then re-run 'sudo dpkg --configure -a' - it says "Setting up tailscale (1.48.2) ..." but then the SSH connection drops

Even with 'screen', Tailscale cannot be updated cleanly. The only clean way to complete the upgrade was:

1. SSH into the machine directly (not over Tailscale)
2. run 'sudo dpkg --configure -a' - it gave the same "Setting up tailscale (1.48.2) ..." output, but then completed correctly and returned me to the command line
3. I ran 'sudo apt update && sudo apt upgrade" once more just to double check there were no continuing issues, and there weren't any.

When I upgraded another machine running Raspbian, here's what happened (it was successful):

1. SSH in over Tailscale
2. run 'screen' 
3. run 'sudo apt update && sudo apt upgrade' - the upgrade proceeds, I hit Y to confirm the upgrade, and then the SSH session drops as expected
4. I SSH back into the machine over Tailscale, and run 'screen -rd' - unlike on the dietpi machine, this time the screen session is resumed correctly, and I can see the output of the resumed session shows that the installation completed correctly.
5. I run 'exit' to get back to a plain shell
6. I run 'tailscale version' to check the updated version (1.48.2) is running, and it is
7. I run 'sudo apt update && sudo apt upgrade' to check it runs cleanly (without the dpkg error from above) and it does.

MichaIng, 2023-09-12:
Hmm, were you able to replicate this? "There is no screen to be detached." sounds like the screen was not detached but closed, like CTRL+D instead of CTRL+A+D would do it. The related packages on Raspbian and DietPi are the same, especially when you switched to OpenSSH on DietPi, and generally screen behaves the same, neither being affected by crashing internal processes, nor by crashing/terminated external connections.

mstgrv, 2023-09-13:
Doesn't seem like I'll be able to try again until another Tailscale update drops.

To my mind, it seems to point back to something in my original post - I installed Tailscale via `dietpi-software` and there is no tailscale.list file in /etc/apt/sources.list.d, and this is what's causing all the issues with all the various methods of updating that I've tried, even when using screen.

mstgrv, 2023-10-07:
Upgrading to Tailscale 1.50 and 1.50.1 have produced the same issue. I've also tried uninstalling Tailscale with dietpi-software and installing it using their own install script (https://tailscale.com/download/linux) but the same issue persists. You cannot cleanly update Tailscale over a Tailscale connection, whether running the update directly or via screen, but another machine running Raspian can do it without issues (via either method).

Let me know if there's anything else to try but I'm running out of ideas myself.

jacko0, 2024-01-04:
I also have this Tailscale 1.54 update issue on two different Raspberry PIs. While trying to update DietPi from v8.23.3 to v8.25.1

Board 1 96201b58-f225-40a6-a098-ec06d47d84fd
Board 2 8d180c74-096d-4998-b0e3-5398bed99249

#### Details:
- Date           | Thu Jan  4 21:54:21 GMT 2024
- Bug report     | 96201b58-f225-40a6-a098-ec06d47d84fd
- DietPi version | v8.23.3 (MichaIng/master)
- Image creator  | 
- Pre-image      | 
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux ts-de 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=0)
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1.  run dietpi-update 
2.  or run an apt update
3. or run 'dpkg --configure -a' 
#### Expected behaviour:

dietpi-update to finish without errors
- ...
#### Actual behaviour:

dietpi-update errors with E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
Running dpkg --configure -a starts to install Tailscale then kills the remote connection.

- ...
#### Extra details:

The update of the tailscale package seems to cause the issue
- ...
#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

Joulinar, 2024-01-04:
@jacko0 are you connected to the systems via Tailscale ??

jacko0, 2024-01-04:
Hi, yes I am. They are both remote connections.

MichaIng, 2024-01-07:
Please see the issue at the Tailscale repo: https://github.com/tailscale/tailscale/issues/9780
When I tested it, the active Tailscale connection remained active during the package upgrade, so I am unable to replicate it. I have no idea why a service restart kills the connection in your both cases, and more strangely even the screen session in OP case, while it all remains up here. We really only install the default APT package from their repository, matching 100% what you get when installing with their own scripts/docs. But maybe Tailscale devs have an idea in which circumstances and why a service restart kills the connection.

jacko0, 2024-01-09:
Thanks for the info. 
My workaround was to get physical access to the PI and run 'dpkg --configure -a' it finished with no problem.
Then I ran dietpi-update and it's all working again.

MichaIng, 2024-01-13:
@jacko0
Did you run `tailscale up` or `tailscale up --ssh` when establishing the Tailscale connection?

This flag lets Tailscale use an internal SSH server. A theory was that, if this is used and you restart the Tailscale service, that internal SSH server is terminated and terminates also active sessions (unlike Dropbear and OpenSSH). Otherwise it is strange that the SSH sessions are terminated in your both cases, while it remains active on my tests.

anoop-b, 2024-04-26:
As a workaround I was able to update the device remotely by first installing the dietpi dashboard and updating packages by using the terminal shell via the dashboard.

MichaIng, 2024-05-11:
So you are facing the same issue, that the SSH connection is aborted? Very strange that I was never able to replicate it. Does a screen session survive in your case?

I am wondering whether OOM kills might be the reason. Quite some coincidence and unlikely if all system memory is consumed right at the time when the Tailscale service is restarted, but just to rule it out: There is more than enough free system memory?

anoop-b, 2024-05-15:
> So you are facing the same issue, that the SSH connection is aborted? 

yes

> Does a screen session survive in your case?

No

IMO I don't believe it could be because of system memory because the installation went alright through dietpi dashboard. 

As a side note, tailscale now lets you enable [auto updates](https://tailscale.com/blog/auto-update-beta) on the clients. But this won't work on dietpi installation because of the `dietpi-` suffix in `dietpi-tailscale.list` sources list.

MichaIng, 2024-05-15:
Yes, that feature IMO does not make any sense, especially not the way it was implemented. Just enable APT auto-upgrades, if you want such. I mean imagine Debian or any distro wants to add Tailscale to their APT repositories, which would be awesome. Now this builtin feature throws this error, hence Debian cannot implement Tailscale without patching the source code to remove the auto-update feature first.

Can one of you guys attack a screen+keyboard, then replicate the issue? I guess even a reinstall would work
```sh
apt install --reinstall tailscale
```
Then check SSH server and Tailscale daemon logs, in the hope to see something which killed the sessions:
```sh
journalctl -u dropbear -u ssh -u tailscaled
```
And kernel logs:
```sh
dmesg -l 0,1,2,3
```

MichaIng, 2025-05-01:
I tested it again, using two VMs, one with a Bookworm system, the other with a Bullseye system:
```sh
dietpi-software install 58 # alternatively just "apt install tailscale"
tailscale up # open printed link to register device
dbclient <IP> # connect via Dropbear SSH client, using IPv4 address of remote VM
apt install --reinstall tailscale
systemctl restart tailscaled
```
I did this on both VMs and connected them to each other via Tailscale IP.

But while this worked, `systemctl stop tailscaled` did indeed hang the connection. Fascinating however: when starting the `tailscaled` service again with a dedicated connection, both Dropbear SSH sessions were immediately usable again.

My connection is between two local clients, so even that the connection is done through Tailscale, in the end it is P2P, so with much less delay. Possible a service restart cuts the connection short enough to not break it, respectively to not have running commands terminated. But at least the SSH connection itself is not terminated. I also tried with OpenSSH clients (but Dropbear server) and OpenSSH server, which all behave the same. I even switched from Dropbear server to OpenSSH server via `dietpi-software` though Tailscale connection on the respective other machine both concurrently, before rerunning above restart tests.

In any case, would be great if one of you can check logs when a package upgrade or reinstall causes a connection/dpkg abortion:
```sh
journalctl -u dropbear -u ssh -u tailscaled
dmesg -l 0,1,2,3
```

queeup, 2025-05-02:
Connection lost while dpkg "Setting up tailscale".

```
$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  docker-ce docker-ce-cli docker-compose-plugin tailscale
4 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 72.5 MB of archives.
After this operation, 956 kB disk space will be freed.
Do you want to continue? [Y/n] Y
Get:1 https://download.docker.com/linux/debian bookworm/stable arm64 docker-ce-cli arm64 5:28.1.1-1~debian.12~bookworm [14.3 MB]
Get:2 https://pkgs.tailscale.com/stable/debian bookworm/main arm64 tailscale arm64 1.82.5 [29.6 MB]
Get:3 https://download.docker.com/linux/debian bookworm/stable arm64 docker-ce arm64 5:28.1.1-1~debian.12~bookworm [16.7 MB]
Get:4 https://download.docker.com/linux/debian bookworm/stable arm64 docker-compose-plugin arm64 2.35.1-1~debian.12~bookworm [12.0 MB]
Fetched 72.5 MB in 8s (8818 kB/s)                                                                                      
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 16332 files and directories currently installed.)
Preparing to unpack .../docker-ce-cli_5%3a28.1.1-1~debian.12~bookworm_arm64.deb ...
Unpacking docker-ce-cli (5:28.1.1-1~debian.12~bookworm) over (5:28.0.4-1~debian.12~bookworm) ...
Preparing to unpack .../docker-ce_5%3a28.1.1-1~debian.12~bookworm_arm64.deb ...
Unpacking docker-ce (5:28.1.1-1~debian.12~bookworm) over (5:28.0.4-1~debian.12~bookworm) ...
Preparing to unpack .../docker-compose-plugin_2.35.1-1~debian.12~bookworm_arm64.deb ...
Unpacking docker-compose-plugin (2.35.1-1~debian.12~bookworm) over (2.34.0-1~debian.12~bookworm) ...
Preparing to unpack .../tailscale_1.82.5_arm64.deb ...
Unpacking tailscale (1.82.5) over (1.82.0) ...
Setting up docker-compose-plugin (2.35.1-1~debian.12~bookworm) ...
Setting up docker-ce-cli (5:28.1.1-1~debian.12~bookworm) ...
Setting up tailscale (1.82.5) ...
Connection to dietpi-zgz closed by remote host.
Connection to dietpi-zgz closed.

$
```

```
May 02 23:59:46 dietpi-zgz tailscaled[9924]: tailscaled got signal terminated; shutting down
May 02 23:59:46 dietpi-zgz systemd[1]: Stopping tailscaled.service - Tailscale node agent...
```

MichaIng, 2025-05-03:
@queeup 
Is it Tailscale's embedded SSH in your case or Dropbear/OpenSSH?

It really might be related to how long the connection is down during service restart. I'll test with an RPi Zero W.

queeup, 2025-05-03:
It is Tailscale's embedded SSH.

MichaIng, 2025-05-06:
Okay for embedded SSH, it is normal, I guess. The service is in fact stopped. OpenSSH/Dropbear connections can survive in certain circumstances, if the Tailscale connection is down short enough, in my case even when stopped longer, but that does not mean that currently running commands do not get a SIGTERM.

MichaIng, 2025-05-28:
Tested it again with a remote connection, with OpenSSH server and OpenSSH client, and Dropbear client, just to test all combinations. It all works well. The Tailscale service restarts, so the TCP connection is cut for a moment, but the SSH connection + the `dpkg` process remain active.

I even added an additional `ExecStopPost=/bin/sleep 10` to the service, so that the connection remains cut for at least 10 seconds. It is, I cannot do CTRL+C or anything during that time, but after the 10 seconds, the upgrade/downgrade/reinstall/restart continues without issue. It is a general behavior of SSH connections to stay active even if the underlying network connection is not. Once the network connection is back again, the SSH session can be continued to be used as before. For this reason we have an additional shutdown service which actively terminates the SSH sessions on shutdown, since otherwise it would remain "active" at the client forever, or until a very long timeout, so you cannot quickly restart a new SSH session but need to terminate the client first.

Now there are firewalls and such which do close connections automatically, if on idle. And others might want that an SSH session is terminated, if the underlying network connection is lost, to avoid unused idle sessions to stack up. For this, SSH servers come with KeepAlive implementations, sending packets to the peer every now and then, to:
1. signal possible firewalls and such that the network connection is still in use, and hence shall not be terminated
2. terminate the SSH session if the network connection is lost

OpenSSH has:
- `TCPKeepAlive`, which just says whether the host system's default KeepAlive mechanism should be used, which is usually to start seinding KeepAlive packets after 1 hour (!) of inactivity, every 75 seconds, and killing the connection after 10 messages have not been answered. So, well, unlikely that a system is configured to send KeepAlive messages and kill the connection all within the 1-3 seconds the Tailscale service is restarted.
- `ClientAliveInterval` and `ClientAliveCountMax`, which is an SSH-wise KeepAlive implementation. So it does not send raw TCP packets outside the SSH connection, but packets within the SSH session (on application level). It is disabled by default, but `ClientAliveCountMax` is 3 by default. So setting `ClientAliveInterval` to 20 would mean that OpenSSH sends KeepAlive messages every 20 seconds, killing the session after 1 minute, if the client does not answer. The same can be configured at the OpenSSH client, with `ServerAliveInterval` and `ServerAliveCountMax`, and logically also the client system can have TCP KeepAlive.

Dropbear has the `-K` option to enable and specify the the interval of such messages, and kills the connection after 3 missing responses, for its server as well as its client.

So if OpenSSH is used, it could be checked whether one of these settings is used, and if so disabled, else enabled, as both can lead to killed connections, though usually at MUCH longer time scales:
```sh
grep -Er 'TCPKeepAlive|ClientAliveInterval' /etc/ssh
```
In case of Dropbear, `-K` would be usually defined in `/etc/default/dropbear`. For its client, I think there is no config file, but `-K` would always need to be defined at the command line.

And in case of other SSH clients: check its settings whether there is some KeepAlive feature active, or any option which would terminate the session quickly on lost network connection.

________________

I spent enough time now testing this issue. Will close it, as it is no DietPi issue anyway. But feel free to further discuss.

closed 2025-05-28 by MichaIng

MichaIng, 2025-11-07:
DietPi updates are not meant to run automatically. You can enable automatic APT upgrades, but not DietPi updates: if you tried, you would miss important information and choices regarding software and kernel migrations and stuff like that.

Does it work for you when using the dedicated Dropbear or OpenSSH server, instead of Tailscale's own one?

MichaIng, 2025-11-07:
> v6.28.0

You run an ancient DietPi version from an unofficial repo. If this was the official repo, you'd get DietPi v6.35.0 offered, and then migrated to v7 and up, depending on the Debian version.

Please check this:
```sh
grep DEV_GIT /DietPi/dietpi.txt
cat /etc/os-release
```

MichaIng, 2025-11-07:
Ah okay, we removed that option shortly after.

Automatic APT package upgrades is fine/good via `CONFIG_CHECK_APT_UPDATES=2` in `/boot/dietpi.txt`. But I'd recommend against implementing manual DietPi updates. They are not mission critical unless you are (re)installing/updating `dietpi-software` options or use other DietPi tools. Hence DietPi updates are fine to do when you anyway spend some time on the console, use some tools, install software, etc, so you can in case read through info of do some choices that may be offered by times. If there are security relevant or other critical things within the scope of DietPi code we'd inform you via MOTD shown in login banner.

MichaIng, 2025-11-07:
> Do I need to reboot after changing `CONFIG_CHECK_APT_UPDATES=2 `in `/boot/dietpi.txt`?

Nope, next daily cron job will take this into account without reboot.

theincredibleman, 2025-11-26:
Just to update everyone following this, setting `CONFIG_CHECK_APT_UPDATES=2 `in `/boot/dietpi.txt` worked perfectly for me, Tailscale updates unattended and I can run dietpi-update while connected trough Tailscale SSH without any issues because Tailscale is already on the latest version.