MichaIng/DietPi #6577: Add option to disable EEE on the Pi 4 eth0 interface

douglascamata, 2023-08-24:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

I've been having a lot of issues lately with my Pi 4 completely disconnecting from my home's network. It's connected via ethernet directly to my internet vendor's router and configured with static IP and DNS servers.

I've done a lot of research around the internet... I got lost in grief and sadness. Then I found [this issue](https://github.com/raspberrypi/linux/issues/4289) and everything made sense. The problem seems related to kernel 6.1.y and the EEE (energy efficient ethernet) feature.

I disabled EEE with the command bellow and the problem went away:

```
sudo ethtool --set-eee eth0 eee off
```

But this is cleared on reboot. 

The problem became so widely spread and accepted that on kernel 6.5.y for the Pi, a kernel parameter called `genet.eee=N` can be used to automatically disable EEE on link (see https://github.com/raspberrypi/linux/commit/47a68cd4d2ec215356194ca1a97c7a9f4f9a3c78).

#### Describe the solution you'd like:

**While we don't have the kernel 6.5.y**: I would love if the network configuration of `dietpi-config` had an option to easily disable EEE for me for me using the systemd unit method from https://github.com/raspberrypi/linux/issues/4289#issuecomment-1351349084. 

**When we have the kernel 6.5.y**: the network configuration could then fix the EEE using the kernel parameter `genet.eee=N`. 

This is so important that it could be worth to be done **by default** by DietPi and those folks who are sure about enabling EEE could do it as they wish. 

It could even be considered to release a hotfix patch. There could be many out there with issues due to this.

#### Describe alternatives you've considered:

I considered the issue could be my router or cable, but I did many tests with other devices (Pi 3, random cheap dumb terminal) and cables I have... always and only the Pi 4 had issues.

#### Additional context

Nothing, already wrote too much.

douglascamata, 2023-08-24:
I suspect there might be some "survival bias" involved when we say "there aren't that many complaints out there". It's pretty difficult to figure out EEE is the actual problem. It also only affects eth0 and not wlan0. I'm 100% fine though with a switch in `dietpi-config`. Some extra documentation somewhere to mention that in some switches EEE on causes would be an awesome cherry on top so that users are aware of **why** this option exists. 

I didn't know about the `/boot/config.txt` path to disable EEE. Will definitely try it out! 

Thanks, @MichaIng! 🙇

douglascamata, 2023-08-24:
Still pending to try the `dtparam`, but it seems like people already tried and it doesn't work on the Pi 4: https://github.com/raspberrypi/linux/issues/4289#issuecomment-839665912

douglascamata, 2023-08-24:
There seems to be an alternative solution that doesn't use a systemd unit, I can also try it out: https://github.com/raspberrypi/linux/issues/3292#issuecomment-777167803

MichaIng, 2023-08-26:
> I suspect there might be some "survival bias" involved when we say "there aren't that many complaints out there". It's pretty difficult to figure out EEE is the actual problem.

We have about 100k users, about 25k of them using an RPi 4 and about 75% of them using Ethernet. I am not aware of unresolved reports about Ethernet connection issues. And compared to how many RPi customers are there in total, the number of reports on their GitHub, and that it took so long for them to actually provide a fix, points into the same direction: It is definitely rare, or the symptoms are usually not as problematic as in your case.

The bias in such cases is usually happening the other way round, when a few people from millions find each other in a bug report 😉. Enough to do something about it, but not urgent.

I'll wait until the kernel module parameter is shipped with a new kernel release and then add a toggle. Until then, indeed the workaround via `interfaces` config is best, is it applies to all `ifup` calls, not just a particular service. But I'd use `pre-up`
```sh
pre-up /sbin/ethtool --set-eee eth0 eee off
```

Joulinar, 2023-08-26:
If I understood the GitHub link correctly, it hardly depends on switch used and if it is supporting this feature.

MichaIng, 2023-08-26:
Seems to. However, EEE is enabled by default on every Ethernet adapter and board if supported, so that would not only affect the RPi only (which is quite possible, but would be probably recognised earlier with other network peers). But they also said something about the particular implementation within the RPi onboard Ethernet driver. Probably it does something more than just turning off power of no cable is attached which causes more often issues than minimal EEE would, or so.

douglascamata, 2023-08-26:
I don’t think pre-up will work. The interface has to be up for the EEE switch to work. That’s why solutions that run at boot time have to retry the command, like the the systemd unit workaround that retries to set EEE off every 30s until it succeeds, for example.

MichaIng, 2023-08-26:
Ah okay, the it behaves differently than the WiFi power save mode.