MichaIng/DietPi #6578: DietPi-Software | X.Org X Server: Start still fails on Odroid C2 with mainline kernel

eevan78, 2023-08-25:
Two weeks ago, I have tried to install LXDE on a freshly installed latest image for Odroid C2 and it didn't want to start. Upon finding this page and looking into commit 572d1b3 I spotted that because of the bad condition value in line 2725 it didn't install libegl1 and libgles2, and created the config file. After I did that manually, LXDE could start.

For my Odroid C2, the value of `G_HW_MODEL` is 12, and the condition in line 2725 reads:
```
elif (( $G_HW_ARCH < 10 && $G_HW_MODEL < 10 ))
			then
...
```

So it seems that the 10 in the second expression should be changed to 13.
But I haven't tried it.

_Originally posted by @eevan78 in https://github.com/MichaIng/DietPi/issues/5234#issuecomment-1674248735_

MichaIng, 2023-08-26:
Many thanks fro your report.

The line 2725 on current DietPi looks different: https://github.com/MichaIng/DietPi/blob/a044838/dietpi/dietpi-software#L2725
The condition you are referring to has been fixed a long time ago: https://github.com/MichaIng/DietPi/blob/a044838/dietpi/dietpi-software#L2687
```sh
elif [[ $G_HW_CPUID == 7 && -d '/dev/dri' ]]
```
So the config file `/etc/X11/xorg.conf.d/02-dietpi-aml-s905.conf` should have been created. And the two packages should have been pulled in as dependencies:
```sh
dpkg -l | grep -E 'lib(egl1|gles2)'
```
Can you verify that both is true?

eevan78, 2023-08-26:
I apologize for this report,
It seams that I flashed by mistake an older image, and that is why I
experienced issue.

I tried again with the current one, and both libraries and config file are
found.

MichaIng, 2023-08-26:
Okay great. Always good to update DietPi to latest version first.

I'll mark this issue as closed. Feel free to reopen if required.

closed 2023-09-01 by MichaIng