MichaIng/DietPi #6581: DietPi - After reboot the Date was set to 12/31/2069 and Failed to auto update NTP settings

daytonajrw, 2023-08-27:
#### Details:
- Date           | Tue Dec 31 19:42:47 EST 2069
- Bug report     | b9491b4a-7818-4a1a-a977-1aa2cbcce5e2
- DietPi version | v8.19.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | `Linux DietPiJRW 5.10.0-23-amd64 #1 SMP Debian 5.10.179-3 (2023-07-27) x86_64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Set_software
#### Steps to reproduce:

1. After a reboot the system time was 12/31/2069
2. all certs were considered invalid and ntp could not be automatically updated
#### Expected behaviour:

- after reboot, the system time should not have changed to 12/31/2069
- the time should have stayed the same or if it was slightly off, it should have updated correctly after checking into ntp
#### Actual behaviour:

- ...
#### Extra details:

- When i noticed issues with Pi-Hole after the restart, i logged into to terminal and i received the following message:  
curl: (60) SSL certificate problem: certificate has expired
More details here: https://curl.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.
- To solve the issue, i manually set the date and rebooted they system
root@DietPiJRW:~# date -s "Sun Aug 27 20:41:10 UTC 2023"
Sun 27 Aug 2023 04:41:10 PM EDT
root@DietPiJRW:~# date
Sun 27 Aug 2023 04:41:11 PM EDT
#### Additional logs:
```
Err:1 https://deb.debian.org/debian bullseye InRelease
  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
Err:2 https://deb.debian.org/debian bullseye-updates InRelease
  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
Err:3 https://deb.debian.org/debian-security bullseye-security InRelease
  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
Err:4 https://deb.debian.org/debian bullseye-backports InRelease
  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
Reading package lists...
E: Failed to fetch https://deb.debian.org/debian/dists/bullseye/InRelease  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
E: Failed to fetch https://deb.debian.org/debian/dists/bullseye-updates/InRelease  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
E: Failed to fetch https://deb.debian.org/debian-security/dists/bullseye-security/InRelease  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
E: Failed to fetch https://deb.debian.org/debian/dists/bullseye-backports/InRelease  Certificate verification failed: The certificate is NOT trusted. The revocation or OCSP data are old and have been superseded. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 199.232.162.132 443]
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

MichaIng, 2023-08-27:
Why did the network time sync fail?
```sh
journalctl -u systemd-timesyncd
```
At boot, the time is first set via `fake-hwclock` to the last time shored at shutdown (if not RTC is attached). This should hence be a little in the past, but 2069 is strange. However, network time sync should align it.

Joulinar, 2023-08-27:
this should manually update system time

```
/boot/dietpi/func/run_ntpd 1
```

daytonajrw, 2023-08-28:
> Why did the network time sync fail?
> 
> ```shell
> journalctl -u systemd-timesyncd
> ```
> 
> At boot, the time is first set via `fake-hwclock` to the last time shored at shutdown (if not RTC is attached). This should hence be a little in the past, but 2069 is strange. However, network time sync should align it.

the error is no longer in the log.  i updated the client version after i was able to get the system up and running again.  
All of the errors when trying to update NTP were due to cert errors which are likely due to the system thinking it was 2069 so every valid cert encountered it thought it was 46 years old and denied anything further.
I even attempted to apt update which failed due to cert issues.

> this should manually update system time
> 
> ```shell
>/boot/dietpi/func/run_ntpd 1
> ```
>
This also failed due to cert issues.

Joulinar, 2023-08-28:
> > this should manually update system time
> >
> > `/boot/dietpi/func/run_ntpd 1`
>
> This also failed due to cert issues.

There is absolutely no certificate involved while updating the system time. How should that fail? Do you have some logs from this attachment?

MichaIng, 2023-08-28:
Right, NTP is not HTTPS. It would be a quite dumb snake biting its tail if NTP updates did require certificate authentication while certificate authentication did require NTP to succeed first 😄.

closed 2023-09-09 by daytonajrw