MichaIng/DietPi #6582: update failed

wdenn, 2023-08-28:
---------------------------------------------------------------------
#### Details:
- Date           | Mon Aug 28 09:48:16 CEST 2023
- DietPi version | v8.20.1 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       | Odroid N2 (aarch64) (ID=15)
- Kernel version | `Linux server2 6.1.42-meson64 #2 SMP PREEMPT Thu Jul 27 06:50:53 UTC 2023 aarch64 GNU/Linux`
- Distro         | bullseye (ID=6)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://repos.influxdata.com/debian stable InRelease
Get:2 https://apt.grafana.com stable InRelease [5984 B]
Hit:3 https://deb.debian.org/debian bullseye InRelease
Hit:4 https://deb.debian.org/debian bullseye-updates InRelease
Hit:5 https://deb.nodesource.com/node_18.x bullseye InRelease
Hit:6 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:7 https://deb.debian.org/debian bullseye-backports InRelease
Err:2 https://apt.grafana.com stable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 963FA27710458545
Get:8 https://mirrors.xtom.de/armbian bullseye InRelease [53.3 kB]
Fetched 59.3 kB in 2s (37.3 kB/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://apt.grafana.com stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 963FA27710458545
E: Failed to fetch https://apt.grafana.com/dists/stable/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 963FA27710458545
E: Some index files failed to download. They have been ignored, or old ones used instead.
```### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
<!-- If you sent a "dietpi-bugreport", please paste the ID here -->
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
<!-- Explain how to reproduce the issue -->
1. ...
2. ...

#### Expected behaviour
<!-- What SHOULD be happening? -->
- ...

#### Actual behaviour
<!-- What IS happening? -->
- ...

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

Joulinar, 2023-08-28:
Sort quest, did you install Grafan from `dietpi-software` repository?

MichaIng, 2023-08-28:
The key was rotated. But actually I added a pre-patch. Can you check:
```sh
date -u '+%Y%m%d' -r '/etc/apt/trusted.gpg.d/dietpi-grafana.gpg'
```

vinhtq115, 2023-08-28:
I don't see any pre-patch so I tried [this guide](https://grafana.com/blog/2023/08/24/grafana-security-update-gpg-signing-key-rotation/) instead:
```
# Remove the old key
apt-key del 0E22EB88E39E12277A7760AE9E439B102CF3C0C6

# Verify that the GPG key was removed. This should return 0.
apt-key list | grep -i grafana | wc -l

# Download the new key and place it in the right location
sudo wget -q -O /usr/share/keyrings/grafana.key https://apt.grafana.com/gpg.key
```

Now I can update normally.

MichaIng, 2023-08-28:
Okay, seems you installed Grafana manually, otherwise a key at `/usr/share/keyrings/` would remain unused.

I'll mark this issue as closed then. Feel free to reopen if required.

closed 2023-08-28 by MichaIng