MichaIng/DietPi #6586: README.md contains an HTML document and not Markdown.

chathawa, 2023-08-29:
### Creating a bug report/issue

#### Required Information
- DietPi v8.21 for Raspberry Pi 2/3/4

MichaIng, 2023-08-30:
More precisely: It contains Markdown code with embedded HTML code, which is fine for most Markdown interpreters. See the entry page of this repo which shows the interpreted README.md. Someone did this to enhance it visually, with the logo centred etc, which is not possible with pure Markdown.

Any problem you have with this embedded HTML code?

chathawa, 2023-08-30:
I was working with it on a system without an X server. Lynx did it no
favors when I tried to view the HTML. It wasn't any better in a browser.
I'll look for a tool that can view it as MD. Thanks.

Best,
Clark Hathaway
Undergraduate at the University of Tennessee, Knoxville

MichaIng, 2023-09-01:
Markdown has the benefit that it is mostly readable in plain text as well. The HTML block indeed breaks this a little.

Browser are indeed usually not able to display Markdown, resp. only in plain text. But with a pipe ready Markdown interpreter/converter it works:
```sh
apt install pandoc
pandoc /boot/dietpi-README.md | lynx --stdin
```