MichaIng/DietPi #6588: Prowlarr service fails to start after latest software update to 1.8.4.3884

githubchonger, 2023-08-30:
### Creating a bug report/issue

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=8
G_DIETPI_VERSION_SUB=21
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version
0

- Kernel version
Linux NAS 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model
RPi 4 Model B (aarch64)

- Power supply used
CanaKit 3.5A USB-C Power Supply

- SD card used
SanDisk 128GB Extreme MicroSDXC

#### Additional Information (if applicable)
- Software title
Prowlarr

- Was the software title installed freshly or updated/migrated? 
Updated first using Prowlarr's install latest functionality and then using `dietpi-software reinstall 151`

- Can this issue be replicated on a fresh installation of DietPi?
Don't know

- Bug report ID
f289a3cc-35bf-4e2d-b3cc-0ed17ce9ce57

#### Steps to reproduce

1. Update Prowlarr using `dietpi-software reinstall 151`
2. Try to launch Prowlarr, observe that the service failed to start

#### Expected behaviour
Prowlarr should launch

#### Actual behaviour
Prowlarr is not launching

#### Extra details
I initially tried to updating to the latest using the Prowlarr UI.  When the service wouldn't come back up, I tried the dietpi-software reinstall route.  Neither seems to be working.

githubchonger, 2023-08-30:
Uninstalled and reinstalled and that fixed the issue.

MichaIng, 2023-09-01:
Let me reopen, as I want to try replicating it. Our Prowlarr implementation should support the web UI updater.

MichaIng, 2023-09-19:
Okay, it works all fine here as expected:
- Prowlarr can and should be updated via its web interface. If you do so, it first spawns an updater process which does the download and kills the parent Prowlarr process. As the systemd services is configured with `KillMode=process`, it does not kill the updater process once the main process is killed. Instead it restarts the main process. When this is done, the changelog shows up on the web UI automatically, along with a page reload button, finishing all successfully. The service log also shows this:
```
Sep 19 18:30:16 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Downloading update 1.8.6.3946
Sep 19 18:30:48 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Verifying update package
Sep 19 18:30:50 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Update package verified successfully
Sep 19 18:30:50 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Extracting Update package
Sep 19 18:31:05 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Update package extracted successfully
Sep 19 18:31:05 VM-Bullseye Prowlarr[909]: [Info] BackupService: Starting Backup
Sep 19 18:31:05 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Preparing client
Sep 19 18:31:09 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Starting update client /tmp/prowlarr_update/Prowlarr.Update
Sep 19 18:31:09 VM-Bullseye Prowlarr[909]: [Info] InstallUpdateService: Prowlarr will restart shortly.
Sep 19 18:31:39 VM-Bullseye systemd[1]: prowlarr.service: Main process exited, code=killed, status=9/KILL
Sep 19 18:31:39 VM-Bullseye systemd[1]: prowlarr.service: Failed with result 'signal'.
Sep 19 18:31:39 VM-Bullseye systemd[1]: prowlarr.service: Unit process 976 (Prowlarr.Update) remains running after unit stopped.
Sep 19 18:31:39 VM-Bullseye systemd[1]: prowlarr.service: Consumed 2min 17.533s CPU time.
Sep 19 18:31:39 VM-Bullseye systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 1.
Sep 19 18:31:39 VM-Bullseye systemd[1]: Stopped Prowlarr (DietPi).
Sep 19 18:31:39 VM-Bullseye systemd[1]: prowlarr.service: Consumed 2min 17.724s CPU time.
Sep 19 18:31:39 VM-Bullseye systemd[1]: prowlarr.service: Found left-over process 976 (Prowlarr.Update) in control group while starting unit. Ignoring.
Sep 19 18:31:39 VM-Bullseye systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Sep 19 18:31:39 VM-Bullseye systemd[1]: Started Prowlarr (DietPi).
Sep 19 18:31:41 VM-Bullseye Prowlarr[999]: [Info] Bootstrap: Starting Prowlarr - /opt/prowlarr/Prowlarr - Version 1.8.6.3946
```
- `dietpi-software reinstall 151` is basically a noop which skips the actual update intentionally, referring to the web UI, and only assures that dependencies are installed, the service user exists, regenerates the log symlinks etc. This also worked well here. It tells you about this:
```
[ INFO ] DietPi-Software | Prowlarr install dir "/opt/prowlarr" already exists. Download and install steps will be skipped.
 - If you want to update Prowlarr, please use the internal updater from web UI.
 - If you need to reinstall (e.g. broken instance), please manually remove the install dir and rerun "dietpi-software reinstall 151".
```
If you or anyone else faces any issue with either the web update or a reinstall, please provide the service logs:
```sh
journalctl -u prowlarr
```

closed 2023-09-19 by MichaIng

githubchonger, 2023-09-20:
So just updating from the web interface as usual should work for future updates then?

Joulinar, 2023-09-20:
yes that's the way. If there are issue, pls share logs as requested by @MichaIng

kinoushe, 2023-11-27:
Hi @MichaIng, I've just tried to update Prowlarr to `1.10.5.4116` using the web updater but it has failed, so attaching logs as requested above: [prowlarr-failed-update.txt](https://github.com/MichaIng/DietPi/files/13471975/prowlarr-failed-update.txt)

Joulinar, 2023-11-27:
This seems to be the issue

> `A fatal error was encountered. The library 'libhostpolicy.so' required to execute the application was not found in '/opt/prowlarr/'.`

kinoushe, 2023-11-27:
Thanks @Joulinar, I've had a look and the file is missing. I can try copying it back in from the release download

Joulinar, 2023-11-27:
Please reboot and share the whole log again

kinoushe, 2023-11-27:
After rebooting, still same error:

```
Nov 27 20:50:09 DietPi64 systemd[1]: Started prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:13 DietPi64 Prowlarr[687]: Unhandled exception. System.IO.FileNotFoundException: Could not load file or assembly 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.
Nov 27 20:50:13 DietPi64 Prowlarr[687]: File name: 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
Nov 27 20:50:13 DietPi64 systemd[1]: prowlarr.service: Main process exited, code=killed, status=6/ABRT
Nov 27 20:50:13 DietPi64 systemd[1]: prowlarr.service: Failed with result 'signal'.
Nov 27 20:50:14 DietPi64 systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 1.
Nov 27 20:50:14 DietPi64 systemd[1]: Stopped prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:14 DietPi64 systemd[1]: Started prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:14 DietPi64 Prowlarr[819]: Unhandled exception. System.IO.FileNotFoundException: Could not load file or assembly 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.
Nov 27 20:50:14 DietPi64 Prowlarr[819]: File name: 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
Nov 27 20:50:14 DietPi64 systemd[1]: prowlarr.service: Main process exited, code=killed, status=6/ABRT
Nov 27 20:50:14 DietPi64 systemd[1]: prowlarr.service: Failed with result 'signal'.
Nov 27 20:50:15 DietPi64 systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 2.
Nov 27 20:50:15 DietPi64 systemd[1]: Stopped prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:15 DietPi64 systemd[1]: Started prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:15 DietPi64 Prowlarr[864]: Unhandled exception. System.IO.FileNotFoundException: Could not load file or assembly 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.
Nov 27 20:50:15 DietPi64 Prowlarr[864]: File name: 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
Nov 27 20:50:15 DietPi64 systemd[1]: prowlarr.service: Main process exited, code=killed, status=6/ABRT
Nov 27 20:50:15 DietPi64 systemd[1]: prowlarr.service: Failed with result 'signal'.
Nov 27 20:50:16 DietPi64 systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 3.
Nov 27 20:50:16 DietPi64 systemd[1]: Stopped prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:16 DietPi64 systemd[1]: Started prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:16 DietPi64 Prowlarr[880]: Unhandled exception. System.IO.FileNotFoundException: Could not load file or assembly 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.
Nov 27 20:50:16 DietPi64 Prowlarr[880]: File name: 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
Nov 27 20:50:16 DietPi64 systemd[1]: prowlarr.service: Main process exited, code=killed, status=6/ABRT
Nov 27 20:50:16 DietPi64 systemd[1]: prowlarr.service: Failed with result 'signal'.
Nov 27 20:50:17 DietPi64 systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 4.
Nov 27 20:50:17 DietPi64 systemd[1]: Stopped prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:17 DietPi64 systemd[1]: Started prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:17 DietPi64 Prowlarr[904]: Unhandled exception. System.IO.FileNotFoundException: Could not load file or assembly 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.
Nov 27 20:50:17 DietPi64 Prowlarr[904]: File name: 'System.Runtime, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
Nov 27 20:50:17 DietPi64 systemd[1]: prowlarr.service: Main process exited, code=killed, status=6/ABRT
Nov 27 20:50:17 DietPi64 systemd[1]: prowlarr.service: Failed with result 'signal'.
Nov 27 20:50:18 DietPi64 systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 5.
Nov 27 20:50:18 DietPi64 systemd[1]: Stopped prowlarr.service - Prowlarr (DietPi).
Nov 27 20:50:18 DietPi64 systemd[1]: prowlarr.service: Start request repeated too quickly.
Nov 27 20:50:18 DietPi64 systemd[1]: prowlarr.service: Failed with result 'signal'.
Nov 27 20:50:18 DietPi64 systemd[1]: Failed to start prowlarr.service - Prowlarr (DietPi).
```

kinoushe, 2023-11-27:
I ended up removing the existing `/opt/prowlarr` directory and reinstalling with `dietpi-software reinstall 151`, which resolved the failed update.

Looking in the `/opt/prowlarr` directory there are now all 417 items, whereas after the failed update there were only about 300, so not sure what went wrong there.

kinoushe, 2023-11-27:
This was the first update since setting up Prolarr new a few weeks ago. I clicked update in the web UI and waited, eventually it just said the same update was still available, it hadn't applied, so I tried clicking the update again.

```
Filesystem     1K-blocks  Used Available Use% Mounted on
tmpfs            1968128     4   1968124   1% /tmp
```
Is this correct?

MichaIng, 2023-11-27:
> Is this correct?

Jep.

Hmm, probably clicking the update button multiple times can cause issues? I'll test it here.

kinoushe, 2023-11-27:
Great, thanks.

Yes, I'm thinking the same. Odd that it would let it proceed multiple times.

MichaIng, 2023-11-27:
Starting with v1.9.4.4039, when I hit the "Install latest" button on the web UI, it turns into a loading cycle which cannot be clicked again:
[image]
When I reload the page it can be again clicked at first, but doing so, this not trigger concurrent update attempts. And after some seconds, it turns automatically into the loading button again.

Did the button turn into the one shown in above image as well?
> eventually it just said the same update was still available,

What do you mean by this? In my case, after the update finished, the page is reloading automatically and then shows this:
[image]

And then
[image]

Probably next time, have an SSH session option when an update is offered. There run:
```sh
journalctl -fu prowlarr
```
This shows any new log entries immediately, so you can follow what happens when you hit the update button the first time, including possibly initial errors, so that the same update is offered again on the web UI.

Here is the full logs my end, one hitting the update button. The first entry (download) did take a while. During this time I did reload the web UI multiple times and hit the "Install latest" button 4 times all together, but the additional hits did not have any effect:
```
Nov 28 00:03:28 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Downloading update 1.10.5.4116
Nov 28 00:05:19 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Verifying update package
Nov 28 00:05:20 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Update package verified successfully
Nov 28 00:05:20 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Extracting Update package
Nov 28 00:05:48 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Update package extracted successfully
Nov 28 00:05:48 VM-Bookworm Prowlarr[926]: [Info] BackupService: Starting Backup
Nov 28 00:05:49 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Preparing client
Nov 28 00:05:53 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Starting update client /tmp/prowlarr_update/Prowlarr.Update
Nov 28 00:05:53 VM-Bookworm Prowlarr[926]: [Info] InstallUpdateService: Prowlarr will restart shortly.
Nov 28 00:06:34 VM-Bookworm systemd[1]: prowlarr.service: Main process exited, code=killed, status=9/KILL
Nov 28 00:06:34 VM-Bookworm systemd[1]: prowlarr.service: Failed with result 'signal'.
Nov 28 00:06:34 VM-Bookworm systemd[1]: prowlarr.service: Unit process 1017 (Prowlarr.Update) remains running after unit stopped.
Nov 28 00:06:34 VM-Bookworm systemd[1]: prowlarr.service: Consumed 3min 20.551s CPU time.
Nov 28 00:06:34 VM-Bookworm systemd[1]: prowlarr.service: Scheduled restart job, restart counter is at 1.
Nov 28 00:06:34 VM-Bookworm systemd[1]: Stopped prowlarr.service - Prowlarr (DietPi).
Nov 28 00:06:34 VM-Bookworm systemd[1]: prowlarr.service: Consumed 3min 20.657s CPU time.
Nov 28 00:06:34 VM-Bookworm systemd[1]: prowlarr.service: Found left-over process 1017 (Prowlarr.Update) in control group while starting unit. Ignoring.
Nov 28 00:06:34 VM-Bookworm systemd[1]: This usually indicates unclean termination of a previous run, or service implementation deficiencies.
Nov 28 00:06:35 VM-Bookworm systemd[1]: Started prowlarr.service - Prowlarr (DietPi).
Nov 28 00:06:39 VM-Bookworm Prowlarr[1039]: [Info] Bootstrap: Starting Prowlarr - /opt/prowlarr/Prowlarr - Version 1.10.5.4116
Nov 28 00:06:39 VM-Bookworm Prowlarr[1039]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Nov 28 00:06:39 VM-Bookworm Prowlarr[1039]: [Debug] Bootstrap: Console selected
Nov 28 00:06:39 VM-Bookworm Prowlarr[1039]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Nov 28 00:06:43 VM-Bookworm Prowlarr[1039]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/prowlarr]
Nov 28 00:06:45 VM-Bookworm Prowlarr[1039]: [Info] MigrationController: *** Migrating data source=/mnt/dietpi_userdata/prowlarr/prowlarr.db;cache size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3;busytimeout=100 ***
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: SQLite notice (283): recovered 5 frames from WAL file /mnt/dietpi_userdata/prowlarr/prowlarr.db-wal
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] FluentMigrator.Runner.MigrationRunner: 38: indexers_freeleech_only_config_contract migrating
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: Beginning Transaction
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] indexers_freeleech_only_config_contract: Starting migration of Main DB to 38
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] FluentMigrator.Runner.MigrationRunner: UpdateData
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: UPDATE "Indexers" SET "ConfigContract" = 'HDSpaceSettings' WHERE "Implementation" = 'HDSpace'
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] FluentMigrator.Runner.MigrationRunner: => 0.3201118s
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] FluentMigrator.Runner.MigrationRunner: UpdateData
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] NzbDrone.Core.Datastore.Migration.Framework.NzbDroneSQLiteProcessor: UPDATE "Indexers" SET "ConfigContract" = 'ImmortalSeedSettings' WHERE "Implementation" = 'ImmortalSeed'
Nov 28 00:06:46 VM-Bookworm Prowlarr[1039]: [Info] FluentMigrator.Runner.MigrationRunner: => 0.0097707s
[... cut ...]
Nov 28 00:06:50 VM-Bookworm Prowlarr[1039]: [Info] CommandExecutor: Starting 3 threads for tasks.
Nov 28 00:06:55 VM-Bookworm Prowlarr[1039]: [Info] Microsoft.Hosting.Lifetime: Application started. Press Ctrl+C to shut down.
Nov 28 00:06:55 VM-Bookworm Prowlarr[1039]: [Info] Microsoft.Hosting.Lifetime: Hosting environment: Production
Nov 28 00:06:55 VM-Bookworm Prowlarr[1039]: [Info] Microsoft.Hosting.Lifetime: Content root path: /
```
As can be seen, after the update and service restart, some database migration steps are done.

kinoushe, 2023-11-27:
Thanks for investigating further. I'll keep in mind to watch the log on next update.

When I clicked "Install Latest" the loading cycle icon loaded as above, and in the bottom left it was saying downloading etc. I left this go, and when checked back it didn't appear to be doing anything, so thought it was finished.

When I refreshed the page there was no pop-over saying it was finished and needed to reload the page. Instead it was still on the old version with the "Install Latest" button visible again.

I've updated the other *arr apps without problem as per your screenshots above, it's just this one that failed.

MichaIng, 2023-11-27:
Ah, I forgot to check or ask whether there is an auto-updater, so that it is possible that an automatic update attempt failed, causing the issues for the manual update (like non-empty `/tmp/prowlarr_update/`)?

kinoushe, 2023-11-27:
I'm just using the stock `dietpi-software` install, and haven't enabled automatic updates in Prowlarr's settings (I like to manually run updates in case of problems)

MichaIng, 2023-11-27:
Okay, so there is a setting, but it is not enabled by default? We do not touch the settings, so it will be the defaults.