MichaIng/DietPi #6589: SKIP_ARCHIVE doesn't appear to work

dirkhh, 2023-08-31:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)
This is a bug report related to the build system, so I think none of the above information really is relevant.
The last test was done with last night's `dev` branch (so my commit on top of bdde5bd73)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. export SKIP_ARCHIVE and ensure that it is set:
```
$ SKIP_ARCHIVE=1
$ sudo -E env | grep SKIP_ARCHIVE
SKIP_ARCHIVE=1
```
2. run the image builder
```
$ sudo -E ./dietpi-build -m 82 -o dirkhh -b adsb-feeder
```

#### Expected behaviour

- the script should honor the request and end with the `.img` and not waste time by compressing and deleting the `.img`, given that all I want to do is write the `.img` to an SD card so I can test it.

#### Actual behaviour

- at the end of the build a `.7z` is created and the `.img` is deleted

#### Extra details

- I tried to spot the part of the code where the value gets lost but I don't get it... [this](https://github.com/MichaIng/DietPi/blob/bdde5bd732142a25468e309d3a9473b1c9407ad8/.build/images/dietpi-imager#L662) should end the build with the `.img` file...

MichaIng, 2023-08-31:
Try to export the variable:
```sh
export SKIP_ARCHIVE=1
```

dirkhh, 2023-08-31:
Sorry, should have provided the full log, I guess.
Yes, I am of course using `export`.
And I'm using `sudo -E` to ensure the environment gets passed on.

[image]
How it started
[image]
How it ends

MichaIng, 2023-09-01:
Ah you actually use `dietpi-build`. Yeah it expects and archive at some points and hence explicitly sets/overwrites all `dietpi-imager` variables. I actually added this flag for `dietpi-build` as for some VM images we need it unpacked to convert to virtual disk image and pack the appliance.

MichaIng, 2024-12-15:
A `--skip-archive` flag has been implemented into `dietpi-build`, and is passed through to `dietpi-imager` respectively: https://github.com/MichaIng/DietPi/commit/269eba3

closed 2024-12-15 by MichaIng